paulober/MicroPico #2: can't seem to install this extension

migs35323, 2022-08-17:
## What are the steps to reproduce this issue?
1. installing pico w go from  extensions page
2. optional:  restart vscode
3. do anything with the pico w go

## What happens?
every pico w go menu crashes saying stuff like it can't start the files
and there is no menu on status bar like it mentioned in the docs

## What were you expecting to happen?
that i could use it <3

## Any logs, error output, etc?
*(If it’s long, please paste to https://gist.github.com and insert the link here)*
[image]

## Any other comments?

ubuntu 22.04 vscode installed from snap and updated (if that matters) python 3.10.4
everything updated to last stuff

## Support info

Copy this from the `Pico-W-Go > Help > Get support info` option in the command palette:

```
it also errors with command not found
```

Warp8, 2022-08-23:
I also have this problem on Manjaro. It works on my windows desktop so I assume it probably a Linux issue.

Warp8, 2022-08-24:
Update: after much research I have learned that this is a problem with the latest version of VS code having an issue with loading some micro-controller programming extensions. They say you can rollback your installation, but I didn't have any luck.

paulober, 2022-08-27:
I am sorry to hear that you are having these problems with the extension. I don't know how the vscode snap version differs from the others in terms of loading the extension, but maybe that could be the reason for skipping the initialization of the extension which would initialize all the commands and the taskbar. Maybe i get this problem fixed with version 2.0 of this extension coming (hopefully) in the next time

paulober, 2022-09-05:
Should be fixed now in version 2.0.0+
Can you confirm?

Warp8, 2022-09-05:
On VS code 1.71.0 and on the plugin release from 9/5/2022 it still does not work for me.

paulober, 2022-09-05:
Ok, can you reproduce this error and then open ther visual studio code terminal -> select the output pane -> select Protocoll (Extensionhost) in the dropdown can past the contenct into pastebin, gist or something like that. This constains log information why the extension crashes on you setup.

migs35323, 2022-09-05:
it's in a big loop now, its worse lol
it's spamming theese window non stop
closed vscode and opened again, i can't use vscode lol

[image]

found this in the log shared tab not sure if its your extention or other:
`[2022-09-05 23:49:18.168] [sharedprocess] [error] [uncaught exception in sharedProcess]: Could not find pty on pty host: CodeExpectedError: Could not find pty on pty host
    at c._throwIfNoPty (/snap/code/106/usr/share/code/resources/app/out/vs/platform/terminal/node/ptyHostMain.js:23:36316)
    at c.resize (/snap/code/106/usr/share/code/resources/app/out/vs/platform/terminal/node/ptyHostMain.js:23:33195)
    at Object.call (/snap/code/106/usr/share/code/resources/app/out/vs/platform/terminal/node/ptyHostMain.js:16:8386)
    at u.onPromise (/snap/code/106/usr/share/code/resources/app/out/vs/platform/terminal/node/ptyHostMain.js:15:5851)
    at u.onRawMessage (/snap/code/106/usr/share/code/resources/app/out/vs/platform/terminal/node/ptyHostMain.js:15:5234)
    at /snap/code/106/usr/share/code/resources/app/out/vs/platform/terminal/node/ptyHostMain.js:15:4520
    at _.invoke (/snap/code/106/usr/share/code/resources/app/out/vs/platform/terminal/node/ptyHostMain.js:11:145)
    at D.deliver (/snap/code/106/usr/share/code/resources/app/out/vs/platform/terminal/node/ptyHostMain.js:11:2275)
    at T.fire (/snap/code/106/usr/share/code/resources/app/out/vs/platform/terminal/node/ptyHostMain.js:11:1853)
    at process.Q (/snap/code/106/usr/share/code/resources/app/out/vs/platform/terminal/node/ptyHostMain.js:9:20703)
    at process.emit (node:events:526:28)
    at emit (node:internal/child_process:938:14)`

the log extension host shows this, nothing important i believe
`[2022-09-05 23:49:37.790] [exthost] [info] Extension host with pid 5582 started
[2022-09-05 23:49:37.924] [exthost] [info] ExtensionService#_doActivateExtension vscode.microsoft-authentication, startup: false, activationEvent: 'onAuthenticationRequest:microsoft'
[2022-09-05 23:49:37.958] [exthost] [info] ExtensionService#_doActivateExtension vscode.debug-auto-launch, startup: true, activationEvent: '*'
[2022-09-05 23:49:37.962] [exthost] [info] ExtensionService#_doActivateExtension vscode.git-base, startup: true, activationEvent: '*', root cause: vscode.git
[2022-09-05 23:49:37.967] [exthost] [info] ExtensionService#_doActivateExtension vscode.ipynb, startup: true, activationEvent: '*'
[2022-09-05 23:49:38.069] [exthost] [info] ExtensionService#_doActivateExtension vscode.git, startup: true, activationEvent: '*'
[2022-09-05 23:49:38.102] [exthost] [info] ExtensionService#_doActivateExtension vscode.github, startup: true, activationEvent: '*'
[2022-09-05 23:49:38.150] [exthost] [info] Eager extensions activated
[2022-09-05 23:49:38.170] [exthost] [info] ExtensionService#_doActivateExtension vscode.emmet, startup: false, activationEvent: 'onStartupFinished'
[2022-09-05 23:49:38.181] [exthost] [info] ExtensionService#_doActivateExtension vscode.merge-conflict, startup: false, activationEvent: 'onStartupFinished'
[2022-09-05 23:49:38.186] [exthost] [info] ExtensionService#_doActivateExtension vscode.npm, startup: false, activationEvent: 'onStartupFinished', root cause: esbenp.prettier-vscode
[2022-09-05 23:49:38.207] [exthost] [info] ExtensionService#_doActivateExtension ms-vscode-remote.remote-containers, startup: false, activationEvent: 'onStartupFinished'
[2022-09-05 23:49:38.283] [exthost] [info] ExtensionService#_doActivateExtension PKief.material-icon-theme, startup: false, activationEvent: 'onStartupFinished'
[2022-09-05 23:49:38.301] [exthost] [info] ExtensionService#_doActivateExtension vscode.github-authentication, startup: false, activationEvent: 'onAuthenticationRequest:github'
[2022-09-05 23:49:38.432] [exthost] [info] ExtensionService#_doActivateExtension esbenp.prettier-vscode, startup: false, activationEvent: 'onStartupFinished'`

and here's what is on pty host

`[2022-09-05 23:49:16.488] [ptyhost] [warning] Shell integration cannot be enabled for executable "/usr/bin/python3" and args ["/home/ubuntu/.vscode/extensions/paulober.pico-w-go-2.0.3-linux-x64/dist/terminalExec.py","1972"]
[2022-09-05 23:49:17.694] [ptyhost] [warning] Shell integration cannot be enabled for executable "/usr/bin/python3" and args ["/home/ubuntu/.vscode/extensions/paulober.pico-w-go-2.0.3-linux-x64/dist/terminalExec.py","2332"]
[2022-09-05 23:49:18.325] [ptyhost] [warning] Shell integration cannot be enabled for executable "/usr/bin/python3" and args ["/home/ubuntu/.vscode/extensions/paulober.pico-w-go-2.0.3-linux-x64/dist/terminalExec.py","1920"]
[2022-09-05 23:49:18.853] [ptyhost] [warning] Shell integration cannot be enabled for executable "/usr/bin/python3" and args ["/home/ubuntu/.vscode/extensions/paulober.pico-w-go-2.0.3-linux-x64/dist/terminalExec.py","1909"]
[2022-09-05 23:49:19.352] [ptyhost] [warning] Shell integration cannot be enabled for executable "/usr/bin/python3" and args ["/home/ubuntu/.vscode/extensions/paulober.pico-w-go-2.0.3-linux-x64/dist/terminalExec.py","2264"]
[2022-09-05 23:49:19.861] [ptyhost] [warning] Shell integration cannot be enabled for executable "/usr/bin/python3" and args ["/home/ubuntu/.vscode/extensions/paulober.pico-w-go-2.0.3-linux-x64/dist/terminalExec.py","1962"]
[2022-09-05 23:49:20.384] [ptyhost] [warning] Shell integration cannot be enabled for executable "/usr/bin/python3" and args ["/home/ubuntu/.vscode/extensions/paulober.pico-w-go-2.0.3-linux-x64/dist/terminalExec.py","1394"]
[... cut ...]
[2022-09-05 23:49:33.314] [ptyhost] [warning] Shell integration cannot be enabled for executable "/usr/bin/python3" and args ["/home/ubuntu/.vscode/extensions/paulober.pico-w-go-2.0.3-linux-x64/dist/terminalExec.py","1368"]
[2022-09-05 23:49:34.164] [ptyhost] [warning] Shell integration cannot be enabled for executable "/usr/bin/python3" and args ["/home/ubuntu/.vscode/extensions/paulober.pico-w-go-2.0.3-linux-x64/dist/terminalExec.py","2065"]
[2022-09-05 23:49:35.224] [ptyhost] [warning] Shell integration cannot be enabled for executable "/usr/bin/python3" and args ["/home/ubuntu/.vscode/extensions/paulober.pico-w-go-2.0.3-linux-x64/dist/terminalExec.py","1643"]`

paulober, 2022-09-06:
I just tested the extension on manjaro Linux at i think i found the problem. Expect fix soon.

closed 2022-09-06 by paulober

paulober, 2022-09-06:
@migs35323 should be fixed now with v2.0.4. Can you confirm?

migs35323, 2022-09-06:
@paulober thank you a lot for the fixing and quick actions, i will be able to test later today, will provide feedback once i have the change to try

commit 9f32f7ff (2022-09-06): Fixes #2, terminal constantly crashing on linux