paulober/MicroPico #3: Manual Connect doesn't work: 'Address not configured. Please go to the settings to configure a valid address or comport'

MySurmise, 2022-08-21:
## What are the steps to reproduce this issue?
1. Install extension
2. Change two things in the global settings:
`"auto_connect":true,"manual_com_device":"COM7",`
3. Click on the "Pico W Disconnected" button in the bottom left

## What happens?
I get an error message: `Address not configured. Please go to the settings to configure a valid address or comport`

## What were you expecting to happen?
The extension tries to connect to COM7 and either connects or doesn't connect, but shows an appropriate error message.

## Any logs, error output, etc?
see above

## Any other comments?
First of all, thank you for continuing the development of this great extension.
I already tried a lot, but other forums don't know this extension, so I hope you can help.

## Support info

When I clicked on 'All commands' > 'Help > Get Support Info', nothing was outputted in the terminal, but the other buttons still printed output afterwards, so the extension did not crash.
I also recorded a video with everything (keep in mind this is a completely fresh install of the extension):

https://user-images.githubusercontent.com/70728036/185813684-ede596c9-4efc-4875-879d-5ae9504cb548.mp4

```
< Paste Here >
```

MySurmise, 2022-08-21:
I am using Windows 11 and Python 3.10.0

paulober, 2022-08-27:
Ok, this is strange... Try `All Commands -> Configure Project` first, then connect the Pico to your PC via USB without changing the global settings. The extension should then automatically detect and connect to the Pico (if you have previously flashed the latest MicroPython firmware to it). In the meantime, I'll look into the problem with "manual_com_device" and try to fix it or report a solution if I find one. I hope the above method works for you and happy coding.

MySurmise, 2022-08-27:
I tried this, sadly, this did not work. I tried the extension on another PC with Windows 10, it worked there. But the support info worked on neither. Maybe you can push an update with a working "Support info" button, so you can find out what's wrong?

MySurmise, 2022-08-27:
I have one additional question: My Intellisense for Pico also does not work on either PC.
[image]

Example: The "machine" module does not get recommended.
Do you have an idea why this is and how I coud fix this?

paulober, 2022-08-28:
> I have one additional question: My Intellisense for Pico also does not work on either PC. [image]
> 
> Example: The "machine" module does not get recommended. Do you have an idea why this is and how I coud fix this?

Have you done `All Commands -> Configure Project` on the computer where to extension works. And make sure to install the recommended extensions in .vscode/extensions.json because they are required for the intellisense.
```json
{
    "recommendations": [
        "ms-python.python",
        "visualstudioexptteam.vscodeintellicode",
        "ms-python.vscode-pylance"
    ]
}

paulober, 2022-08-28:
> I tried this, sadly, this did not work. I tried the extension on another PC with Windows 10, it worked there. But the support info worked on neither. Maybe you can push an update with a working "Support info" button, so you can find out what's wrong?

Ok, i have no idea why this doesn't work for you. I can only say wait for the next update. I hopefully will fix your problems...

MySurmise, 2022-08-29:
> Have you done `All Commands -> Configure Project` on the computer where to extension works. And make sure to install the recommended extensions in .vscode/extensions.json because they are required [...]

I'm sorry, apparently I disabled Pylance and enabled Jedi because it was giving me some weird (and wrong) recommendations. Anyway, I reenabled it and it works now and the wrong recommendations are gone no, so thanks!

paulober, 2022-09-05:
@MySurmise 

The reason for that is that you have to deactivate auto_connect in order to make use of the manual_com_port. The auto_connect means it automaticly searches for a valid COM port and tries it out and therefor it skips manual_com_port.

Now that finaly version 2.0.0 is out, the settings moved to VS Code integrated settings so you can configure this now via the GUI (VS Code settings) or your local settings.json in .vscode. For reference please see README.md where no the settings are explained in more detail.

Does this now work for you?

paulober, 2022-09-12:
After 7 days of no response, I will mark this issue as closed. If the issue has not been resolved, feel free to reopen it.

closed 2022-09-12 by paulober

jerrens, 2022-10-11:
To work around this problem, I had to:
1. Disconnect the device that was created COM8 and COM9 (a Bluetooth headset) 
2. Change the COM port number of my Pico to be the highest port number used (instructions below)
3. Disconnect the Pico
4. Delete the 'pymakr.devices.configs' section of my user settings in VS Code
5. Reconnect the Pico
6. Reload VS Code (F1 > Developer: Reload Window)
7. Configure the project for Pico-Go (F1 > Pico-W-Go > Configure project)
8. Attempt to connect

For those not familiar with the steps to change a COM port number in Windows:
1. Press WIN + X
2. Choose 'Device Manager'.  If prompted by UAC, click 'Yes' to allow
9. Scroll down to 'Ports (COM & LPT), expand the group, then right-click on your device (it must be connected to your computer at this point, it is likely labeled "USB Serial Device") and choose 'Properties'  (you could also double-click on the device)
10. In the Properties dialog, switch over to Port settings, and click on the 'Advanced' button
11. At the bottom of the Advanced Settings dialog, there is a drop down menu for the COM port Number.  You must choose one that isn't being used by another device (marked).
12. Once the new com port number has been selected, 'OK' out of each dialog and then close the device manager