paulober/MicroPico #4: command 'picowgo.listCommands' not found

GuybrushThreepwood75, 2022-08-25:
## What are the steps to reproduce this issue?
1. New install of extension
2. try to open the "All commands"

## What happens?
- Error-windows with "command 'picowgo.listCommands' not found"

## Support info
Version: 1.70.2 (user setup)
Commit: e4503b30fc78200f846c62cf8091b76ff5547662
Date: 2022-08-16T05:35:13.448Z
Electron: 18.3.5
Chromium: 100.0.4896.160
Node.js: 16.13.2
V8: 10.0.139.17-electron.0
OS: Windows_NT x64 10.0.19044

paulober, 2022-08-27:
I have no idea what could be the problem here. I hope that the extension will work for you after the next updates. I will come back to this problem. But anyway thanks for reporting this error.

LGH, 2022-09-03:
I have the same problem. Just installed the extension in VS Code and no Pico-Go-W commands work:

* Command 'Pico-W-Go > Configure project' resulted in an error (command 'picowgo.initialise' not found)

```
Version: 1.71.0
Commit: 784b0177c56c607789f9638da7b6bf3230d47a8c
Date: 2022-09-01T07:25:38.437Z (2 days ago)
Electron: 19.0.12
Chromium: 102.0.5005.167
Node.js: 16.14.2
V8: 10.2.154.15-electron.0
OS: Darwin x64 21.6.0
Sandboxed: No
```

paulober, 2022-09-05:
@GuybrushThreepwood75 and @LGH great news should be fixed now with version 2.0.0+

Can you check this on your machines please?

paulober, 2022-09-12:
After 7 days of no response, I will mark this issue as closed. If the issue has not been resolved, feel free to reopen it.

ollar, 2022-09-16:
Version: 1.71.2

command 'picowgo.connect' not found

seems the problem is still actual

paulober, 2022-09-16:
> 

Ok, but can you please provide some more information like: Your OS and it's version, do you use the current version of the extension, do you have Python 3.9+ installed....

LGH, 2022-09-18:
Sorry, I've been away for work so I couldn't report back (only running this on my personal laptop).

Issue still persists. 

Python3.9 (3.9.13) and 3.10 both installed using Homebrew and available in /usr/local/bin/ (python3 linked to 3.10.6). pico-w-go v2.0.7, MacOS 12.6 (Intel), VS Code as per below.

`Version: 1.71.2
Commit: 74b1f979648cc44d385a2286793c226e611f59e7
Date: 2022-09-14T21:05:37.721Z (3 days ago)
Electron: 19.0.12
Chromium: 102.0.5005.167
Node.js: 16.14.2
V8: 10.2.154.15-electron.0
OS: Darwin x64 21.6.0
Sandboxed: No`

paulober, 2022-09-18:
Ok, tanks. I currently don't have a MacOS machine available for testting but i assume that the problem is caused how to python installations are installed but it's juts a guess. Can you please leave your system as it is now and reproduce the error and than copy the part which concerns "Pico-W-Go" from the "Terminal > output > (Select something with "(Extensionhost)")" and paste it here.

paulober, 2022-09-18:
> Sorry, I've been away for work so I couldn't report back (only running this on my personal laptop).
> 
> Issue still persists.

No problem, thanks for the response, but i've also ment @ollar.

LGH, 2022-09-19:
If I select the Output window and "Log (Extension host)", there is nothing related to Pico-W-Go there. Did I misinterpret what you wanted us to do?

paulober, 2022-09-24:
If you press `Ctrl+F` or the macos alternative of it inside the "Log (Extension host)" tab and type "picowgo" and it doesn't find anything then it looks like the extension is not installed. Try to uninstall it and the restart VS Code, install it again. Open VS Code, open a folder and run the "Pico-W-Go > Configure project" VS Code command inside the new project.

paulober, 2022-09-24:
@ollar 
Have you run "Pico-W-Go > Configure project" in the project folder you opened in VS Code first?

Drmikej, 2022-11-01:
I have the same problem - Pi 400 running PiOS latest Python 3.92
Clean install of PiOs Not other additional software installed.
Fresh install of VS Code
Installation of all extensions listed in readme.
Install Pico-W-go
set up hello.py with hello world
try "Pico-W-Go > Configure project"
and get command picowgo intialise not found
I am connecting via remote desktop

paulober, 2022-11-01:
@Drmikej Ok, sad to here that you have problems with the extension. Would you like to give me some debuging output from the terminal -> dropdown -> ... extension host or sth like that -> there should be something concerning picowgo.

Drmikej, 2022-11-02:
And just found the error when trying to a configure:

paulober, 2022-11-02:
@Drmikej I found the issue. I'll send the details later. As for now you can try to install the newest vsix package from the github releases and see if it works if you want.

Drmikej, 2022-11-02:
Just tried pico-w-go-2.1.2.vsix
same error - paulober.pico-w-go' failed: No native build was found for platform=linux
Sorry :-(

paulober, 2022-11-02:
Ok, the reason why it's not running is because support for linux-arm has been droped for this extension and it cannot start on your platform because the extension doesn't provide native build for this platform currently. If you setup your Raspberry Pi with the new 64-bit Raspbian it should work as the extension supports linux-arm64.

Drmikej, 2022-11-03:
That isn't good news - there are a lot of 32 bit Raspbian installations after all its the default.
How difficult would it be to add 32 bit support?
The original pico-go supported it.

Drmikej, 2022-11-03:
I can confirm that it does work on PiOS 64. 
On reflection I'm not sure that 32-bit support is essential as the move to 64-bit as the default has to be sometime soon.
If it was easy to add then it is worth doing as the majority of Pi installations are 32-bit at the moment but I don't think its a deal breaker given how easy it is to change to PiOS 64 - I think adding a note to the effect that you have to use PiOS 64 is a good and cheap idea :-)   This is what it is planned to say in the new edition of the book.
PS if you would like a copy - no problem

paulober, 2022-11-04:
@Drmikej So i looked some things up and as i know the RBPI4 and 400 uses a armv8 based CPU and the extension only includes binaries for armv6 and armv7 or the 64 arch equivalent. But in the 32-bit mode i thought Raspbian is running in armv7 compatibility mode so it can run armv7 binaries. You can see all included prebuilds here: https://serialport.io/docs/guide-platform-support#supported-platforms-and-architectures
The reason why pico-go supported the 32-bit RBPI is that the developer generated binaries for each VS Code -Electron - Node- OS - Arch combo when a new one released. With version v2.1.0 i moved over to that prebuild one to avoid crashes when VS Code has a new updated and so on... So the reason why your v2.0.5 version of this ext isn't running on your pi is because it doesn't include binaries compatible with the newest VS Code version.

I might put a note into readme for raspberry pi users as it seems to me that there are lots of Pico developers using a RBPI 4 or 400 for dev.
Which book are you talking about?

Drmikej, 2022-11-05:
I'm not sure I follow because I'm not familiar with VS Code extensions. Adding the note seems like the best solution as I have been happy using the 64 bit OS with Pico-W-Go.
From previous comment:
Also I'm the coauthor of https://www.amazon.com/dp/1871962692 and harry and I are working on the update for the PicoW. The first edition used PicoGo and this second edition is planning to use Pico-W-Go. If we can't get it to work on the Pi 400 we will probably drop VS Code from the edition. So we are keen to help but know little about VS Code extensions.

paulober, 2022-11-05:
@Drmikej I added a note in the wiki and the README shows for linux only support for x64 and arm64. 
Sounds great, what problem do you still have with the extension on the Pi 400 as i thought the info about the 64-bit os solved your issue?

Drmikej, 2022-11-05:
Making the switch to PiOS 64 solves all problems - but most users think that an arm64 extension will work on PiOS 32 - I think its all solved now.

paulober, 2022-11-05:
@Drmikej Ok, glad to hear. I'm looking forward to seeing the expansion referenced in a book. If there are any problems, please feel free to create an issue.

Nabla128k, 2022-11-08:
Hi, so if I understand it correctly support for 32bit arm OS has been dropped forever?
Is it too complicated to provide step by step how to build extension at home for 32bit use? For me, currently, switching to 64bit is not an option, unfortunately.

paulober, 2022-11-09:
@Nabla128k Yes, i think so.
Details on how to generate native bindings for the Raspberry Pi OS 32-bit i refer to this section of the serialport.io documentation: https://serialport.io/docs/guide-installation#compilation-on-unsupported-systems

paulober, 2022-11-09:
@Nabla128k Maybe could help if you can paste your output of the vscode help -> info dialog option where your electron, vscode,  os and many other versions are listed.

paulober, 2022-11-09:
@Nabla128k Version v2.1.3 added support for armv7 (aka Raspberry Pi OS 32-bit support on Raspberry Pi 3b+ or 4b).

closed 2022-11-09 by paulober

Drmikej, 2022-11-11:
Thanks! 
I just tried it and it all seems to work on Pi OS 32 and I will change what the book says to indicate this.
Excellent work and much appreciated  :-)

ollar, 2022-11-24:
@paulober sorry, github did not mentioned me in any way :(
>  Have you run "Pico-W-Go > Configure project" in the project folder you opened in VS Code first?
that helped me, and now vs-code can see the board and the plugin can connect to it.
Thanks a lot!

ningelsohn, 2022-12-20:
I had the same Error and the reason was, that although I reinstalled vscode, another pico extension still was installed which somehow interfered with the new one.

commit 4b10f40d (2022-11-09): v2.1.3

Added support for armv7 which also fixes #4.