paulober/MicroPico #8: First time I "Run current file": `No module named ...`, 2nd time is OK

sgbaird, 2022-08-27:
## What are the steps to reproduce this issue?
1. Follow basic setup instructions
2. Create a module at the same level as the file to run, e.g. `secrets.py`
3. Import the module in the file to run
4. Click "Run current file"

## What happens?
Can't import the module the first time, imports when clicked again.

## What were you expecting to happen?
Import the module per normal.

## Any logs, error output, etc?
*(If it’s long, please paste to https://gist.github.com and insert the link here)*
```python
>>> Running c:\Users\sterg\Documents\GitHub\sparks-baird\self-driving-lab-demo\scripts\pico-go-w\microdot_hello.py

♦Traceback (most recent call last):
  File "<stdin>", line 4, in <module>
ImportError: no module named 'secrets'

>>> Running c:\Users\sterg\Documents\GitHub\sparks-baird\self-driving-lab-demo\scripts\pico-go-w\microdot_hello.py
connected
```

## Any other comments?
It's an every other time sort of thing.

[Screencast: ](https://www.screencast.com/t/YSr5iYyXY)

## Support info

Copy this from the `Pico-W-Go > Help > Get support info` option in the command palette:

[image]

Pico-Go-W `v1.5.6`
OS: Windows 11 Home

paulober, 2022-08-27:
Thank you for mentioning. This one will be a bit more complicated to fix but i aim to fix this as soon as possible PRs wellcome.

I've been thinking about rewriting the extension in Typescript from ground up, since the current codebase has far too many dependencies on old, vulnerable versions of packages, and the asynchronous mess is a pain to maintain. So maybe then I will implement a fix for this directly when implementing the "run" function.

paulober, 2026-09-14:
Thanks for the report! This is the same problem as #44, so I'm closing this one to keep everything in one place.

closed 2026-09-14 by paulober