paulober/MicroPico #9: Pico-Go status bar is visible in non-pico projects

timstrasser, 2022-08-27:
## What are the steps to reproduce this issue?
1. Install extension
2. Open any folder in VSCode

## What happens?
The Pico-Go status bar appears, no matter if the current folder is a Pico project or not.

## What were you expecting to happen?
I would expect the Pico-Go extension, to be loaded only if I'm inside a Pico project.

## Any logs, error output, etc?
*(If it’s long, please paste to https://gist.github.com and insert the link here)*

## Any other comments?

## Support info

Copy this from the `Pico-W-Go > Help > Get support info` option in the command palette:

```
Get support info" does not print anything into console.
```

paulober, 2022-08-27:
I'm also anoyed by this behaviour and try to fix this in an upcoming version. Thanks for mentioning...

timstrasser, 2022-08-27:
Thanks. :) If I'm not mistaken, this problem did not occur in the latest cpwood repo. I'm not sure what caused the issue, though.

paulober, 2022-08-27:
Ok, that's wired... As mentioned in #8 I'm currently working on a clean re-write with up-to date dependencies etc.... Should I make good progress with this this should probably fix a lot of these issues.

timstrasser, 2022-08-27:
Looking forward to that! Do you plan on also rewriting the automation scripts? (e.g. for generating the stubs and the bindings)

paulober, 2022-08-27:
Maybe via my Azure DevOps, but that depends on the dependency I will use for serial communication. For example the current serialport package does not work in its latest version with the extension so maybe there are some changes in the requirement for the native bindings. And no, I probably won't rewrite the script that generates the stubs on the pico, if that's what you mean. I was thinking of reusing the stubs repo that already exists and just rewriting the extension and the communication between the three parts. If I have enough time for it...

paulober, 2022-09-12:
Should be fixed since update 2.0.0, can you confirm?

paulober, 2022-09-18:
After 7 days of no response, I will mark this issue as closed. If the issue has not been resolved, feel free to reopen it.

closed 2022-09-18 by paulober