paulober/MicroPico #10: Unexpected token { in JSON at position 120

timstrasser, 2022-08-28:
## What are the steps to reproduce this issue?
1. Install extension
2. run "Download" command or try connecting to the FTP server

## What happens?
If I try to download the project from my Pico, this gets printed in the console:
```
Downloading from / ...
Reading files from board
Download failed.Unexpected token { in JSON at position 120
Download done, resetting board...
```
Similarly, if I connect to the FTP server via FileZilla, this gets printed in the FileZilla console:
```
Command:	LIST
Response: 	451 Unexpected token { in JSON at position 120
```

## What were you expecting to happen?
Download/FTP to work as expected.

## Any logs, error output, etc?
*(If it’s long, please paste to https://gist.github.com and insert the link here)*

## Any other comments?

## Support info

Copy this from the `Pico-W-Go > Help > Get support info` option in the command palette:

```
Get support info does not print anything...
```

paulober, 2022-09-05:
I have now idea what caused this problem. Maybe the on pico project file... 🤷‍♂️
But anyway should be fixed now with release 2.0.0+

Can you please test if it now works for you. Maybe you first have to delete your .vscode folder or the stubs in it as well as the folder Pico-W-Stub in %APPDATA%\Code\User as well as the files on the Pico or at least the project.pico-w-go file.

paulober, 2022-09-12:
After 7 days of no response, I will mark this issue as closed. If the issue has not been resolved, feel free to reopen it.

closed 2022-09-12 by paulober

timstrasser, 2022-09-29:
The issue persists for me, even in the newest version.