paulober/MicroPico #14: unsupported operator errors in typings for `int`s?

Karel-Kroeze, 2022-09-05:
## What are the steps to reproduce this issue?
1. Start a project
2. use Pico-W-Go to generate project files (Configure Project)
3. attempt to compare integers (e.g., `while max_wait > 0:`)

## What happens?
[image]

## What were you expecting to happen?
no errors

## Any other comments?
The stubs packaged with the vanilla Pico-Go extension seem to include more files, and do not exhibit this problem. Removing the typeshed line from the generated pylance settings also removes the operator errors (but of course now complains about micropython imports).

## Support info

Copy this from the `Help -> Info -> Copy` option in Visual Studio Code:

_There is no 'Info' option under Help, this is the output from 'About'_

```
Version: 1.70.2 (system setup)
Commit: e4503b30fc78200f846c62cf8091b76ff5547662
Date: 2022-08-16T05:35:13.448Z
Electron: 18.3.5
Chromium: 100.0.4896.160
Node.js: 16.13.2
V8: 10.0.139.17-electron.0
OS: Windows_NT x64 10.0.22622
```

paulober, 2022-09-05:
Thank you for reporting this bug (next time please open it in github.com/paulober/Pico-W-Stub).

I also had errors in the linting concerning basic things. So after some searching it seems like the problem is in the builtin stubs. So i grabt the official ones from python/typeshed which should also be the one make this linting work when no custom settings were made. Expect PR with fix soon...

Ok i think the "About" is renamed in the german translation to "Info". I'll correct this in the template...

closed 2022-09-05 by paulober

paulober, 2022-09-05:
No problem and glad to here that it fixes your issue. 

It should have a mention, when you try to create a new issue, but maybe i have to describe it a bit more detailed, sorry for that...

PR #15 (2022-09-05): Fixes pylance linting for builtins
## What does this implement/fix? Explain your changes.
Fixes pylance linting for builtins with parts from official typings (github.com/python/typeshed).

## Does this close any currently open issues?
Should fix #14