paulober/MicroPico #16: Extension not working with VSCode 1.71.0 [Linux]

ma3oun, 2022-09-06:
## What are the steps to reproduce this issue?
1. Update to latest VSCode 1.71.0 on linux
2. Remove Pico-Go and install Pico-W-Go
3. Try to execute any command (e.g. Upload current file only)

## What happens?

An error window pops up: "command 'picowgo.uplodFile' not found"

## What were you expecting to happen?

File upload

## Any logs, error output, etc?
*(If it’s long, please paste to https://gist.github.com and insert the link here)*

## Support info

Copy this from the `Pico-W-Go > Help > Get support info` option in the command palette:

This command is not listed !

paulober, 2022-09-06:
You have to run 'Pico-W-Go > configure project first' as this will setup all the commands and linting for your workspace first. See README.md

ma3oun, 2022-09-06:
Indeed, this solves the problem. Thanks :-)

closed 2022-09-06 by ma3oun

SteveBannister1, 2022-09-07:
Thanks! That worked!