paulober/MicroPico #21: Cannot use REPL console.

Lak4CYUT, 2022-09-16:
## What are the steps to reproduce this issue?
1. Open VSCode.
2. Open a Pico W Go project.
3. Wait.

## What happens?
A dialog show the "Unable to start the terminal".

## What were you expecting to happen?
Show REPL console terminal.

## Any logs, error output, etc?
*(If it’s long, please paste to https://gist.github.com and insert the link here)*
Platform: Ubuntu
VSCode version: 1.70.2

## Any other comments?

## Support info

Copy this from the `Help -> Info -> Copy` option in Visual Studio Code:
Version: 1.70.2
Commit: e4503b30fc78200f846c62cf8091b76ff5547662
Date: 2022-08-16T05:36:25.715Z
Electron: 18.3.5
Chromium: 100.0.4896.160
Node.js: 16.13.2
V8: 10.0.139.17-electron.0
OS: Linux x64 5.4.0-96-generic
```
< Paste Here >
```

aaaaahaaaaa, 2022-09-16:
I experience the same issue now. The extension forces a terminal to be opened. Sometimes even 2.

apazdnikov, 2022-09-17:
Hello, experience the same issue.

migs35323, 2022-09-17:
probably related, in my case vscode opens like 4 terminals and i can't seem to close them, they respawn

paulober, 2022-09-18:
> ## What happens?
> A dialog show the "Unable to start the terminal".

Ok, can you please reproduce the error, open a new "normal" terminal > output panel > select something with "(Extensionhost)" at then end, and then copy the output concerning picowgo to something like a gist or directly into the issue?

paulober, 2022-09-18:
> probably related, in my case vscode opens like 4 terminals and i can't seem to close them, they respawn

I know, that's a known issue. I'm working on it as it affects all platforms.

paulober, 2022-09-18:
> probably related, in my case vscode opens like 4 terminals and i can't seem to close them, they respawn

There is a workaround. Instead of opening VS Code via a shortcut just run `code C:\path\to\project\folder` in terminal as this should not reload the terminals from your last session.

Lak4CYUT, 2022-09-19:
> probably related, in my case vscode opens like 4 terminals and i can't seem to close them, they respawn

Yes, I got the same problem.

Lak4CYUT, 2022-09-19:
> Ok, can you please reproduce the error, open a new "normal" terminal > output panel > select something with "(Extensionhost)" at then end, and then copy the output concerning picowgo to something [...]

Sure.

`[2022-09-19 09:25:09.036] [exthost] [info] Extension host with pid 757754 started
[2022-09-19 09:25:09.036] [exthost] [info] Skipping acquiring lock for /home/lake/.config/Code/User/workspaceStorage/d1ad1b1f274557b9c7d821c7e2615b9d.
[2022-09-19 09:25:09.962] [exthost] [info] ExtensionService#_doActivateExtension ms-python.python, startup: false, activationEvent: 'onLanguage:python'
[2022-09-19 09:25:11.278] [exthost] [info] ExtensionService#_doActivateExtension ms-toolsai.jupyter, startup: false, activationEvent: 'onLanguage:python'
[2022-09-19 09:25:11.646] [exthost] [info] ExtensionService#_doActivateExtension VisualStudioExptTeam.intellicode-api-usage-examples, startup: false, activationEvent: 'onLanguage:python'
[2022-09-19 09:25:11.966] [exthost] [info] ExtensionService#_doActivateExtension VisualStudioExptTeam.vscodeintellicode, startup: false, activationEvent: 'onLanguage:python'
[2022-09-19 09:25:12.189] [exthost] [info] ExtensionService#_doActivateExtension vscode.debug-auto-launch, startup: true, activationEvent: '*'
[2022-09-19 09:25:12.229] [exthost] [info] ExtensionService#_doActivateExtension vscode.git-base, startup: true, activationEvent: '*', root cause: vscode.git
[2022-09-19 09:25:12.287] [exthost] [info] ExtensionService#_doActivateExtension vscode.ipynb, startup: true, activationEvent: '*'
[2022-09-19 09:25:12.348] [exthost] [info] ExtensionService#_doActivateExtension cl.keil-assistant, startup: true, activationEvent: '*'
[2022-09-19 09:25:13.107] [exthost] [info] ExtensionService#_doActivateExtension lucasazzola.vscode-csproj, startup: true, activationEvent: '*'
[2022-09-19 09:25:14.214] [exthost] [info] ExtensionService#_doActivateExtension paulober.pico-w-go, startup: true, activationEvent: 'workspaceContains:.picowgo'
[2022-09-19 09:25:17.726] [exthost] [info] Not activating extension 'ms-dotnettools.csharp': Timed out while searching for 'workspaceContains' pattern **/*.{csproj,sln,slnf,csx,cake}
[2022-09-19 09:25:17.726] [exthost] [info] Not activating extension 'ms-vscode.cmake-tools': Timed out while searching for 'workspaceContains' pattern */CMakeLists.txt,*/*/CMakeLists.txt,*/*/*/CMakeLists.txt
[2022-09-19 09:25:17.726] [exthost] [info] Not activating extension 'ms-vscode.makefile-tools': Timed out while searching for 'workspaceContains' pattern **/makefile,**/Makefile,**/GNUmakefile
[2022-09-19 09:25:17.794] [exthost] [info] ExtensionService#_doActivateExtension vscode.microsoft-authentication, startup: false, activationEvent: 'onAuthenticationRequest:microsoft'
[2022-09-19 09:25:18.402] [exthost] [info] ExtensionService#_doActivateExtension vscode.git, startup: true, activationEvent: '*'
[2022-09-19 09:25:18.538] [exthost] [info] ExtensionService#_doActivateExtension vscode.github, startup: true, activationEvent: '*'
[2022-09-19 09:25:18.596] [exthost] [info] ExtensionService#_doActivateExtension ms-toolsai.jupyter-renderers, startup: false, activationEvent: 'api', root cause: ms-toolsai.jupyter
[2022-09-19 09:25:18.750] [exthost] [info] ExtensionService#_doActivateExtension ms-python.vscode-pylance, startup: false, activationEvent: 'onLanguage:python'
[2022-09-19 09:25:19.609] [exthost] [info] ExtensionService#_doActivateExtension vscode.emmet, startup: false, activationEvent: 'onStartupFinished'
[2022-09-19 09:25:19.772] [exthost] [info] ExtensionService#_doActivateExtension vscode.merge-conflict, startup: false, activationEvent: 'onStartupFinished'
[2022-09-19 09:25:19.903] [exthost] [info] ExtensionService#_doActivateExtension jemc.vscode-implicit-indent, startup: false, activationEvent: 'onStartupFinished'
[2022-09-19 09:25:20.794] [exthost] [info] ExtensionService#_doActivateExtension marus25.cortex-debug, startup: false, activationEvent: 'onStartupFinished'
[2022-09-19 09:25:21.007] [exthost] [info] ExtensionService#_doActivateExtension ms-vscode-remote.remote-containers, startup: false, activationEvent: 'onStartupFinished'
[2022-09-19 09:25:21.315] [exthost] [info] ExtensionService#_doActivateExtension ms-vscode-remote.remote-wsl, startup: false, activationEvent: 'onStartupFinished'
[2022-09-19 09:25:21.538] [exthost] [info] ExtensionService#_doActivateExtension psioniq.psi-header, startup: false, activationEvent: 'onStartupFinished'
[2022-09-19 09:25:28.433] [exthost] [info] ExtensionService#_doActivateExtension vscode.github-authentication, startup: false, activationEvent: 'onAuthenticationRequest:github'
[2022-09-19 09:25:28.763] [exthost] [info] ExtensionService#_doActivateExtension vscode.configuration-editing, startup: false, activationEvent: 'onLanguage:jsonc'
[2022-09-19 09:25:28.898] [exthost] [info] ExtensionService#_doActivateExtension vscode.json-language-features, startup: false, activationEvent: 'onLanguage:jsonc'
[2022-09-19 09:25:29.069] [exthost] [info] ExtensionService#_doActivateExtension vscode.typescript-language-features, startup: false, activationEvent: 'onLanguage:jsonc'
[2022-09-19 09:25:30.143] [exthost] [info] Eager extensions activated
`

paulober, 2022-09-24:
Should be fixed now with version v2.1.0.
Can you confirm?

aaaaahaaaaa, 2022-09-25:
I confirm it's working on my side.

Small feedback: I've never worked on a VSCode extension myself, but from UX point of view, it does feel like an oddity that it always forces a terminal to be opened. It would feel better if the `Pico Console` was available in the list of available terminals, and could be opened and closed freely.

paulober, 2022-09-25:
> Small feedback: I've never worked on a VSCode extension myself, but from UX point of view, it does feel like an oddity that it always forces a terminal to be opened. It would feel better if the `Pico Console` was available in the list of available terminals, and could be opened and closed freely.

As of now you can set "picowgo.openOnStart" to false in your porject settings, this would avoid opening the terminal on start. But i'll add this feature to the list.

Lak4CYUT, 2022-09-27:
@paulober 
> Should be fixed now with version v2.1.0. Can you confirm?
Confirmed, it is fixed on my Ubuntu. Thanks.

closed 2022-09-27 by Lak4CYUT