paulober/MicroPico #23: No stubs installed

bf194, 2022-09-20:
Installed a fresh VS Code (V 1.71.2) on windows 10, plus extensions Python, Pylance, Intellisense and Pico-W-Go.

Following Pico-W-Go getting started, opened a folder and (ctrl+shift+p) Pico-W-Go configure project. A .vscode folder is being installed, but is empty.
No stub files are created.
[image]

NowanIlfideme, 2022-09-21:
Got the same bug. It could be because I'm using conda/mamba environments locally. This is quite literally a new environment (installed VSCode + synced extensions on a new Windows install). Could be something Windows-specific too.

NowanIlfideme, 2022-09-21:
After some digging I found some way to manually install for now: https://github.com/paulober/Pico-W-Stub#using-the-stubs (same author as extension btw 😉)

paulober, 2022-09-24:
Ok, strange. Have a look i you have a `Pico-W-Stub` folder in `%APPDATA%\Code\User`. 

The Python environment shouldn't be a problem as long as it provides a "py", "python" or "python3" command to the $PATH. You can try to set the python interpreter executable via "picowgo.pythonPath" in you workspace settings.json or via VS Code settings GUI.

Please provide information requested by the bug report template to help getting this error solved as soon as possible.

quittung, 2022-09-24:
I have the same issue. When running "Configure Projects" the toolbar appears, no errors are shown and an empty .vscode folder is created. 

I tried Python 3.8, 3.9, 3.10 (from python.org). Now I am running 3.10 from the Microsoft Store. 
I rebooted twice, reloaded numerous times and disabled all extensions but Python, Pylance, Intellicode and Pico-W-Go.
I tried setting picowgo.pythonPath to "C:\Users\USER_NOSPACES\AppData\Local\Microsoft\WindowsApps\python3.10.exe".

The only relevant line in in Terminal > Output > Log (Extension Host) is:
[2022-09-24 22:17:02.823] [exthost] [info] ExtensionService#_doActivateExtension paulober.pico-w-go, startup: false, activationEvent: 'onCommand:picowgo.initialise'

Help -> About -> Copy
Version: 1.71.2 (user setup)
Commit: 74b1f979648cc44d385a2286793c226e611f59e7
Date: 2022-09-14T21:03:37.738Z
Electron: 19.0.12
Chromium: 102.0.5005.167
Node.js: 16.14.2
V8: 10.2.154.15-electron.0
OS: Windows_NT x64 10.0.19044
Sandboxed: No

paulober, 2022-09-24:
@quittung Thanks for the extensive debug Information. I think the problem could be related to the symbolic link creation permission. 

I'll publish a powershell script as a workaround for now...

paulober, 2022-09-25:
Here is a workaround for project setup. Place it in your porject folder and then run it via powershell. (On linux you maybe have to install powershell 7 first)
[Configure-Project.zip](https://github.com/paulober/Pico-W-Go/files/9640624/Configure-Project.zip)

quittung, 2022-09-25:
That was quick. I think we are getting closer, the files are there now. But importing ```machine``` still cannot be resolved. The json files seem corrupted, maybe an encoding issue:
[image]
[settings.zip](https://github.com/paulober/Pico-W-Go/files/9641014/settings.zip)

According to Notepad++ the encoding is ```UTF-16 LE BOM```. When I reencode it as ```UTF-8```, everything works.

paulober, 2022-09-26:
Thank's i missed that one... I'll update the script later.

paulober, 2022-09-27:
[Configure-Project.zip updated version](https://github.com/paulober/Pico-W-Go/files/9651807/Configure-Project.zip)

quittung, 2022-10-01:
Works for me, thanks for all the work!

paulober, 2022-10-11:
@quittung Glad to hear it works for you. I'm adding a description of this script to the known errors so others may have the same problem. Looking for a solution/fix.
@bf194 Since I have not heard anything more from you, I assume that the problem is also solved for you.

paulober, 2022-10-11:
Reopening for tracking

bf194, 2022-10-12:
Hello all, bf194 here.
I apologize for not showing up in the discussion; you made a very good job in fixing the issue. Thanks a lot for that. I am not a skilled software engineer (rather than experienced in the microprocessors that brought the apollo's to the moon) but an E-hardware engineer and not able to follow your technical details and jargon. I had to overcome diffidence before posting this issue on Github, but I am happy you took it seriously and fixed it.
It seems the issue is fixed. Everything works fine now in VS code (on my Linux). Even the pylance problem of not recognizing classes in the micropython module machine (ADC, Pin) seems to be resolved. A minor issue is that every time I start running my micropython (RUN button on Pico-W-Go), the very first import line in the code raises an import error. However you can simply ignore that. Pressing RUN again and it runs.
Thank you for the professional problem fixing
Kind Regards, bf194

jtayl0r, 2022-10-26:
For what it's worth... I had the same issue, but ran VSCode as an administrator and Pico-w-go > configure project properly installed the stubs in my project directory. I didn't need to run the script. Putting this out there as maybe it will help you debug, or help another user that doesn't want to run the separate script. Admin mode isn't ideal, but its something to work with.

paulober, 2022-10-26:
> @paulober For what it's worth... I had the same issue, but ran VSCode as an administrator and Pico-w-go > configure project properly installed the stubs in my project directory. I didn't need to run the script. Putting this out there as maybe it will help you debug, or help another user that doesn't want to run the separate script. Admin mode isn't ideal, but its something to work with.

Thanks for the feedback. This confirms my suspicion that there is a problem with the permissions for symlinks under some Windows installations. I'm just surprised because I don't have any problems with my Windows 11 installation.

jtayl0r, 2022-10-26:
@paulober I've got it figured out. Windows, by default, sets a GPO that limits who/what can create symlinks. As soon as I added my user to the GPO it worked perfectly. 

From GP editor... Computer Config > Windows Settings > Security Settings > User Rights Assignment > Create Symbolic Links 

In my testing, adding groups to this policy did not work. I set "everyone" "administrators" "local users" and a few others and it didn't work until I added the specific user. Tested again removing everything and adding only the user and it worked. gpudate /force didn't work for this policy and a reboot was required for the change to take effect. 

Hopefully you can just add that to the readme and not worry about messing with your code.

jtayl0r, 2022-10-27:
> @jtayl0r<https://nam12.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2Fjtayl0r&data=05%7C01%7C%7C855da1e5db8547b61b2208dab84752ac%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C638024913711369335%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000%7C%7C%7C&sdata=3z2uDwCZ15mJzmIxMDhztpYIjmAVPIxci19VF0Hcyu0%3D&reserved=0> Thanks for doing all this research. I don't like the idea that most Windows users would have to find a workaround listed in this repo for running basic features of this extension. So i think i found a fix for this issue, implemented in update v2.1.1. I appended a preview build of this version. I you want you can test it on your system, without your workaround enabled of course.
> pico-w-go-2.1.1.zip containing the vsix package for VS Code<https://nam12.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2Fpaulober%2FPico-W-Go%2Ffiles%2F9882137%2Fpico-w-go-2.1.1.zip&data=05%7C01%7C%7C855da1e5db8547b61b2208dab84752ac%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C638024913711369335%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000%7C%7C%7C&sdata=9d0Oabatk05ljMVmQVPe9Fcc2gbfuSBZcD9lhewjUeM%3D&reserved=0>

Sure thing! I’ll give it a go this evening, if not tomorrow.

Jonathan Taylor

jtayl0r, 2022-10-28:
> @jtayl0r Thanks for doing all this research. I don't like the idea that most Windows users would have to find a workaround listed in this repo for running basic features of this extension. So i [...]

@paulober I wouldn't say it is the most exhaustive test ever done, but whatever voodoo magic you did seemed to have resolved the issue. I tried it on two different Win11 machines after resetting the GPO and rebooting and it configured properly and works as advertised! Thanks for the fix!

paulober, 2022-10-28:
@jtayl0r Thanks for testing again. I'm glad to here that my fix is working. Will be included in upcoming update v2.1.1

closed 2022-10-28 by paulober

Image8080, 2023-04-24:
I have had *exactly* the same problem, works ok on mac, but windows is completely broken. I have tried everything including complete reinstall of vscode pyhton etc. Nada, totally borked. Unfortunately just do not have time to try figure this  out so will have to abandon pico-w-go for now but many thanks for your work

Image8080, 2023-04-24:
OK weird, now it just started working after reboot, there seems to be some sequence(?) issue on windows, but seriously I spent hours on this and still have no idea why it just started working. And I have no idea if this is just randomly going to happen again.

PR #31 (2022-10-28): v2.1.1 develop merge
## What does this implement/fix? Explain your changes.
Fixes #23, No stubs installed

## Does this close any currently open issues?
#23 

## Where has this been tested?

**Operating system:**
- Windows 11 22H2

**VSCode version:**
v1.72.2