paulober/MicroPico #24: No stubs installed in VS code

bf194, 2022-09-22:
Because the PICO-W-GO extension on VS Code does not work as promised (it does not create stubs), I tried a manual setup to get my stubs:

I followed the docs on https://micropy-cli.readthedocs.io/en/latest/base.html.

First, I installed micropy-cli via pip.
following the docs, creating a new project folder is simple:

Step 1: Excecuting micropy init project1 (in the bash terminal of VS code):
[image]

So far so good. It comes with 5 features (step 2). I selected them all. Execute a return to the next step:

Step 3: never comes. No selection of target device/firmware (I expected the RP2 stuff here). This is what comes:
[image]

Step 4: no Boom. No ready workspace! Still no stubs!

This command created a .micropy/ folder under $HOME, with an empty sybfolder /stubs. No symlinks, no micropy.json file and no project 'project1'

OK, no stubs! But a micropy.log file was created. After line 11 there is an error. What means the 1535 after the http code?
[image]

Not giving up yet. I tried the micropy stubs add <STUB_NAME>, but have not any idea what STUB_NAME to select. Again, after some browsing, with the help of google, I found something useful:

https://github.com/paulober/Pico-W-Stub/blob/main/micropy.md

I downloaded stubs for the RP2 from https://pypi.org/project/micropython-rp2-stubs/#files (the built distibution [micropython_rp2_stubs-1.19.1.post6-py3-none-any.whl](https://files.pythonhosted.org/packages/92/e6/d732b2a816eb0172f39d3dc9059ea5b0bf5c21796796ea8b393c20576f31/micropython_rp2_stubs-1.19.1.post6-py3-none-any.whl) and unpacked in some folder I made.

To add the stubs, one should execute the command: micropy stubs add "<path to dist/micropy-cli folder in this repo>". What does this mean? There is an example, but that is a windows one and I am in Linux.

Is it a site package in Python? I checked this and found something under pyhon 3.9 sitepackages, at least a folder named stubs under the micropy folder. But my active Python is 3.10.2 And that holds no site packages concerning stubs or micropy.

Strange! I give up! Just following instructions is not working.

This string of actions removed me far from my original goal: starting a project in a Pico-Pi, using VS Code 1.71.2

Back to the good old Thonny IDE

paulober, 2022-09-24:
I'm sorry to hear that you have problems installing the stubs. Why have you downloaded:
> stubs for the RP2 from https://pypi.org/project/micropython-rp2-stubs/#files
https://github.com/paulober/Pico-W-Stub/ is the repo containing the stub files i created to use in this extension. They should be automaticly installed in your VS Code User config dir on linux something like: "/home/bf194/.config/Code/User/Pico-W-Stub" and then you have to run `Pico-W-Go > Configure project` inside a VS Code workspace folder thought the VS Code command palett.

If you want to use micropy [https://github.com/paulober/Pico-W-Stub/blob/main/micropy.md](https://github.com/paulober/Pico-W-Stub/blob/main/micropy.md) is the right place for the docs. And as it says `micropy stubs add "<path to dist/micropy-cli folder in this repo>"` is the command to install the stubs into a repo you allready initilaized with micropy. For this you have clone the repo localy like into your Downloads folder or sth like that. Then run `micropy stubs add "here you put in the fully path to the folder dist/micropy-cli inside the repo you cloned localy"`. If you cloned my stubs repo in your downloads folder the command might look like: `micropy stubs add "/home/bf194/Downloads/Pico-W-Go/dist/micropy-cli".
I hope this fixes your problems.

paulober, 2022-09-24:
>Because the PICO-W-GO extension on VS Code does not work as promised (it does not create stubs), I tried a manual setup to get my stubs

Have you opened a folder thought VS Code and then pressed something like `Ctrl-Shift-P` and searched for `Pico-W-Go > Configure project`?
This should create a .vscode folder inside your project folder and link the stubs into it.

torpinouche, 2022-09-26:
I have the same problem. The command creates a .vscode but the folder is empty. This used to work with Pico-Go but cannot get it working properly with Pico-W-Go. Is there any manual way to do it? I have tried uninstall/reinstall everything including vscode and python

paulober, 2022-09-27:
@torpinouche Strange...
I also Tester the extension on a Windows 11 Pro system and it works fine...
You can found the workaround in the other issue created by the same author like this one (duplicate)

paulober, 2022-09-27:
https://github.com/paulober/Pico-W-Go/issues/23#issuecomment-1258996898

paulober, 2022-10-11:
Tracked in #23

closed 2022-10-11 by paulober