paulober/MicroPico #25: Upload of picozero library fails

mariomie, 2022-09-22:
## What are the steps to reproduce this issue?
1. Create a project including picozero library 
2. Upload project to pi pico

## What happens?
Upload fails for picozero.py 

## What were you expecting to happen?
Upload picozero.py

## Any logs, error output, etc?

```
>>>
Uploading project (main folder)...
Not safe booting, disabled in settings

Uploading to / ...
Reading file status
Creating dir lib
Creating dir lib/picozero
[1/3] Writing file 'Hello_World.py' (69 bytes)
[2/3] Writing file 'lib/picozero/__init__.py' (1 kB)
[3/3] Writing file 'lib/picozero/picozero.py' (65 kB)
Upload failed.Hashes do not match between computer and board.
Hashes do not match between computer and board.
Hashes do not match between computer and board.
Hashes do not match between computer and board.
Upload done, resetting board...

MicroPython v1.19.1 on 2022-06-18; Raspberry Pi Pico with RP2040
Type "help()" for more information.
>>>
MPY: soft reboot
MicroPython v1.19.1 on 2022-06-18; Raspberry Pi Pico with RP2040
Type "help()" for more information.
>>>
```

## Any other comments?
It is possible to upload the library with Thonny

## Support info

```
Version: 1.71.2
Commit: 74b1f979648cc44d385a2286793c226e611f59e7
Date: 2022-09-14T21:12:14.256Z
Electron: 19.0.12
Chromium: 102.0.5005.167
Node.js: 16.14.2
V8: 10.2.154.15-electron.0
OS: Linux x64 5.19.7-1-MANJARO
Sandboxed: No
```

paulober, 2022-09-24:
Try to run "All commands > Delete all files from board" first. And the re-run the "Upload project command to upload all your workspace files.

mariomie, 2022-09-24:
Unfortunately that is not working. It is a bigger file...Is this a problem? 
```
Searching for boards on serial devices...
Connecting to /dev/ttyACM0...

>>>  
Deleting '/Hello_World.py' ...
Deleting '/lib/picozero/__init__.py' ...
Deleting '/lib/picozero' ...
Deleting '/lib' ...

All files and directories have been deleted from the board.

 
>>> 
Uploading project (Pico)...

Not safe booting, disabled in settings
Uploading to / ...
Reading file status
Creating dir lib
Creating dir lib/picozero
[1/3] Writing file 'Hello_World.py' (69 bytes)
[2/3] Writing file 'lib/picozero/__init__.py' (1 kB)
[3/3] Writing file 'lib/picozero/picozero.py' (65 kB)
Upload failed.Hashes do not match between computer and board.
Hashes do not match between computer and board.
Hashes do not match between computer and board.
Hashes do not match between computer and board.
Upload done, resetting board...

MicroPython v1.19.1 on 2022-06-18; Raspberry Pi Pico with RP2040
Type "help()" for more information.
>>> 
MPY: soft reboot
MicroPython v1.19.1 on 2022-06-18; Raspberry Pi Pico with RP2040
Type "help()" for more information.
>>>
```

blakeglucas, 2022-11-30:
I took a look at this tonight, seems like something is off in the pyboard module, since the [MicroPython version](https://docs.micropython.org/en/latest/reference/pyboard.py.html#using-the-pyboard-library) handles the file upload fine. Might be worth it to just use a child process to manipulate their pyboard.py script instead of using a ported TS version?

paulober, 2023-04-11:
@mariomie V3 is there. @blakeglucas  I've taken your suggestion with the pyboard.py seriously and it was hard as sub-process communication is very error-prone, but here it is. Has this update solved your issue?

mariomie, 2023-04-14:
The upload is working with the latest version of Pico-W-Go. Thanks!

closed 2023-04-14 by mariomie