paulober/MicroPico #26: File explorer

aaaaahaaaaa, 2022-09-25:
# Feature Request 🚀

## Describe the solution you'd like

It would be great to have access to the content of the board as a file explorer from VSCode extension sidebar on the left. It would list the files currently on the Pico and files could also be draggred, dropped or even deleted directly from there.

paulober, 2022-09-25:
Good idea, I've been thinking about it as well. I need to find some details about the APIs that VS Code offers developers for this kind of function. This could be a good replacement for the FTP server integration so I can remove this dependency as it has a major security vulnerability.

paulober, 2022-11-09:
@aaaaahaaaaa I've implemented a Visual Studio Code Remote workspace integration which enables you to open the pico filesystem in vscode as a workspace folder and edit in it. The problem ist, that to get the filesystem information the extension has to run python commands in the repl which can cause issues when the user tries to run a command at the same time. A solution to this would be to block the terminal like when starting the ftp server but you'll still have the problem, that vscode will reload after mounting the remote filesystem which disconnects your pico and the reconnects it. This is realy anoying as you then have to manually reload the filesystem your self.

... in short: For the above reasons, the current options are not very user friendly, so I will either drop the Merge feature until vscode changes their API or I will add this as an experimental feature. What's your opinion on this? @aaaaahaaaaa & @Lak4CYUT

PabloPoche, 2022-12-18:
Se hace MUY necesario ver/listar los archivos cargador dentro de las raspberry pico W.

aaaaahaaaaa, 2023-04-11:
Nice! I'll check it out soon ❤️