paulober/MicroPico #29: When connected for keyboard shortcuts

JoeSc, 2022-10-19:
# Feature Request 🚀

## Is your feature request related to a problem? Please describe.
vscode allows me to map `picowgo.run` to `ctrl+enter`  However I have not been able to find a way to set the "When" for the shortcut to "When picowgo is connected" or something similar

## Describe the solution you'd like
Assistance with setting the "When" or perhaps picowgo has to export a variable that can be used in the "When" field

## Describe alternatives you've considered
I can set the shortcut to `ctrl+enter`, but then that breaks jupyter notebooks I may be working on

paulober, 2022-11-09:
Hi, thanks for the feedback and sorry for the delay. I'll look into it.

PabloPoche, 2022-12-18:
Cuando se esta auto-ejecutando un script(main.py) en la raspberry pico W, es muy dificil conectarce (Pico-W-Go > Connect) para deterner su ejecucion y el status de coneccion siempre dice:  Pico W Disconnected

paulober, 2023-07-15:
@JoeSc You can now check for the extension being active with `pico-w-go.isActivated` or for it being connected to a Pico with `pico-w-go.isConnected` in a when-clause context. 
Changes will be published with release v3.2.0.

commit 73938ede (2023-07-15): Fix #29, Add custom when clause context keys
commit 6feca9d6 (2023-07-15): Fix #29, Add custom when clause context keys
PR #116 (2023-07-15): Patch v3.2.0
### Added
- `vscode:uninstall` script to remove Pico-W-Stub after extension gets uninstalled
- `picowgo.rtc.sync` command to sync the Pico's RTC with the host's time
- Fix #29, Add custom when clause context keys

### Changed
- Auto-detected Python3 commands are't automatically saved to global settings anymore to prevent erros when switching between different systems with different python installations and settings sync turned on
- Contributed menu entries are now only shown when the extension is active
- Updated dependencies
closed 2023-07-15 by paulober