paulober/MicroPico #33: Micropython Libs

brettnak, 2022-11-21:
I've added a library to my pico via mip directly on the pico (umqtt.simple), but I'm not sure how to pull that into the project in VSCode.

I've done nothing beyond the standard pico-w-go setup, so obviously the extension knows nothing about it:

[image]

Is there currently a way to pull these into the project?

paulober, 2022-11-22:
The reason why VS Code knows something about modules like "machine" or "rp2" is because i've written custom python files to allow the Python VS Code extension know what is available. So at the moment you could download the umqtt module and copy the files in it into `<Your-workspace>/.vscode/Pico-W-Stub/stubs/umqtt`. I think this should work as a temporary solution.

paulober, 2023-01-15:
@brettnak Whats your status on that?

alexlukas, 2023-01-21:
Is it possible to add/create stubs from a uf2 file? I want to create a program with a Pimoroni pico display (see [github repo](https://github.com/pimoroni/pimoroni-pico), however the problem is that one module seems to be built from C++ files, and therefore there is no python file I can copy to the stubs. Do you have any idea on how to proceed here? Thanks!

paulober, 2023-01-25:
> Is it possible to add/create stubs from a uf2 file? I want to create a program with a Pimoroni pico display (see [github repo](https://github.com/pimoroni/pimoroni-pico), however the problem is that [...]

I think the only option is that you create a python file which contains all the raw functions or classes available in the native module and use this for the autocompletion like other python modules

derekgraham, 2023-04-25:
just wanted to add that I was also enjoying trying Pico-W-Go out until I ran into exact same issue as Alexlukas - wanting to use Pimoroni pico display. 

If I understood better what your response actually meant in practice I might attempt at some point. Conceptually it makes sense, but I just haven't done anything (yet) with python / C interaction, etc. that might inform me. Fascinating that every 'IDE' solution has to address this problem (new or libraries / modules / etc outside original scope). I imagine again and again... and I imagine there is no reason this can't be solved 'automatically' at some point and totally get how it may not be your top priority. Looking forward to checking back in at some point.

paulober, 2023-04-26:
@alexlukas @derekgraham If you user their modified MicroPython firmware you only need the stubs for your MicroPython dev no C library or anything like that as they are included on the Pico. You then have to run https://github.com/Josverl/micropython-stubber on your Pico like i do in [paulober/Pico-W-Stub](https:://github.com/paulober/Pico-W-Stub) and copy the files. Then rename to .pyi and you have some very basic stubs for the modules.

paulober, 2026-09-14:
Thanks! MicroPython type hints come from micropython-stubs, so this needs to be requested there: https://github.com/Josverl/micropython-stubs/issues

closed 2026-09-14 by paulober