paulober/MicroPico #36: Upload issues with a larger project

gilesknap, 2022-12-02:
## What are the steps to reproduce this issue?
1. clone git@github.com:gilesknap/pico-xmas.git
2. launch the folder in vscode
3. connect a pico
4. Command Pallette Pico-w-Go->Delete all files from board
5. right click Pico-w-Go->Upload project from file Explorer

## What happens?
The files start to upload but after one or two the process stalls 
sending a soft reset via the command pallette makes the upload move on to the next file
If this process is repeated a few times and the timing is right then the project does download OK.
If I don't do the soft reset then we see ``Upload failed.timeout``.

The output in the Terminal looks like this
```
>>> 
Uploading project (main folder)...

Not safe booting, disabled in settings
Uploading to / ...
Reading file status
Creating dir morse
Creating dir utils
[1/6] Writing file '01_start_morse.py' (271 bytes)
[2/6] Writing file '02_count.py' (1 kB)
[3/6] Writing file 'LICENSE' (11 kB)
[4/6] Writing file 'morse/morse_code.py' (1 kB)
```

## What were you expecting to happen?
All project files should upload with no additional intervention.
Thonny is able to upload this without issue.

## Any logs, error output, etc?
*(To get extension logs see Terminal > Output > Protcoll (Extensionhost) and copy output concerning picowgo)*
*(If it’s long, please paste to https://gist.github.com and insert the link here)*

## Any other comments?
Nice project !

## Support info

I'm using a pico H not a pico W - does that make a difference?

```
Version: 1.73.1
Commit: 6261075646f055b99068d3688932416f2346dd3b
Date: 2022-11-09T03:54:53.913Z
Electron: 19.0.17
Chromium: 102.0.5005.167
Node.js: 16.14.2
V8: 10.2.154.15-electron.0
OS: Linux x64 5.15.0-56-generic
Sandboxed: No
```

vifeng, 2022-12-08:
same error I have a pico H.
VSCODE : Version: 1.74.0
Python 3.11.1
all latest extensions
Mac os Monterey 12.6.1

magabriel, 2022-12-17:
This bug started to bite me hard just a few days ago, while the "Upload project" command started to fail at random and finally stopped working at all. This behavior seems related to either the number of files or its size because it was getting worse fast as my project grew in size.
So now I have to upload the whole project with Thonny and then upload the file I'm working on separately. Far from ideal, more so because the FTP server never worked for me either.

gavinbaker999, 2022-12-17:
My problem is that the file upload does not work at all. The error message
is hash size mismatch.

I am looking into shell and ampy but I don't know if they are fully windows
compatible or fit the use case I am looking at.

phasiclabs, 2023-01-21:
I'm having similar trouble on a Pico W with 24 files in the project. 

I have a magic ritual I have to follow to get my changes to upload to the board - whenever I make changes, I have to delete all files from the board, unplug it re-plug it in, upload 1 file only, unplug/replug, modify a file, then finally upload the project.

If I don't do this, it will either freeze mid upload, decide there is nothing to upload, or sometimes it removes existing files from the board.

It seems like there is some kind of caching going on that is not working - even with all files deleted from the board, it won't upload my project unless I make new modifications.

bmurphy1976, 2023-01-22:
@phasiclabs oof that's a lot.  Have you tried using [rshell](https://github.com/dhylands/rshell) instead?  I've found both Pico-W-Go and ampy to be very unreliable.  My current workflow is to us Pico-W-Go to help with auto completion but leave it disconnected.  I'm then using rshell on the upload, sync, and run files.

gavinbaker999, 2023-01-22:
I seem to have found a solution - the pymakr vscode extension.  This allows
connection to the pico board. Also can display in terminal and files as a
directory in the project file window. You can run files on the board and
upload them to the board by options in the right click on the local file
menu.

You can also drag and drop files between the local and remote.

A CLI solution I found is ampy which again connected to the pico, run files
on the pico and upload/download files.

I also tried rshell, seemed to install fine but crashed on running.

phasiclabs, 2023-01-22:
@bmurphy1976 it's only just started happening to me now I'm working on a larger project, so I've not looked for other workarounds yet. 
I'm starting to think modifications to main.py make things worse - the project upload fails 100% if that is modified - possibly it's locked by the board? It still fails very often with other files though. 
I'll check out rshell though - thanks for the tip.

paulober, 2023-01-30:
@gilesknap I tryed reporducing your issue by uploading the repo you mentioned to my pico (with PicoWGo). But its all working fine, i got no errors, nothing. All files get uploaded like they should.

Can you confirm this issue still exists for you/

paulober, 2023-01-30:
@phasiclabs Do you have a sample project for me to reproduce this error?

gilesknap, 2023-01-31:
@paulober I'll give this another go this evening.

paulober, 2023-02-03:
> @paulober I'll give this another go this evening.

Can I mark this as resolved?

gilesknap, 2023-02-04:
Sorry for the delay.

I tried this again and am still seeing the issue. I note that if I do a 'delete all files on the board' followed by upload it always hangs on the same file as shown below.

```

>>> 
Uploading project (main folder)...

Not safe booting, disabled in settings
Uploading to / ...
Reading file status
Creating dir hardware
Creating dir morse
Creating dir music
Creating dir utils
[1/16] Writing file 'LICENSE' (11 kB)
[2/16] Writing file 'd01_start_morse.py' (292 bytes)
[3/16] Writing file 'd02_count.py' (498 bytes)
[4/16] Writing file 'd03_buttons.py' (497 bytes)
[5/16] Writing file 'd04_controller.py' (3 kB)
[6/16] Writing file 'd05_music.py' (143 bytes)
[7/16] Writing file 'd06_photo_t.py' (1 kB)
[8/16] Writing file 'd07_pir.py' (1 kB)
[9/16] Writing file 'hardware/advent.py' (1 kB)
[10/16] Writing file 'morse/morse_code.py' (1 kB)
```

The project that exhibits this issue is here https://github.com/gilesknap/pico-xmas
I'm on ubuntu 22.04

paulober, 2023-04-11:
@gilesknap V3 is here, completely rewritten from ground up, so i wanted to ask you if this issue is fixed by the new release?

gilesknap, 2023-04-12:
Great. I'll try it out this evening (BST).

This e-mail and any attachments may contain confidential, copyright and or privileged material, and are for the use of the intended addressee only. If you are not the intended addressee or an authorised recipient of the addressee please notify us of receipt by returning the e-mail and do not use, copy, retain, distribute or disclose the information in or attached to the e-mail.
Any opinions expressed within this e-mail are those of the individual and not necessarily of Diamond Light Source Ltd. 
Diamond Light Source Ltd. cannot guarantee that this e-mail or any attachments are free from viruses and we cannot accept liability for any damage which you may sustain as a result of software viruses which may be transmitted in or with the message.
Diamond Light Source Limited (company no. 4375679). Registered in England and Wales with its registered office at Diamond House, Harwell Science and Innovation Campus, Didcot, Oxfordshire, OX11 0DE, United Kingdom

gilesknap, 2023-04-13:
OK this looks good. I had no issues with project upload with the same set of files. Thanks for the fix!

A couple of minor things I noticed:

- I was a little bemused that the REPL did not appear by default and I had to click on the + in the terminal. Maybe this is #60?
- Also there is now no logging when uploading files / projects, just a pop up notification while it is happening. That initially threw me as I did not believe that it was working. I'm pretty OK with this is if project upload is rock solid - so just an observation.

paulober, 2023-04-13:
@gilesknap I'm glad to here that the issue is resolved for you by the new version. 
As for the other things you mentioned:
- If you want the vREPl auto open you can set the `picowgo.openOnStart` settings like documented in the [README.md](https://github.com/paulober/Pico-W-Go#extension-settings)
- I also thought about printing the upload status of each file into the vREPL, as for now i decided to use the integrated progress bar notification type to not remove stuff already typed into the vREPL by the user but not submitted. But maybe i'll make the progress notification have a growing progress indicator and if something fails print out the file and progress where it stops into the vREPL.

gilesknap, 2023-04-14:
Thanks Paul.

> * But maybe i'll make the progress notification have a growing progress indicator and if something fails print out the file and progress where it stops into the vREPL.

That would be ideal