paulober/MicroPico #39: Hangs at "Connecting to /dev/tty.usbmodem"

dComposer, 2022-12-27:
## What are the steps to reproduce this issue?
1. Trying to connect to my pico using pico-w-go (on a Mac M1 Mini)

## What happens?
Terminal hangs at "Connecting to /dev/tty.usbmodem213201..."

## What were you expecting to happen?
I expect it to connect to my Pico

## Any logs, error output, etc?
* No error output, just hangs at trying to connect to my pico

## Any other comments?
I was able to use pico-w-go to discover my serial ports

## Support info

Copy this from the `Help -> Info/About -> Copy` option in Visual Studio Code:

```
Version: 1.74.2 (Universal)
Commit: e8a3071ea4344d9d48ef8a4df2c097372b0c5161
Date: 2022-12-20T10:26:09.430Z
Electron: 19.1.8
Chromium: 102.0.5005.167
Node.js: 16.14.2
V8: 10.2.154.15-electron.0
OS: Darwin arm64 22.1.0
Sandboxed: No
```

paulober, 2023-01-15:
Which version of Pico-W-Go and Python have you installed on you system?

paulober, 2023-01-15:
This could be due to permission erros. Some unix like operating systems do have very restricted permission on device files like the one for the serial connection to your pico. Try connection to you Pico with a diffrent serial console like minicom without sudo. If this does not that could be an indicator that it's the permission thing. Here is a description on how to solve this problem on a linux machine: [Pico-W-Go -> Wiki -> Linux](https://github.com/paulober/Pico-W-Go/wiki/Linux) mybe that helps you finding a way to get access on your machine. Unfortunately I don't have access to a macOS bases machine so i cannot give you more advice on the system part.

dburles, 2023-01-17:
I have this same issue (MacOS), it is indicating that it's connected in the status bar, however the console hangs at this message on starting vscode. I would expect it should stop the running program and land on the REPL, though this never happens.

paulober, 2023-04-11:
@dComposer @dburles Does the new version 3 of the extension resolve your issue?

dburles, 2023-04-13:
@paulober It does. Though sometimes the REPL occasionally hangs upon opening and causing the other commands (connect/disconnect, run, reset etc) to become unresponsive. Can't seem to find a reliable way to reproduce it.
[image]

paulober, 2023-06-14:
Fixed with v3+

closed 2023-06-14 by paulober