paulober/MicroPico #40: Config "syncFolder" - allow for contents only upload (no parent path)

plesatejvlk, 2022-12-29:
# Feature Request 🚀

As I manage multiple pico-w projects in the same repository I would appreciate a config setting which would allow me to:
- set the syncFolder per directory (already supported)
- allow for "contents only" sync on the "Upload Project" action, for example:
```
repository-name
-- project-one
---- .picowgo (syncFolder="project-one")
---- main.py
-- project-two
---- .picowgo (syncFolder="project-two")
---- main.py
```
I would expect that once running the "Upload Project" action from within the `project-one` directory the `project-one/main.py` would get uploaded as `/main.py` to the pico (treating the `syncFolder` setting as a project target root).

Thanks for considering this!

paulober, 2023-04-11:
Sorry for having overlooked this issue for so long. Now that V3 of the extension is here. I'll take your request on the TODO list. I'll keep you updated.

ffunenga, 2023-06-18:
Is there a work around available until this issue is resolved? For now, the only way I am seeing to overcome this problem is to *only* use the Pico-W-FS and write my code directly there.

paulober, 2023-06-18:
@ffunenga I'm working on adding support for multi root projects but for now you can either open only the directory with the project you want to upload or set the syncRoot in the  workspace `settings.json` to the project folder you are currently working on.

commit ac646922 (2023-06-19): Added additionalSyncFolders 

- Fixes #40, Config additional sync folders per project
closed 2023-06-20 by paulober