paulober/MicroPico #41: Configure project fails on VSCode 1.74.2 on MacOS 12.6

echelon2020, 2023-01-05:
## What are the steps to reproduce this issue?
Updated Python to 3.10.4 (via brew)
Installed dependencies pylons and microphone-cli via pip3
Installed Pico-W-Go ver 2.1.4 in vscode 1.74.2
created and opened new directory
tried Configure Project

## What happens?
Command 'Pico-Go > Configure project' resulted in an error (command 'picogo.initialise' not found)

In fact any command fails with a similar error message.....

## What were you expecting to happen?

## Any logs, error output, etc?
*(To get extension logs see Terminal > Output > Protcoll (Extensionhost) and copy output concerning picowgo)*
*(If it’s long, please paste to https://gist.github.com and insert the link here)*

## Any other comments?

## Support info

Copy this from the `Help -> Info/About -> Copy` option in Visual Studio Code:

```
Version: 1.74.2
Commit: e8a3071ea4344d9d48ef8a4df2c097372b0c5161
Date: 2022-12-20T10:26:09.430Z (2 wks ago)
Electron: 19.1.8
Chromium: 102.0.5005.167
Node.js: 16.14.2
V8: 10.2.154.15-electron.0
OS: Darwin x64 21.6.0
Sandboxed: No
```

echelon2020, 2023-01-05:
log is

/Users/uwehagmaier/.vscode/extensions/chriswood.pico-go-1.4.3/node_modules/@serialport/bindings/lib/binding/node-v106-darwin-x64/bindings.node
	at bindings (/Users/uwehagmaier/.vscode/extensions/chriswood.pico-go-1.4.3/node_modules/bindings/bindings.js:126:9)
	at Object.<anonymous> (/Users/uwehagmaier/.vscode/extensions/chriswood.pico-go-1.4.3/node_modules/@serialport/bindings/lib/darwin.js:2:36)
	at u._compile (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/loader.js:4:1313)
	at Module._extensions..js (node:internal/modules/cjs/loader:1220:10)
	at Module.load (node:internal/modules/cjs/loader:1035:32)
	at Module._load (node:internal/modules/cjs/loader:876:12)
	at c._load (node:electron/js2c/asar_bundle:5:13343)
	at r._load (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/api/node/extensionHostProcess.js:120:13379)
	at w._load (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/api/node/extensionHostProcess.js:115:61900)
	at v._load (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/api/node/extensionHostProcess.js:115:61293)
	at Module.require (node:internal/modules/cjs/loader:1059:19)
	at p (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/loader.js:4:699)
	at Object.<anonymous> (/Users/uwehagmaier/.vscode/extensions/chriswood.pico-go-1.4.3/node_modules/@serialport/bindings/lib/index.js:10:22)
	at u._compile (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/loader.js:4:1313)
	at Module._extensions..js (node:internal/modules/cjs/loader:1220:10)
	at Module.load (node:internal/modules/cjs/loader:1035:32)
	at Module._load (node:internal/modules/cjs/loader:876:12)
	at c._load (node:electron/js2c/asar_bundle:5:13343)
	at r._load (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/api/node/extensionHostProcess.js:120:13379)
	at w._load (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/api/node/extensionHostProcess.js:115:61900)
	at v._load (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/api/node/extensionHostProcess.js:115:61293)
	at Module.require (node:internal/modules/cjs/loader:1059:19)
	at p (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/loader.js:4:699)
	at Object.<anonymous> (/Users/uwehagmaier/.vscode/extensions/chriswood.pico-go-1.4.3/node_modules/serialport/lib/index.js:2:17)
	at u._compile (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/loader.js:4:1313)
	at Module._extensions..js (node:internal/modules/cjs/loader:1220:10)
	at Module.load (node:internal/modules/cjs/loader:1035:32)
	at Module._load (node:internal/modules/cjs/loader:876:12)
	at c._load (node:electron/js2c/asar_bundle:5:13343)
	at r._load (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/api/node/extensionHostProcess.js:120:13379)
	at w._load (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/api/node/extensionHostProcess.js:115:61900)
	at v._load (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/api/node/extensionHostProcess.js:115:61293)
	at Module.require (node:internal/modules/cjs/loader:1059:19)
	at p (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/loader.js:4:699)
	at Object.<anonymous> (/Users/uwehagmaier/.vscode/extensions/chriswood.pico-go-1.4.3/lib/connections/pyserial.js:20:19)
	at u._compile (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/loader.js:4:1313)
	at Module._extensions..js (node:internal/modules/cjs/loader:1220:10)
	at Module.load (node:internal/modules/cjs/loader:1035:32)
	at Module._load (node:internal/modules/cjs/loader:876:12)
	at c._load (node:electron/js2c/asar_bundle:5:13343)
	at r._load (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/api/node/extensionHostProcess.js:120:13379)
	at w._load (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/api/node/extensionHostProcess.js:115:61900)
	at v._load (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/api/node/extensionHostProcess.js:115:61293)
	at Module.require (node:internal/modules/cjs/loader:1059:19)
	at p (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/loader.js:4:699)
	at Object.<anonymous> (/Users/uwehagmaier/.vscode/extensions/chriswood.pico-go-1.4.3/lib/pymakr.js:18:17)
	at u._compile (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/loader.js:4:1313)
	at Module._extensions..js (node:internal/modules/cjs/loader:1220:10)
	at Module.load (node:internal/modules/cjs/loader:1035:32)
	at Module._load (node:internal/modules/cjs/loader:876:12)
	at c._load (node:electron/js2c/asar_bundle:5:13343)
	at r._load (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/api/node/extensionHostProcess.js:120:13379)
	at w._load (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/api/node/extensionHostProcess.js:115:61900)
	at v._load (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/api/node/extensionHostProcess.js:115:61293)
	at Module.require (node:internal/modules/cjs/loader:1059:19)
	at p (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/loader.js:4:699)
	at Object.<anonymous> (/Users/uwehagmaier/.vscode/extensions/chriswood.pico-go-1.4.3/lib/activator.js:28:15)
[... cut ...]
	at v._load (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/api/node/extensionHostProcess.js:115:61293)
	at Module.require (node:internal/modules/cjs/loader:1059:19)
	at require (node:internal/modules/cjs/helpers:102:18)
	at Function.r [as __$__nodeRequire] (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/loader.js:5:101)
	at d.pb (/Applications/Visual Studio Code.app/Contents/Resources/app/out/vs/workbench/api/node/extensionHostProcess.js:115:63164)
	at process.processTicksAndRejections (node:internal/process/task_queues:96:5)
	at async Promise.all (index 0)

paulober, 2023-01-10:
You are running the old extension `chriswood.pico-go-1.4.3` which does no longer even exists in VS Code store. The command for the new extension if you have installed like is not "Pico-Go > Configure project" it's "Pico-W-Go > Configure project".
But i would probably be better if you first uninstall "Pico-Go" before using "Pico-W-Go".

echelon2020, 2023-01-10:
So simple. Thank you very much.

closed 2023-01-10 by echelon2020