paulober/MicroPico #42: Pico-Go-W > Configure Project Fails

freesurfer-rge, 2023-01-16:
## What are the steps to reproduce this issue?
1. Installed latest version of VS Code plugin
2. Created empty directory
3. Ran `code .`
4. Tried to run Pico-G-W > Configure Project

## What happens?

Command failed, with message 'picowgo.initialise not found'

## What were you expecting to happen?

To be able to complete the 'getting started' part of the ReadMe

## Any logs, error output, etc?

From the VS Code terminal log, the relevant portion appears to be:

```
2023-01-16 14:34:33.621 [info] ExtensionService#_doActivateExtension paulober.pico-w-go, startup: false, activationEvent: 'onCommand:picowgo.initialise'
2023-01-16 14:34:33.909 [error] Activating extension paulober.pico-w-go failed due to an error:
2023-01-16 14:34:33.909 [error] Error: No native build was found for platform=linux arch=arm runtime=electron abi=106 uv=1 armv=default libc=glibc node=16.14.2 electron=19.1.8
    loaded from: /home/pi/.vscode/extensions/paulober.pico-w-go-2.1.5-linux-armhf/node_modules/@serialport/bindings-cpp

	at load.path (/home/pi/.vscode/extensions/paulober.pico-w-go-2.1.5-linux-armhf/node_modules/node-gyp-build/index.js:60:9)
	at load (/home/pi/.vscode/extensions/paulober.pico-w-go-2.1.5-linux-armhf/node_modules/node-gyp-build/index.js:22:30)
	at Object.<anonymous> (/home/pi/.vscode/extensions/paulober.pico-w-go-2.1.5-linux-armhf/node_modules/@serialport/bindings-cpp/dist/load-bindings.js:10:46)
	at u._compile (/usr/share/code/resources/app/out/vs/loader.js:4:1313)
```

This is on a Raspberry Pi 4b, which had completed runinng `apt update ; apt upgrade` about half an hour previously.

## Any other comments?

## Support info

Copy this from the `Help -> Info/About -> Copy` option in Visual Studio Code:

```
Version: 1.74.3
Commit: 97dec172d3256f8ca4bfb2143f3f76b503ca0534
Date: 2023-01-09T16:53:02.931Z
Electron: 19.1.8
Chromium: 102.0.5005.167
Node.js: 16.14.2
V8: 10.2.154.15-electron.0
OS: Linux arm 5.15.84-v7l+
Sandboxed: No
```

paulober, 2023-04-11:
@freesurfer-rge  Hi, sorry for having overlooked this issue. I was working hard on the newly released V3 of the extension. I hope this new update solves your issue as it uses python pyserial under the hood which should work great on the Raspberry Pi. The old system was affected by a bug in node-gyp not correctly detecting the arm version and therefore even if the extension is capable of armhf not loading it.

So in short. Has the new update solved your issue?

freesurfer-rge, 2023-04-16:
I have updated Pico-W-Go from within VS Code, although that is saying it's only version 2.1.8.

Having said that, I can now run the 'Pico-W-Go'->'Configure Project' command successfully.

Getting `flash.py` working is another matter. When I connect the Pico, am I supposed to hold down the 'BOOTSEL' button on the Pico or not? If I do, my Pi4 can successfully mount it as a drive. If I don't, then I can tell that the Pico is running the Python script I'd previously put on it with Thonny. However, either way, I'm not seeing anything new under `/dev/tty*`

Is there a longer 'Getting Started' guide which I've overlooked? The one on the GitHub landing page feels like it's missing a step or two.

paulober, 2023-04-18:
No, don't hold BOOTSEL down when connecting to your Pi4 as this would put the Pico in USB mode and it will be connected as a drive. If your `pi` user on your Pi4 has the sufficient rights (which if i remember correctly should be the case on Raspbian) you can now (with the new version of the extension) open the configured project folder in VSCode and it should automatically try to connect to you Pico. You see the status of the connection by the status bar and notifications. Hope this works for you?

closed 2023-04-18 by paulober