paulober/MicroPico #44: Run multifile OOP main.py from terminal

shanks13, 2023-01-21:
# Feature Request 🚀

When creating multiple files (one per class) and attempting to run main.py from terminal, the following error occurs
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: no module named 'led'

In the test only one class in a file led.py was created and line 1 of main.py was 
import led

Have tried running the same code on Thonny and works fine via the terminal. An __init__.py was created in the project folder too.

Right now the only way to execute a multi-file OOP in micropython is by uploading the project files to the pico in order to test each change.

shanks13, 2023-01-21:
I am running rp2-pico-20220618-v1.19.1.uf2 micropython the Pico

shanks13, 2023-01-25:
Update - it actually started working - sporadic, but it does run main.py from the terminal with repeated tries and imports the other  class files.

paulober, 2023-01-25:
I'm working on a new implmenetation of the part of this extension which handles the communication between the pico and vscode. This should also fix this issue...

shanks13, 2023-01-25:
That's fantastic - you're the best. This is a great extension BTW - thank you !

KernelBypass, 2023-07-03:
I am having this issue as well - everything works on Thonny, but in VS code running code with multiple files is a no go -- which limits me to using monolithic single file source code.

Is there an ETA for the fix?

Thank you

shanks13, 2023-07-03:
It's been working well for me now. Make sure there is a __init__.py file uploaded to the pico. Then running main.py through the pico-vREPL terminal works great with the multi-file OOP projects. Print statements back to the terminal also working just fine.