paulober/MicroPico #45: Upload Project is not uploading any files

rodrigobomfim, 2023-01-29:
## What are the steps to reproduce this issue?
1. Set "src" as the sync folder
2. Command: Delete all files from the device
3. Command: Upload Project

## What happens?

No files are uploaded.

> Uploading project (src)...
> Not safe booting, disabled in settings
> 
> Uploading to / ...
> Reading file status
> No files to upload
> Upload done

## What were you expecting to happen?

All files under "src" folder are uploaded. This used to work but has stopped working now. 

## Any logs, error output, etc?
*(To get extension logs see Terminal > Output > Protcoll (Extensionhost) and copy output concerning picowgo)*
*(If it’s long, please paste to https://gist.github.com and insert the link here)*

## Any other comments?

## Support info

Copy this from the `Help -> Info/About -> Copy` option in Visual Studio Code:

```
< Paste Here >
```

paulober, 2023-02-03:
Have you tried deleting the `project.pico-w-go` file from the board by running following commands after running the "Delete all files from board" feature? Then reopen vscode and reconnect the board before uploading again.
```python3
import os; os.remove("project.pico-w-go")
```
It seems that this file has been corrupted on your Pico.

calston, 2023-03-23:
This appears to happen on every upload though

mattb1969, 2023-03-26:
Hi,
I am suffering the same issue, having tried the suggested fix, it is still not uploading any files.

I have been trying to figure this out and I wonder if it is because I am not using default folders on my Mac, it is not picking up any files. 
The error has come about as I have written some libraries for my project and they are not being transferred to the Pico board.

If I try to just upload and run a single file, it is fine, but as soon as there are multiple files, it does not work.

Hanqaqa, 2023-04-05:
Have you guys tried saving the main.py file before uploading?

I mean not just writing the code in VS Code and hitting `Upload project`. I mean Ctrl+S or File -> Save in your main.py file VS Code before hitting  `Upload project`

I got a Pico yesterday and started using this extension right away. Everything was working perfectly: writing python instructions on the REPL, running the main.py file, but as soon as a I uploaded something. Nothing was happening.  This was driving me nuts until I opened the main.py file with Notepad++ and saw that the file was empty. VS Code keeps a cache of the data you are writing, but it doesn't actually save the file.

paulober, 2023-04-11:
@rodrigobomfim, @mattb1969, @calston, V3 of the extension now uses the official pyboard tool from the MicroPython project under the hood, does this solve your issue?

mattb1969, 2023-04-12:
Hi @paulober, I am still at a complete loss as to why my project is not working.

I have noticed that in V3, there is no longer dialogue identifying which files are being uploaded and I don’t appear to be able to find a separate list of them as Output is empty.

paulober, 2023-04-13:
@mattb1969 Have you check if you set the `picowgo.syncFolder` and `picowgo.syncFileTypes` or `picowgo.syncAllFileTypes` (maybe also `picowgo.pyIgnore`) settings correctly?

closed 2023-05-24 by paulober