paulober/MicroPico #46: Running file deletes all other files from the board

radimd, 2023-02-01:
## What are the steps to reproduce this issue?
1. Upload project (pico-w-go > Upload project)
2. Open any .py file
3. Click Run or right click in project explorer and select pico-w-go > Run current file

## What happens?
File runs, but all other files and folders are deleted from the board

## What were you expecting to happen?
File runs and the rest of the files and folders stays intact.

## Any logs, error output, etc?
*(To get extension logs see Terminal > Output > Protcoll (Extensionhost) and copy output concerning picowgo)*
*(If it’s long, please paste to https://gist.github.com and insert the link here)*

[image]

## Any other comments?

## Support info

Copy this from the `Help -> Info/About -> Copy` option in Visual Studio Code:

```
< Paste Here >
```

busslina, 2023-02-09:
I have same issue. For me only works if everything is in one file. If I had any import (on same folder) it crashes. Ampy workis fine. I only use this extension (for the moment) to enjoy the stubs..

alaub81, 2023-02-09:
I have the same problem. But for me everything is working when I flash my pico w with the old MicroPython Version:

```rp2-pico-w-20221201-unstable-v1.19.1-724-gfb7d21153.uf2```

So perhaps the problem is caused by newer MicroPython versions

busslina, 2023-02-09:
> 

On readme it is said:

> Auto-completion based on Raspberry Pi Pico W MicroPython firmware: [rp2-pico-w-20230113-unstable-v1.19.1-796-gf4811b0b4.uf2](https://micropython.org/resources/firmware/rp2-pico-w-20230113-unstable-v1.19.1-796-gf4811b0b4.uf2)

Maybe we have to wait this libs get adapted to last version

alaub81, 2023-02-09:
Nope... just tested it with that old Version. Same Problem here.

theobvious89, 2023-02-12:
> I have the same problem. But for me everything is working when I flash my pico w with the old MicroPython Version:
> 
> `rp2-pico-w-20221201-unstable-v1.19.1-724-gfb7d21153.uf2`
> 
> So perhaps the problem is caused by newer MicroPython versions

I'm having the same issue. I'd like to install the old firmware, but I'm unable to find it. Do you have a link for this version that worked?

alaub81, 2023-02-12:
> I'm having the same issue. I'd like to install the old firmware, but I'm unable to find it. Do you have a link for this version that worked?

No Sorry, I have the binary just in my download folder. But here is the upload:

* [rp2-pico-w-20221201-unstable-v1.19.1-724-gfb7d21153.uf2.zip](https://github.com/paulober/Pico-W-Go/files/10716631/rp2-pico-w-20221201-unstable-v1.19.1-724-gfb7d21153.uf2.zip)

theobvious89, 2023-02-13:
> No Sorry, I have the binary just in my download folder. But here is the upload:
>
> * [...]

Thanks a lot! That makes the development process a lot simpler for me.

closed 2023-03-02 by paulober

paulober, 2023-03-02:
Can you confirm that this [experimental build pico-w-go-2.1.7.zip](https://github.com/paulober/Pico-W-Go/files/10875581/pico-w-go-2.1.7.zip) fixes this issue for you even when running the latest micropython firmware?

theobvious89, 2023-03-03:
> Can you confirm that this [experimental build pico-w-go-2.1.7.zip](https://github.com/paulober/Pico-W-Go/files/10875581/pico-w-go-2.1.7.zip) fixes this issue for you even when running the latest micropython firmware?

Yes, I can confirm that build fixes the issue with the latest micropython firmware.

alaub81, 2023-03-03:
> Can you confirm that this [experimental build pico-w-go-2.1.7.zip](https://github.com/paulober/Pico-W-Go/files/10875581/pico-w-go-2.1.7.zip) fixes this issue for you even when running the latest micropython firmware?

Tested it, too! It is working for me!
Thank you @paulober

paulober, 2023-03-03:
> > Can you confirm that this [experimental build pico-w-go-2.1.7.zip](https://github.com/paulober/Pico-W-Go/files/10875581/pico-w-go-2.1.7.zip) fixes this issue for you even when running the latest micropython firmware?
> 
> Tested it, too! It is working for me! Thank you @paulober

Glad to here that it's working for you
but credits go to @connorcl

commit d84537bf (2023-03-02): Merge pull request #49 from connorcl/bug/fix-fs-erase-on-run

"Wait for Ctrl-D to finish to prevent filesystem reformat upon following Ctrl-C"

Fixes #46, #50