paulober/MicroPico #50: Uploading Project results in removing all files (?)

ningelsohn, 2023-03-01:
Fist and foremost: I am not sure if this is a problem with this plugin, or if my pico is in some weird state because I had to flash-nuke it.

## What are the steps to reproduce this issue?
1. Unsure, but since the flash-nuke  it happens here and there
2. After a hard-reset the upload works normal

## What happens?

Instead of the normal Upload info, the following is shown:

```
Uploading project (main folder)...
Not safe booting, disabled in settings
Uploading to / ...
Reading file status
Deleting 27 files and folders
Removing file hcsr04.py
Removing file controller.py
...
Removing file app.py
Upload done, resetting board...
OK
```

## What were you expecting to happen?

Normally, after uploading the project, I get a progress screen for which files have already been uploaded and afterwards the program starts automatically, neither of that happens

## Support info

Copy this from the `Help -> Info/About -> Copy` option in Visual Studio Code:

```
Version: 1.75.1 (Universal)
Commit: 441438abd1ac652551dbe4d408dfcec8a499b8bf
Date: 2023-02-08T21:34:59.000Z
Electron: 19.1.9
Chromium: 102.0.5005.194
Node.js: 16.14.2
V8: 10.2.154.23-electron.0
OS: Darwin arm64 22.2.0
Sandboxed: No
```

paulober, 2023-03-02:
Thanks for reporting this bug. I'll include a fix for this in a later update

paulober, 2023-03-02:
Should be fixed by #49. Can you confirm with [experimental build pico-w-go-2.1.7.zip](https://github.com/paulober/Pico-W-Go/files/10875581/pico-w-go-2.1.7.zip)

ningelsohn, 2023-03-03:
Hi! Thanks for the lightning fast fix! For now it seems like everything is working, but somehow I couldn't reproduce the error with the normal version available in VSCode. Were you able to reproduce it?

paulober, 2023-03-03:
Actually no. But i'm planing some big changes to the communication component so then this should be fixed at the least

closed 2023-03-03 by paulober