paulober/MicroPico #53: Failed to connect to com port

kfaubel, 2023-03-19:
## What are the steps to reproduce this issue?
1. Plug in my Pico W
2. Launch VSCode with Pico-W-Go installed and enabled
3. 

## What happens?
Terminal shows:
(AutoConnect enabled, ignoring 'manual com port' address setting)
Searching for boards on serial devices...
Connecting to COM4...

If I close VSCode and open Thonny, it automatically opens COM4 and gives me the REPL prompt.

Device Manager shows: USB Serial Device (COM4)

I rebooted to make sure nothing else was using COM4 but it did not help.

## What were you expecting to happen?
I was expecting COM4 to be opened.

I saw an earlier post about disabling Auto Connect and manually specifying a COM port but that did not help so I reverted to auto connect.

## Any logs, error output, etc?
*(To get extension logs see Terminal > Output > Protcoll (Extensionhost) and copy output concerning picowgo)*
*(If it’s long, please paste to https://gist.github.com and insert the link here)*
2023-03-19 07:49:14.179 [info] Extension host with pid 15076 started
2023-03-19 07:49:14.179 [info] Skipping acquiring lock for c:\Users\ken\AppData\Roaming\Code\User\workspaceStorage\c1c2ea82a97bb4735bd875400adb74c7.
2023-03-19 07:49:14.496 [info] ExtensionService#_doActivateExtension vscode.github-authentication, startup: false, activationEvent: 'onAuthenticationRequest:github'
2023-03-19 07:49:14.528 [info] ExtensionService#_doActivateExtension vscode.emmet, startup: false, activationEvent: 'onLanguage'
2023-03-19 07:49:14.538 [info] ExtensionService#_doActivateExtension vscode.git-base, startup: true, activationEvent: '*', root cause: vscode.git
2023-03-19 07:49:14.544 [info] ExtensionService#_doActivateExtension christian-kohler.path-intellisense, startup: true, activationEvent: '*'
2023-03-19 07:49:14.597 [info] ExtensionService#_doActivateExtension curdeveryday.vscode-m5stack-mpy, startup: true, activationEvent: '*'
2023-03-19 07:49:14.693 [info] ExtensionService#_doActivateExtension Shan.code-settings-sync, startup: true, activationEvent: '*'
2023-03-19 07:49:15.232 [info] ExtensionService#_doActivateExtension WallabyJs.quokka-vscode, startup: true, activationEvent: '*'
2023-03-19 07:49:15.245 [info] ExtensionService#_doActivateExtension paulober.pico-w-go, startup: true, activationEvent: 'workspaceContains:.picowgo'
2023-03-19 07:49:15.849 [info] ExtensionService#_doActivateExtension vscode.git, startup: true, activationEvent: '*'
2023-03-19 07:49:15.881 [info] ExtensionService#_doActivateExtension vscode.github, startup: true, activationEvent: '*'
2023-03-19 07:49:16.205 [info] ExtensionService#_doActivateExtension vscode.microsoft-authentication, startup: false, activationEvent: 'onAuthenticationRequest:microsoft'
2023-03-19 07:49:16.230 [info] ExtensionService#_doActivateExtension ms-python.python, startup: true, activationEvent: 'workspaceContains:*.py', root cause: ms-python.isort
2023-03-19 07:49:18.450 [info] ExtensionService#_doActivateExtension ms-python.isort, startup: true, activationEvent: 'workspaceContains:*.py'
2023-03-19 07:49:19.843 [info] Eager extensions activated
2023-03-19 07:49:20.897 [info] ExtensionService#_doActivateExtension vscode.debug-auto-launch, startup: false, activationEvent: 'onStartupFinished'
2023-03-19 07:49:20.899 [info] ExtensionService#_doActivateExtension vscode.merge-conflict, startup: false, activationEvent: 'onStartupFinished'
2023-03-19 07:49:20.906 [info] ExtensionService#_doActivateExtension ms-vscode-remote.remote-wsl-recommender, startup: false, activationEvent: 'onStartupFinished'
2023-03-19 07:49:20.918 [info] ExtensionService#_doActivateExtension dbaeumer.vscode-eslint, startup: false, activationEvent: 'onStartupFinished'
2023-03-19 07:49:20.935 [info] ExtensionService#_doActivateExtension GitHub.copilot, startup: false, activationEvent: 'onStartupFinished'
2023-03-19 07:49:21.146 [info] ExtensionService#_doActivateExtension GitHub.vscode-pull-request-github, startup: false, activationEvent: 'onStartupFinished'
2023-03-19 07:49:21.190 [info] ExtensionService#_doActivateExtension ms-vscode-remote.remote-containers, startup: false, activationEvent: 'onStartupFinished'
2023-03-19 07:49:21.275 [info] ExtensionService#_doActivateExtension PKief.material-icon-theme, startup: false, activationEvent: 'onStartupFinished'
2023-03-19 07:49:21.511 [info] ExtensionService#_doActivateExtension vscode.configuration-editing, startup: false, activationEvent: 'onLanguage:jsonc'
2023-03-19 07:49:21.517 [info] ExtensionService#_doActivateExtension vscode.json-language-features, startup: false, activationEvent: 'onLanguage:jsonc'
2023-03-19 07:49:21.545 [info] ExtensionService#_doActivateExtension vscode.typescript-language-features, startup: false, activationEvent: 'onLanguage:jsonc'
2023-03-19 07:49:21.563 [info] ExtensionService#_doActivateExtension ms-dotnettools.vscode-dotnet-runtime, startup: false, activationEvent: 'onLanguage:jsonc', root cause: ms-azuretools.vscode-bicep
2023-03-19 07:49:22.508 [info] ExtensionService#_doActivateExtension ms-azuretools.vscode-bicep, startup: false, activationEvent: 'onLanguage:jsonc'
2023-03-19 07:49:22.636 [info] ExtensionService#_doActivateExtension msazurermtools.azurerm-vscode-tools, startup: false, activationEvent: 'onLanguage:jsonc'
2023-03-19 07:49:28.963 [error] Error: timeout
    at Timeout._onTimeout (c:\Users\ken\.vscode\extensions\paulober.pico-w-go-2.1.8-win32-x64\dist\extension.js:2:687995)
    at listOnTimeout (node:internal/timers:559:17)
    at processTimers (node:internal/timers:502:7)

## Any other comments?
OS: Windows 10 (OS Build: 19045.2728)
VSCode: 1.76.2
Pico-W-Go: 
More Info
Published 8/10/2022, 10:43:54
Last released 3/4/2023, 06:04:16
Last updated 3/11/2023, 06:43:32
Identifier paulober.pico-w-go

## Support info

Copy this from the `Help -> Info/About -> Copy` option in Visual Studio Code:

```
Version: 1.76.2 (system setup)
Commit: ee2b180d582a7f601fa6ecfdad8d9fd269ab1884
Date: 2023-03-14T17:55:54.936Z
Electron: 19.1.11
Chromium: 102.0.5005.196
Node.js: 16.14.2
V8: 10.2.154.26-electron.0
OS: Windows_NT x64 10.0.19045
Sandboxed: Yes
```

paulober, 2023-03-19:
Thanks for opening an issue here.
Normaly this kind of timeout problems can be solved by clicking the "Pico Disconnected" button, wait a second and click it again, or by unpluging the pico and then pluging it  back in. The extension should then auto-connect.

As mentioned in other issues, i'm currently working on a complete rewrite of the communitcation system in the extension.
That should fix all these connection, upload issues some people are having.

kfaubel, 2023-03-23:
Thanks for the suggestion.  This is the output when I tried it.  I did a fresh reboot and did not do anything else that touched the COM port.  Device Manager reports COM7 is present.

```
(AutoConnect enabled, ignoring 'manual com port' address setting)
Searching for boards on serial devices...
Connecting to COM7...                  <== This is where it gets stuck when I launch VSCode
 
Disconnected                           <== Here I pressed Pico W Disconnected, waited a couple of seconds and clicked again
Connecting to COM7...
 
Disconnected                           <== Tried again
Connecting to COM7...
 
Disconnected

Previous board is not available anymore      <== Here I unplugged the Pico W and then plugged it back in again.
Connecting to COM7...
```

If I shut down VSCode and then run Thonny I see it connect right away on COM7
```
MicroPython v1.19.1-966-g05bb26010 on 2023-03-13; Raspberry Pi Pico W with RP2040

Type "help()" for more information.

>>> 
```

David00, 2023-04-04:
I am having this problem too, running VS Code on Win 11 with Ubuntu 22.04 installed via WSL.

The extension does not work in WSL, but it does outside of WSL.

I created a new project in my Windows environment, configured it, and the extension connects to the Pico quickly.

However, if I create a new project in my WSL environment, the extension fails to connect to the Pico. (The extension is installed in WSL as well)

@paulober if you'd like me to provide any logs or debugs for this, let me know. I just wanted to share my findings in case this helps you with your efforts in rewriting the comm system.

paulober, 2023-04-11:
@David00 Thanks for the offering and some more detail on this issue. In the last days i release the new version (v3) of the extension (completely rewritten from ground up). Can you confirm if the issue still exists for you which the new version? (also @kfaubel)

David00, 2023-04-11:
@paulober, awesome! I just tested in both of my environments (WSL and Windows) and it is now connecting properly in both environments.  

I see that the Soft Reset functionality is exposed through the keyboard shortcut menu but does not have a default mapping.  I mapped it to `ctrl-d` and unmapped the VSCode default for ctrl-d, but I cannot seem to get the soft reload to work when pressing ctrl-d.  

I see the Reset button in the bottom toolbar, which seems to work fine, but I was hoping to have a similar serial console experience like when using puTTY or Thonny.  I'm fine with using the Reset button in the toolbar, but I wanted to let you know about this small issue too.

paulober, 2023-04-11:
@David00 I'm glad to here that at least for you it solved the issue. 

I'll take you suggestion with the Ctrl-d as feature request for an upcoming release. Have i understand it right: You set a keyboard shortcut to the softreset command but it is not working if you are working in the integrated vREPL, but you would like to use it there without having to go back to the UI buttons? I've thats the case the issue is due to most keystrokes are redirected to the vREPL which alters these, and not handled by vscode. This could be implemented....

David00, 2023-04-11:
Actually, it looks like the ctrl-d does work, but _only_ when not focused on VSCode's integrated terminal.  If I am actively editing a text file, pressing ctrl-d does in fact perform the soft reset.  It's only when actively typing in the console that the soft reset (via ctrl-d) does not work.

This is a super minor thing - I was just hoping to mirror the same experience as puTTY or Thonny's serial connection.

Here's a screenshot of what I mean:

[image]

paulober, 2023-04-11:
I know, that's what i mean. Because if you focus the terminal the extension as to deal with the input and vscode does not care only for a few sequences.

David00, 2023-04-11:
> I know, that's what i mean. Because if you focus the terminal the extension as to deal with the input and vscode does not care only for a few sequences.

I see - so in a "standard" serial connection (like via puTTY), is the ctrl-d captured by the application and replaced with a special signal or code? Or, is the ctrl-d sequence actually sent over the wire to the target device?

kfaubel, 2023-04-12:
@paulober Nice job.  Connection issue is no longer happening.  Some of the other issues I was seeing are corrected as well.

Thank you!

paulober, 2023-04-12:
@kfaubel I'm glad to here that the connection issue is resolved for you. What do you mean by other issues?

closed 2023-04-12 by paulober