paulober/MicroPico #56: Port update in settings is not reflected while connecting

hex007, 2023-04-05:
## What are the steps to reproduce this issue?
1.  Set port to /dev/ttyUSB0 with auto connect disabled
2. Restart VScode
3. Update port to /dev/ttyACM0 (pico's port) with auto connect disabled
4. Try connecting to Pi Pico

## What happens?
Connection attempt is made on /dev/ttyUSB0 rather than updated port and needs a restart to connect successfully

## What were you expecting to happen?
Connection attempt is made on /dev/ttyACM0, the updated port without needing a restart to connect successfully

## Any logs, error output, etc?

## Any other comments?

## Support info
```
Version: 1.76.2
Commit: ee2b180d582a7f601fa6ecfdad8d9fd269ab1884
Date: 2023-03-14T17:57:21.103Z
Electron: 19.1.11
Chromium: 102.0.5005.196
Node.js: 16.14.2
V8: 10.2.154.26-electron.0
OS: Linux x64 6.1.18-200.fc37.x86_64
Sandboxed: No
```

paulober, 2023-04-11:
Is the issue solved with V3, if not i'll take a look at it.

paulober, 2023-06-14:
... closed due to no response for over a month. Feel free to reopen if the issue still persists

closed 2023-06-14 by paulober