paulober/MicroPico #59: Cannot run anything since 3.0.0 update

bcseh202, 2023-04-10:
## What are the steps to reproduce this issue?
1. VS Code auto installed the update
2. Trying to run a code on the Pico or upload, doesn't matter
3. Installed pyserial correctly, reinstalled Pico-W-Go, even reinstalled the older version

## What happens?
VS Code installed the new 3.0.0 update and it broke everything for me. On the newer version, it says ''command 'picowgo.initialise' not found" and it doesn't start. I installed pyserial and reinstalled the Pico-W-Go but it still asks for pyserial everytime (the paths are correct for Python) and crashes with the error above.
I tried rolling back to a previous version but then the REPL doesn't come to life, it says that "Connecting on COM3 (correct)", new line and nothing, i cannot put a single character there, and the Run button simply "time outs", it is like it does something but then nothing happens. The board got the newest firmware and i nuked everyhing from it before.

## What were you expecting to happen?
I expect at least the roll-back version to work.

## Any logs, error output, etc?
*(To get extension logs see Terminal > Output > Protcoll (Extensionhost) and copy output concerning picowgo)*
*(If it’s long, please paste to https://gist.github.com and insert the link here)*
I cannot find logs there.

## Any other comments?
[image]
[image]
[image]

## Support info

Copy this from the `Help -> Info/About -> Copy` option in Visual Studio Code:

```
Version: 1.77.1 (user setup)
Commit: b7886d7461186a5eac768481578c1d7ca80e2d21
Date: 2023-04-04T23:21:11.906Z
Electron: 19.1.11
Chromium: 102.0.5005.196
Node.js: 16.14.2
V8: 10.2.154.26-electron.0
OS: Windows_NT x64 10.0.22621
Sandboxed: No
```

paulober, 2023-04-10:
I'm sorry to here that you are having troubble with the new version. I haven't had any issues during my testing on Windows 11 so i'm confused. If you can provide me some detail about the crash i'll try to fix it.
1. Do you have a default Python PATH user install or a different environment like Conda?
2. What is the output for `<your path to python.exe> -m pip show pyserial`?
3. Have you tried settings the path to your python.exe in the picowgo global settings in vscode?

kchimera, 2023-04-10:
Hi, I also have encountered this issue and even when forcing the extension via the settings to the python.exe is hangs saying 'Activating Extension' at the bottom and does nothing although it does suppress  the 'pyserial' not installed' message.

Here's my details:

Version: 1.77.1 (user setup)
Commit: b7886d7461186a5eac768481578c1d7ca80e2d21
Date: 2023-04-04T23:21:11.906Z
Electron: 19.1.11
Chromium: 102.0.5005.196
Node.js: 16.14.2
V8: 10.2.154.26-electron.0
OS: Windows_NT x64 10.0.19044
Sandboxed: Yes

Name: pyserial
Version: 3.5
Summary: Python Serial Port Extension
Home-page: https://github.com/pyserial/pyserial
Author: Chris Liechti
Author-email: cliechti@gmx.net
License: BSD
Location: c:\users\disko\appdata\local\programs\python\python310\lib\site-packages
Requires:
Required-by:

[image]

paulober, 2023-04-10:
@kchimera Thanks for adding more details. 
So you don't get a "Python3 is not installed" warning or "Install pyserial" popup when opening a workspace containing a .picowgo file?

kchimera, 2023-04-10:
I originally received the message to install 'pyserial', even when confirmed installed using 'pip pip install pyserial'; the installation command seemed to do nothing also?
Confirmed I have my 1 python installation in the 'PATH' inc 'scripts'
The message went away when I set the path to the python.exe manually in the Pico-W-Go' extension settings; Although I believe this doesn't make the extension work as any command through the palette seems to always attempt a ' Activating Extensions...' message in VSCode.

Thanks for you support!

paulober, 2023-04-10:
Nope that's the old terminal

kchimera, 2023-04-10:
Ah ok, my bad, It was my patience. Because the message disappears, didn't wait long enough for it to say it had installed the 'pySerial package'; I just assumed it was doing nothing sorry.

I've got it working now, it seems to be that (from the settings merge it showed me) that my path in the  for python was blank, and after I deleted it, its now written 'python.exe'  in there, which I assume should be the default; even though it says 'null' in the box?

[image]

paulober, 2023-04-10:
@kchimera I'm glad that you found the problem. Yes, the default of the settings null but when running the extension it will chose a default for you platform if the setting is null and then also sets the setting to this value.

The `autoConnect` feature is currently out of order and will be added in a future minor update.

kchimera, 2023-04-11:
Ok thanks, it seems then that having nothing written in the 'Python Path' field makes it fail to work.

bcseh202, 2023-04-11:
Hi!
Sorry, i'm from Hungary, the time zones are far away :)

Pip: 
` 
Name: pyserial
Version: 3.5
Summary: Python Serial Port Extension
Home-page: https://github.com/pyserial/pyserial
Author: Chris Liechti
Author-email: cliechti@gmx.net
License: BSD
Location: c:\users\cseh_\appdata\local\programs\python\python310\lib\site-packages      
Requires:
Required-by:
PS C:\Users\cseh_\.vscode\Workspace\Pico\MPython\Get Started With MicroPython\Chapter 3>
`

Also the path for the python interpreter as well as Pico-W-Go are correct and set.
The 'install pyserial' that the extension gives me does nothing also i noticed that everytime i leave my pico plugged in to my omputer and install the 3.0.0 version it messes up the firmware so i had to nuke.uf2 and reflash it before it can even work on the rollback version.

paulober, 2023-04-11:
@kchimera I added the autoConnect now with Patch 2, thanks for the hint.

paulober, 2023-04-11:
@bcseh202 Thanks for the more details, the issue you describe with v2.1.8 should be resolved with v3.0.1+ but concerning the other problems you mentioned i'll take a look at.

bcseh202, 2023-04-11:
@paulober Thanks for your help in advance! :)
Also i cannot use 3.0.1 too (same pyserial wanted but already installed and unable to start the extension or run anything), idk if i mentioned it earlier.

paulober, 2023-04-11:
Yes so if you get a command not found error you should find a log output int the output pane (in the terminal panel) there you can select 'ExtensionHost' in there you should find some logs concerning pico-w-go, it could help me diagnose the problem, if you could send me this logs please?

bcseh202, 2023-04-12:
@paulober So i got the logs below, i hope this helps!
Also ```C:\Users\cseh_\AppData\Local\Programs\Python\Python310\python.exe``` is my python path but i installed it manually yesterday, now when i installed 3.0.2 just for the logs i got the same old pyserial missing message.

```
2023-04-12 09:13:14.838 [error] Activating extension paulober.pico-w-go failed due to an error:
2023-04-12 09:13:14.838 [error] Error: Faild to install pyserial pip package. Manual install required!
	at gr.activate (c:\Users\cseh_\.vscode\extensions\paulober.pico-w-go-3.0.2-win32-x64\dist\extension.cjs:10:84769)
	at async exports.activate (c:\Users\cseh_\.vscode\extensions\paulober.pico-w-go-3.0.2-win32-x64\dist\extension.cjs:10:98612)
	at async E.n (c:\Users\cseh_\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:96:6646)
	at async E.m (c:\Users\cseh_\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:96:6609)
	at async E.l (c:\Users\cseh_\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:96:6066)

```

asltpo, 2023-04-13:
Hi @paulober, I'm getting similar behaviour to the OP.  v2.1.8 works fine but v3.0.2 gives me the following:

```
2023-04-13 13:14:24.100 [error] Activating extension paulober.pico-w-go failed due to an error:
2023-04-13 13:14:24.100 [error] Error: Faild to install pyserial pip package. Manual install required!
	at gr.activate (c:\Users\tomof\.vscode\extensions\paulober.pico-w-go-3.0.2-win32-x64\dist\extension.cjs:10:84769)
	at async exports.activate (c:\Users\tomof\.vscode\extensions\paulober.pico-w-go-3.0.2-win32-x64\dist\extension.cjs:10:98612)
	at async E.n (c:\Users\tomof\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:96:6646)
	at async E.m (c:\Users\tomof\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:96:6609)
	at async E.l (c:\Users\tomof\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:96:6066)
```
I have tried with and without installing pyserial manually beforehand.  Both python and pyserial executables are local.  VSCode details:

[image]

Thanks for a really useful bit of software.

paulober, 2023-04-13:
@bcseh202 @asltpo Thanks for all the logs. To fix this problem as fast as possible I would have a special version of v3.0.2 for you which generates a trace after the failed activation into the "ExtensionHost" output like the other logs:
[pico-w-go-win32-x64-3.0.2.zip](https://github.com/paulober/Pico-W-Go/files/11222636/pico-w-go-win32-x64-3.0.2.zip)
If you have time and want to try it out, please send me the output back so i can work on a fix. And to be sure, this is not a fix, just extended logging output enabled!

asltpo, 2023-04-13:
Hi @paulober, I ran the special version and got this:

```
2023-04-13 15:01:20.415 [info] ExtensionService#_doActivateExtension paulober.pico-w-go, startup: false, activationEvent: 'workspaceContains:.picowgo'
2023-04-13 15:01:20.779 [info] ExtensionService#_doActivateExtension vscode.configuration-editing, startup: false, activationEvent: 'onLanguage:jsonc'
2023-04-13 15:01:20.783 [info] ExtensionService#_doActivateExtension vscode.json-language-features, startup: false, activationEvent: 'onLanguage:jsonc'
2023-04-13 15:01:20.798 [info] ExtensionService#_doActivateExtension vscode.typescript-language-features, startup: false, activationEvent: 'onLanguage:jsonc'
2023-04-13 15:01:24.774 [error] Error: spawn C:\Users\tomof\AppData\Roaming\Microsoft\Windows\Start Menu\Programs\Python 3.11 ENOENT
	at ChildProcess._handle.onexit (node:internal/child_process:283:19)
	at onErrorNT (node:internal/child_process:478:16)
	at process.processTicksAndRejections (node:internal/process/task_queues:83:21)
```

paulober, 2023-04-13:
@asltpo Could it be that you've set you start menu path as python path? And also i'm not sure that this error is concerning Pico-W-Go..

asltpo, 2023-04-13:
Hi @paulober, apologies I'm new to VSCode so it's quite possible I've got some incorrect settings, although it must be something that only shows up with v3+ since v2.1.8 works fine.

My python path is C:\Users\tomof\AppData\Local\Programs\Python\Python311\python.exe

paulober, 2023-04-13:
@asltpo Np, there log i'm searching to fix the bug concerning the "Failed to install pyserial" bug should contain something like "at ... .vscode\extensions\paulober.pico-w-go-3.0.2-win32-x64 ..." below it and you can find it where you found "Error: Faild to install pyserial pip package. Manual install required!" error before. And make sure to uninstall Pico-W-Go before installing the vsix package i uploaded here in the issue.

bcseh202, 2023-04-13:
@paulober I installed it succesfully, now the pyserial need doesn't pop up but i don't get any logs. (the extension cannot start either)
Only interesting or relevant stuffs in logs are:
```
2023-04-13 16:29:12.743 [error] Error: spawn C:\Users\cseh_\AppData\Local\Programs\Python\Python310\ ENOENT
	at ChildProcess._handle.onexit (node:internal/child_process:283:19)
	at onErrorNT (node:internal/child_process:478:16)
	at process.processTicksAndRejections (node:internal/process/task_queues:83:21)

2023-04-13 16:29:12.229 [info] ExtensionService#_doActivateExtension paulober.pico-w-go, startup: true, activationEvent: 'workspaceContains:.picowgo'
```

asltpo, 2023-04-13:
Hi @paulober, I've just repeated the test.  To clarify this involved:
1. Uninstall the current version (2.1.8)
2. Install special version from vsix
3. Ctrl-shift-P -> Configure Project

The snippet below is the entirety of the Extension Host output from the point I started the test to the end.

```
2023-04-13 15:27:31.603 [info] Extension host terminating: renderer closed the MessagePort
2023-04-13 12023-04-13 15:27:31.603 [info] Extension host terminating: renderer closed the MessagePort
2023-04-13 15:27:31.608 [info] Extension host with pid 10032 exiting with code 0
2023-04-13 15:27:40.571 [info] Extension host with pid 10296 started
2023-04-13 15:27:40.571 [info] Skipping acquiring lock for c:\Users\tomof\AppData\Roaming\Code\User\workspaceStorage\f4d8913a4731384a0611a9289bd41822.
2023-04-13 15:27:40.586 [info] ExtensionService#_doActivateExtension vscode.emmet, startup: false, activationEvent: 'onLanguage'
2023-04-13 15:27:40.591 [info] ExtensionService#_doActivateExtension vscode.github-authentication, startup: false, activationEvent: 'onAuthenticationRequest:github'
2023-04-13 15:27:40.620 [info] ExtensionService#_doActivateExtension vscode.git-base, startup: true, activationEvent: '*', root cause: vscode.git
2023-04-13 15:27:40.625 [info] ExtensionService#_doActivateExtension vscode.git, startup: true, activationEvent: '*'
2023-04-13 15:27:40.641 [info] ExtensionService#_doActivateExtension vscode.github, startup: true, activationEvent: '*'
2023-04-13 15:27:40.643 [info] ExtensionService#_doActivateExtension vscode.microsoft-authentication, startup: false, activationEvent: 'onAuthenticationRequest:microsoft'
2023-04-13 15:27:40.687 [info] Eager extensions activated
2023-04-13 15:27:40.688 [info] ExtensionService#_doActivateExtension vscode.debug-auto-launch, startup: false, activationEvent: 'onStartupFinished'
2023-04-13 15:27:40.691 [info] ExtensionService#_doActivateExtension vscode.merge-conflict, startup: false, activationEvent: 'onStartupFinished'
2023-04-13 15:27:40.697 [info] ExtensionService#_doActivateExtension ms-vscode-remote.remote-wsl-recommender, startup: false, activationEvent: 'onStartupFinished'
2023-04-13 15:27:42.398 [info] ExtensionService#_doActivateExtension paulober.pico-w-go, startup: false, activationEvent: 'workspaceContains:.picowgo'
2023-04-13 15:27:42.739 [info] ExtensionService#_doActivateExtension vscode.configuration-editing, startup: false, activationEvent: 'onLanguage:jsonc'
2023-04-13 15:27:42.743 [info] ExtensionService#_doActivateExtension vscode.json-language-features, startup: false, activationEvent: 'onLanguage:jsonc'
2023-04-13 15:27:42.762 [info] ExtensionService#_doActivateExtension vscode.typescript-language-features, startup: false, activationEvent: 'onLanguage:jsonc'
2023-04-13 15:27:46.802 [error] Error: spawn C:\Users\tomof\AppData\Roaming\Microsoft\Windows\Start Menu\Programs\Python 3.11 ENOENT
	at ChildProcess._handle.onexit (node:internal/child_process:283:19)
	at onErrorNT (node:internal/child_process:478:16)
	at process.processTicksAndRejections (node:internal/process/task_queues:83:21)5:27:31.608 [info] Extension host with pid 10032 exiting with code 0
2023-04-13 15:27:40.571 [info] Extension host with pid 10296 started
2023-04-13 15:27:40.571 [info] Skipping acquiring lock for c:\Users\tomof\AppData\Roaming\Code\User\workspaceStorage\f4d8913a4731384a0611a9289bd41822.
2023-04-13 15:27:40.586 [info] ExtensionService#_doActivateExtension vscode.emmet, startup: false, activationEvent: 'onLanguage'
2023-04-13 15:27:40.591 [info] ExtensionService#_doActivateExtension vscode.github-authentication, startup: false, activationEvent: 'onAuthenticationRequest:github'
2023-04-13 15:27:40.620 [info] ExtensionService#_doActivateExtension vscode.git-base, startup: true, activationEvent: '*', root cause: vscode.git
2023-04-13 15:27:40.625 [info] ExtensionService#_doActivateExtension vscode.git, startup: true, activationEvent: '*'
2023-04-13 15:27:40.641 [info] ExtensionService#_doActivateExtension vscode.github, startup: true, activationEvent: '*'
2023-04-13 15:27:40.643 [info] ExtensionService#_doActivateExtension vscode.microsoft-authentication, startup: false, activationEvent: 'onAuthenticationRequest:microsoft'
2023-04-13 15:27:40.687 [info] Eager extensions activated
2023-04-13 15:27:40.688 [info] ExtensionService#_doActivateExtension vscode.debug-auto-launch, startup: false, activationEvent: 'onStartupFinished'
2023-04-13 15:27:40.691 [info] ExtensionService#_doActivateExtension vscode.merge-conflict, startup: false, activationEvent: 'onStartupFinished'
2023-04-13 15:27:40.697 [info] ExtensionService#_doActivateExtension ms-vscode-remote.remote-wsl-recommender, startup: false, activationEvent: 'onStartupFinished'
2023-04-13 15:27:42.398 [info] ExtensionService#_doActivateExtension paulober.pico-w-go, startup: false, activationEvent: 'workspaceContains:.picowgo'
2023-04-13 15:27:42.739 [info] ExtensionService#_doActivateExtension vscode.configuration-editing, startup: false, activationEvent: 'onLanguage:jsonc'
2023-04-13 15:27:42.743 [info] ExtensionService#_doActivateExtension vscode.json-language-features, startup: false, activationEvent: 'onLanguage:jsonc'
2023-04-13 15:27:42.762 [info] ExtensionService#_doActivateExtension vscode.typescript-language-features, startup: false, activationEvent: 'onLanguage:jsonc'
2023-04-13 15:27:46.802 [error] Error: spawn C:\Users\tomof\AppData\Roaming\Microsoft\Windows\Start Menu\Programs\Python 3.11 ENOENT
	at ChildProcess._handle.onexit (node:internal/child_process:283:19)
	at onErrorNT (node:internal/child_process:478:16)
	at process.processTicksAndRejections (node:internal/process/task_queues:83:21)
```
It doesn't produce any output like "at ... .vscode\extensions\paulober.pico-w-go-3.0.2-win32-x64 ...", although I do get that output when I run the release v3.0.2.

paulober, 2023-04-13:
@bcseh202 @asltpo Thanks for the quick responses. Here is a final logging version which should finally narrow don't the root of the bug: 
[pico-w-go-win32-x64-3.0.2.zip](https://github.com/paulober/Pico-W-Go/files/11223341/pico-w-go-win32-x64-3.0.2.zip)

bcseh202, 2023-04-13:
@paulober i rage uninstalled every python instance from my computer and reinstalled the latest 3.11.3 version and changed the paths in VS Code as well as your extension to the default 'python.exe'. And after a reboot you know what? Despite the 'Connection to Pico lost' spam every 5-10 seconds it finally seems to be working!
The previous version of python (3.10.10) was only because i wrote a small game using pygame and it was incompatible with the latest python at the time.
I installed the in-store version and this time i got another error but everything working just fine.

```
2023-04-13 16:51:57.705 [error] TypeError: this.proc.disconnect is not a function
	at ChildProcess.<anonymous> (c:\Users\cseh_\.vscode\extensions\paulober.pico-w-go-3.0.2-win32-x64\dist\extension.cjs:1:4352)
	at ChildProcess.emit (node:events:526:28)
	at onSpawnNT (node:internal/child_process:483:8)
	at process.processTicksAndRejections (node:internal/process/task_queues:82:21)
```

Still leaves this thread open until a fix or at least a fix for @asltpo if you don't mind :)

asltpo, 2023-04-13:
Hi @paulober, I have added the following line to `.vscode/settings.json`:

`"picowgo.pythonPath": "C:\\Users\\tomof\\AppData\\Local\\Programs\\Python\\Python311\\python.exe"`

The release version of v3.0.2 now works without errors.  I must have somehow created two competing installations of python on different paths.  Apologies!

Thanks for the prompt support.

paulober, 2023-04-13:
Ok, so i'll close this issue as it seem to be caused by conflicting Python installations on Windows.

closed 2023-04-13 by paulober