paulober/MicroPico #65: Extension not working

KristianBalaj, 2023-04-12:
## What are the steps to reproduce this issue?
1. Run `Pico-W-Go > Configure project`

Or:

1. Open Pico W vRepl in terminal window

## What happens?

Only the activating extensions at the bottom bar appears for a few seconds and then nothing.

[image]

## What were you expecting to happen?

Some configuration to happen and having these options in the bottom bar 
[image]

Or opening the pico repl.

## Any logs, error output, etc?
*(To get extension logs see Terminal > Output > Protcoll (Extensionhost) and copy output concerning picowgo)*
*(If it’s long, please paste to https://gist.github.com and insert the link here)*

[image]

## Any other comments?

Connecting to pi works inside the Thonny IDE.

## Support info

Copy this from the `Help -> Info/About -> Copy` option in Visual Studio Code:

```
Version: 1.77.2 (Universal)
Commit: e344f1f539a80912a0e9357cec841f36ce97a4e2
Date: 2023-04-06T10:12:31.126Z
Electron: 19.1.11
Chromium: 102.0.5005.196
Node.js: 16.14.2
V8: 10.2.154.26-electron.0
OS: Darwin x64 22.3.0
Sandboxed: Yes
```

paulober, 2023-04-13:
Hi @KristianBalaj, what is your concrete problem? Do you have any notifications after running "Configure Project" and what does the vREPL show after you opened it? Have you check that you have a clean python 3.9+ python installation? What have you chosen in the installation dialog for `pyserial`?

KristianBalaj, 2023-04-13:
@paulober thanks for reaching out to me

> what is your concrete problem? 

The problem is that nothing happens after the `Configure project`

> what does the vREPL show after you opened it?

As I've posted - just the Activating extension in the bottom bar appears and nothing happens afterwards.

I'm selecting auto install:
[image]

I do have python 3.9+ version:
[image]

paulober, 2023-04-13:
So after "Auto Install" what notification have you got in vscode and what logs appear in the extension host output after selecting "Auto install"?

Teckdev, 2023-04-14:
I have the exact same issue. I get the `Activating extensions` in the toolbar for a couple of seconds then it disappears and nothin happens. It does that when I try to use **any** `Pico-W-Go` item in the command palette.

I tried all the 3.x version and I always get the issue. Reverting to 2.1.8 works as expected. 

```
2023-04-13 20:41:28.931 [info] Extension host with pid 18076 started
2023-04-13 20:41:28.931 [info] Skipping acquiring lock for /Users/dumontd/Library/Application Support/Code/User/workspaceStorage/b15ef720e3d10c8a413ccbe96036bfaf.
2023-04-13 20:41:28.964 [info] ExtensionService#_doActivateExtension vscode.emmet, startup: false, activationEvent: 'onLanguage'
2023-04-13 20:41:28.974 [info] ExtensionService#_doActivateExtension ms-python.python, startup: false, activationEvent: 'onLanguage:python', root cause: ms-python.isort
2023-04-13 20:41:29.094 [info] ExtensionService#_doActivateExtension ms-toolsai.jupyter, startup: false, activationEvent: 'onLanguage:python'
2023-04-13 20:41:29.189 [info] ExtensionService#_doActivateExtension VisualStudioExptTeam.intellicode-api-usage-examples, startup: false, activationEvent: 'onLanguage:python'
2023-04-13 20:41:29.262 [info] ExtensionService#_doActivateExtension VisualStudioExptTeam.vscodeintellicode, startup: false, activationEvent: 'onLanguage:python'
2023-04-13 20:41:29.455 [info] ExtensionService#_doActivateExtension ms-toolsai.jupyter-renderers, startup: false, activationEvent: 'api', root cause: ms-toolsai.jupyter
2023-04-13 20:41:29.463 [info] ExtensionService#_doActivateExtension vscode.github-authentication, startup: false, activationEvent: 'onAuthenticationRequest:github'
2023-04-13 20:41:29.507 [info] ExtensionService#_doActivateExtension ms-python.isort, startup: false, activationEvent: 'onLanguage:python'
2023-04-13 20:41:29.513 [info] ExtensionService#_doActivateExtension ms-python.vscode-pylance, startup: false, activationEvent: 'onLanguage:python'
2023-04-13 20:41:29.574 [info] ExtensionService#_doActivateExtension vscode.microsoft-authentication, startup: false, activationEvent: 'onAuthenticationRequest:microsoft'
2023-04-13 20:41:29.950 [info] ExtensionService#_doActivateExtension vscode.git-base, startup: true, activationEvent: '*', root cause: vscode.git
2023-04-13 20:41:29.951 [info] ExtensionService#_doActivateExtension paulober.pico-w-go, startup: true, activationEvent: 'workspaceContains:.picowgo'
2023-04-13 20:41:30.721 [info] ExtensionService#_doActivateExtension vscode.git, startup: true, activationEvent: '*'
2023-04-13 20:41:30.733 [info] ExtensionService#_doActivateExtension vscode.github, startup: true, activationEvent: '*'
2023-04-13 20:41:30.745 [info] ExtensionService#_doActivateExtension vscode.configuration-editing, startup: false, activationEvent: 'onLanguage:jsonc'
2023-04-13 20:41:30.747 [info] ExtensionService#_doActivateExtension vscode.json-language-features, startup: false, activationEvent: 'onLanguage:jsonc'
2023-04-13 20:41:30.754 [info] ExtensionService#_doActivateExtension vscode.typescript-language-features, startup: false, activationEvent: 'onLanguage:jsonc'
2023-04-13 20:41:30.760 [info] ExtensionService#_doActivateExtension ms-vscode-remote.remote-containers, startup: false, activationEvent: 'onLanguage:jsonc'
2023-04-13 20:41:39.953 [info] ExtensionService#_doActivateExtension vscode.debug-auto-launch, startup: false, activationEvent: 'onStartupFinished'
2023-04-13 20:41:39.956 [info] ExtensionService#_doActivateExtension vscode.merge-conflict, startup: false, activationEvent: 'onStartupFinished'
```

```
Version: 1.77.3 (Universal)
Commit: 704ed70d4fd1c6bd6342c436f1ede30d1cff4710
Date: 2023-04-12T09:19:37.325Z (1 day ago)
Electron: 19.1.11
Chromium: 102.0.5005.196
Node.js: 16.14.2
V8: 10.2.154.26-electron.0
OS: Darwin arm64 22.4.0
Sandboxed: No
```

Davis8483, 2023-04-14:
Also have the same issue.

paulober, 2023-04-14:
@Davis8483 @Teckdev  have you read and checked the requirements? Python 3 installed and in PATH. To be sure set it in the extension settings (the path to your python executable) and select "auto install" for `pyserial` after setting it up. That is the reason why it tries to activate each time you want to run a command. Also if it's not that, i don't have a Mac so I won't be able to search for logs and so on my own.

Teckdev, 2023-04-14:
> @Davis8483 @Teckdev have you read and checked the requirements? Python 3 installed and in PATH. To be sure set it in the extension settings (the path to your python executable) and select "auto [...]

I do have python3 installed and in my path. I can't set it in the extension settings as it won't even open - when I click on the menu item for the settings, I get the `Activating extensions` appear/disappear and the settings page does not open. If you need some logs from my Mac, I can provide them. Just let me know what you need!

paulober, 2023-04-14:
@Teckdev Thanks for the offering. If you cannot open the vscode settings you can manually edit the settings by opening `$HOME/Library/Application\ Support/Code/User/settings.json` json file and set `picowgo.pythonPath` to your python executable and `picowgo.autoConnect` to true

Teckdev, 2023-04-14:
@paulober I get the same behaviour. I explicitly set my python path in `picowgo.pythonPath`.  Note that there was no `picowgo.autoConnect`. I tried changing `picowgo.ctrlCOnConnect` to `picowgo.autoConnect` but no luck.
My `settings.json` is pretty slim so I will paste it here:
```
{
    "explorer.confirmDragAndDrop": false,
    "explorer.confirmDelete": false,
    "files.autoSave": "afterDelay",
    "[python]": {
        "editor.formatOnType": true
    },
    "python.defaultInterpreterPath": "/usr/local/bin/python",
    "picowgo.ctrlCOnConnect": true,
    "picowgo.pythonPath": "/usr/local/bin/python3",
    "picowgo.statusbarButtons": [
        "status"
    ]
}
```

This is the python version for `/usr/local/bin/python3`:
```
$ /usr/local/bin/python3
Python 3.9.13 (v3.9.13:6de2ca5339, May 17 2022, 11:37:23)
[Clang 13.0.0 (clang-1300.0.29.30)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>>
```

paulober, 2023-04-14:
You should remove "picowgo.statusbarButtons" and if there is no picowgo.autoConnect then just add it.

Teckdev, 2023-04-14:
Same behaviour with those configs:
```
{
    "explorer.confirmDragAndDrop": false,
    "explorer.confirmDelete": false,
    "files.autoSave": "afterDelay",
    "[python]": {
        "editor.formatOnType": true
    },
    "python.defaultInterpreterPath": "/usr/local/bin/python",
    "picowgo.ctrlCOnConnect": true,
    "picowgo.autoConnect": true,
    "picowgo.pythonPath": "/usr/local/bin/python3"
}
```
Also tried without `"picowgo.ctrlCOnConnect": true,` but doesn't make any difference.

paulober, 2023-04-14:
Of course `"picowgo.ctrlCOnConnect": true,` does nothing as it does no longer exist with v3+.
I have no idea what the problem source could be, maybe some mac specific stuff. But i currently don't have time to play tech support so i'll look into it once i have more time or you could implement a fix and create a PR yourself.

Teckdev, 2023-04-14:
To be clear I am quite happy with `2.1.8` so I am not blocked, extension is quite helpful. 

I was just following up to help with the issue other users are also facing. I had no intentions of waisting your time **playing tech support.** - Code Owners are usually happy having feedback from users.

paulober, 2023-04-14:
> I was just following up to help with the issue other users are also facing. I had no intentions of waisting your time **playing tech support.** - Code Owners are usually happy having feedback from users.

That was not ment angry just to inform that i would like to fix this bug now but don't have the time and so that all in this thread know that any don't think why i'm not answering for 2 Months. Also i would like to see people contribute for this kind of bug (MacOS) as i'm not able to test on this platform so i don't have to block downloads for MacOS from the store to prevent other people from having the same issue.

paulober, 2023-04-14:
> Same error here, on Windows 11, pico works whit the extension version 2.1.8 or lower

Sure? What are your logs? Do you get any notifications by the extension? What have you already tried to fix the issue (settings etc...)?

JorgeL339, 2023-04-14:
Yup, here are the details Win11 22H2 Build 22624, python 3.11.3
1. Install the lates version of Pico-W-Go
2. Create a new folder, and select, configure proyect
3.Extension ask for Pyserial to be installed (Py serial is alreade installed, even so, it ask for it)
4.Click on AutoInstallPyserial
5.Pop up dissapears, but the proyect is not yet configured
6. Select configure proyect again
7.Extension gives error mesagge, something like, the picowgo.configure dosent exist
8.Try to open any other, functions of the extension (Upload, Download, etc) i get the same error, picowgo.upload or picowgo.download dosen exist

I can give you the logs in a few days, as now i am working on a proyect whit the 2.1.8 version, and i cant upgrade the version maybe it would damage the proyect

paulober, 2023-04-14:
Ok, thanks for posting the details and sorry for having to deal with this kind of annoying bug...

Teckdev, 2023-04-14:
For reference:

On my _MacMini (M1)_ I uninstalled the extension and pyserial. I then re-installed the extension and it did successfully auto install `pyserial`, but the issue remains.

On my _windows 11_ machine which never had the extension installed, I installed the extension and it auto installed pyserial. Everything works as expected.

I will try the same setup on my _MacBook Pro (M1 Pro)_ which also never had the extension installed. Need to find a usb-c adaptor so will try to do that tomorrow.

JorgeL339, 2023-04-18:
Hi, i have tested the lates release 3.0.4 in my windows 11 system and still get the same error wich is (When trying to configure the proyect for the first time after installing the extension):
[image]

Then if i try to use any othe commands on i get:
[image]
Or:
[image]
Depending on the command i want to use

Here are the logs:
https://gist.github.com/JorgeL339/fa0af84545bc37ecdf11cf6f85a461b4

They say that pyserial is not installe, but it is:
[image]

paulober, 2023-04-18:
@JorgeL339 Thanks for all the detailed logs, I think the problem could cause by your python installation which is not a user install but the Microsoft Store one. I'll do some more testing on this when i have time again...

JorgeL339, 2023-04-18:
Yup, it was that, i unistalle the python from the store, and installed it normally, version 3.11.3, and it worked as intended, thanks for your help

MortimerWittgenstein, 2023-04-20:
Hi,

I have the same Problem.
When running a Pico-W-Go command, the Status Bar says "Activating extension..." and then nothing happens.

On the first time running it, I got the notification to install pylance. I installed it automatically and got notified that the installation was successful.

This is the Extension Log:
2023-04-20 08:16:31.095 [info] ExtensionService#_doActivateExtension paulober.pico-w-go, startup: false, activationEvent: 'onCommand:picowgo.initialise'

Do you have an idea, how I can fix this?
Thanks

stig-bjorlykke, 2023-04-22:
Anything I can do to help figure out the reason for this issue?
I also have to downgrade to `2.1.8` to make this extension work on macOS M2.

paulober, 2023-04-23:
@MortimerWittgenstein First, you don't mean "Pylance" but "pyserial", please by detailed!
Second, please provide more detail on the bug you're encountering if you want it to be solved: You System details from the VSCode help menu, The version of Pico-W-Go you're using, The python version and installation method/path you're using.

MortimerWittgenstein, 2023-04-24:
> @MortimerWittgenstein First, you don't mean "Pylance" but "pyserial", please by detailed! Second, please provide more detail on the bug you're encountering if you want it to be solved: You System [...]

I'm sorry! Yes, I meant pyserial.

**System details:**
Version: 1.77.3 (system setup)
Commit: 704ed70d4fd1c6bd6342c436f1ede30d1cff4710
Date: 2023-04-12T09:16:02.548Z
Electron: 19.1.11
Chromium: 102.0.5005.196
Node.js: 16.14.2
V8: 10.2.154.26-electron.0
OS: Windows_NT x64 10.0.19045
Sandboxed: No

**Pico-W-Go:**
Version: v3.0.5

**Python:**
Version: 3.11.3
Installation: Python x64 Windows Installer (user install, non-store)

**Problem:**
When running a Pico-W-Go command, the Status Bar says "Activating extension..." and then nothing happens.

**I could fix it by myself now:**
I completely uninstalled Python and reinstalled it for all users.

I als have had the problem on another PC (Win11). Maybe reinstalling python works there, too.

Thanks for your help.

paulober, 2023-04-26:
Ok, i wonder what the problem with the extension and you first python install was...
On you second PC before reinstalling python i could be helpfull if you could provide error output from the Output window in vscode > Select "ExtensionHost" in the dropdown

martinowitsch, 2023-04-30:
I got exactly the same issue. Nothing happend with python 3.9 and picowgo 3.0.7 but it still worked with version 2.1.8. However, after upgrading to python 3.11 it seems to work without any issues with picowgo 3.0.7. So upgrading python version seems to do the workaround.

closed 2023-05-05 by paulober

KristianBalaj, 2023-05-10:
The latest version has fixed this for me 👏

PR #88 (2023-05-05): Patch v3.0.8
### Changed
- Fixes Python 3.9 compatibility
- Upgraded to `pyboard-serial-com` v1.4.24

### Related issues
- Should fix #65