paulober/MicroPico #66: Problem with uploading project and connection

sdmomo, 2023-04-12:
## What are the steps to reproduce this issue?
1. connect the Pico H on VSCode and configure a new project with Pico-W-Go
2. create a script MicroPython like blinking a LED
3. when uploading project, nothing happen. Run "uploading project" infinitely

## What happens?
If I connect the pico, I have two windows :
Pico vREPL not open and Connection to Pico lost

## What were you expecting to happen?

## Any logs, error output, etc?
*(To get extension logs see Terminal > Output > Protcoll (Extensionhost) and copy output concerning picowgo)*
*(If it’s long, please paste to https://gist.github.com and insert the link here)*

## Any other comments?

## Support info

Copy this from the `Help -> Info/About -> Copy` option in Visual Studio Code:

```
< Paste Here >
<img width="436" alt="MicrosoftTeams-image" src="https://user-images.githubusercontent.com/57638385/231547788-f0e8a9b7-e579-40c9-895f-4eb02a45c256.png">

```

sdmomo, 2023-04-12:
[image]

paulober, 2023-04-13:
> when uploading project, nothing happen. Run "uploading project" infinitely

Do you mean by that, the operation run infinitely or have you tried it infinitely. If it's the first, what is the indicator you get during the upload?
Also please profile a proper value for *Support info*

sdmomo, 2023-04-14:
The operation run infinitely. Nothing happen in the terminal when I plug the Pico in the computer and upload. 
The REPL terminal doesn't show and after the connection to Pico lost. After, I try to uploading my project, he doesn't work because the connection his lost.

It's arrive after I update to the version 3. 

I try to flash and re-install the micropython firmware and it's the same pattern.

I try in Thonny and the Pico work. 

Here the support info:
Version : 1.77.3 (user setup)
Validation : 704ed70d4fd1c6bd6342c436f1ede30d1cff4710
Date : 2023-04-12T09:16:02.548Z
Electron : 19.1.11
Chromium: 102.0.5005.196
Node.js : 16.14.2
V8 : 10.2.154.26-electron.0
Système d’exploitation : Windows_NT x64 10.0.22621
Sandboxé: No

paulober, 2023-04-26:
@borys-wwa Can you please open the vREPL before running you script: New Terminal > Pico W Go vREPL. Does this fix the problem with executing files?

borys-wwa, 2023-04-26:
@paulober yes - this did help, but only to some extent. The symptom now is as follows: if I open the Pico W Go vREPL terminal manually, I can run the script, and it does execute on a device, but I can do it only once. Once it's running, I can't stop it, I can't reset the pico, basically can do nothing. 
But perhaps it's a user error and I'm doing something wrong? I'm new to the Pico platform. Is there some special way to the python script that is running on the device, if started from the VS code? 
If I write anything on the previously opened terminal, it simply hungs. 
I can't even disconnect from Pico. I need to close VS Code window and reopen it, to be able to do anything.

paulober, 2023-04-26:
@borys-wwa Try to add a print("Something") at the beginning of you script if possible. I'm working on a fix for this issue.

martinowitsch, 2023-04-30:
I can confirm this. I got the same behaviour. The code is running at first on the PicoW but there is no output on my console and I read the same error:
[image]

dangjinghao, 2023-05-01:
Hi! I have the same problem looks like @borys-wwa 's [description](https://github.com/paulober/Pico-W-Go/issues/66#issuecomment-1523739646) . 
According to [raspberrypi forums post](https://forums.raspberrypi.com/viewtopic.php?p=2102005) I **temporary** resolved this by installing 2.1.8 (older version) Pico-W-Go extension.
I think there should be a better way to resolve the problem.

dangjinghao, 2023-05-01:
> @borys-wwa Try to add a print("Something") at the beginning of you script if possible. I'm working on a fix for this issue.

- print without opening  vREPL terminal
```python
from machine import Pin
from time import sleep
pin = Pin("LED",Pin.OUT)
print(3.14)
while True:
    pin.toggle()
    sleep(.1)
```

I can control the pico by button on the status bar (Run,stop and Reset after Stop (Even through I don't know is the Reset effective))

- print with opening  vREPL terminal
same as the status of print without opening  vREPL terminal

- don't print without  opening  vREPL terminal
```python
from machine import Pin
from time import sleep
pin = Pin("LED",Pin.OUT)
# print(3.14)
while True:
    pin.toggle()
    sleep(.1)
```

I can upload the code only once .The status bar will be no change and I can do nothing for pico.After vscode restart , it can't connect pico while the  pico is still blinding.

- don't print with  opening  vREPL terminal
same as the status of don't print without  opening  vREPL terminal

It seems that I must print something ,the status bar could be normal working.
vscode info:
```
Version: 1.77.3
Commit: 704ed70d4fd1c6bd6342c436f1ede30d1cff4710
Date: 2023-04-12T09:16:52.732Z
Electron: 19.1.11
Chromium: 102.0.5005.196
Node.js: 16.14.2
V8: 10.2.154.26-electron.0
OS: Linux x64 6.2.13-arch1-1
Sandboxed: No

Pico-W-Go v3.0.7
```

kingtimm, 2023-05-02:
> Please try following steps in this order:
> 
> * Check `picowgo.autoConnect` in the global settings to be true
> * The correct path to your python.exe in `picowgo.pythonPath` also in global settings
> * Connect the Pico to your PC
> * Open Visual Studio Code
> * Open a directory containing a `.picowgo`  (or a empty one and run "> Pico-W-Go > Configure Project")
> * Select ''auto install" for the pyserial requirement prompt
> * Wait until you get a error notification or success notification from the extension
> * Make sure that `picowog.syncFolder` in you `.vscode/settings.json` is set to the correct syncing directory (relative to the root of the folder you opened in vscode without '/' at the beginning or end of the path)
> * Now right click in the explorer (in vscode) and select "Pico-W-Go > Upload project"
> * Wait for a success or error notification

I can move this to another issue if you'd like, but my issue is similar to @sdmomo 's original issue where upload process just runs indefinitely. I'm not sure the other replies have the same issue.

Selecting "Pico-W-Go > Upload project" results in it a loop where the "Uploading Project..." notification doesn't go away. Trace log from the extension host reports the following:

```
2023-05-01 21:11:00.996 [trace] ExtHostCommands#$executeContributedCommand picowgo.upload
2023-05-01 21:11:01.382 [error] SyntaxError: Unexpected token c in JSON at position 18
	at JSON.parse (<anonymous>)
	at Socket.<anonymous> (c:\Users\kingt\.vscode\extensions\paulober.pico-w-go-3.0.7-win32-x64\dist\extension.cjs:1:9389)
	at Socket.emit (node:events:526:28)
	at addChunk (node:internal/streams/readable:315:12)
	at readableAddChunk (node:internal/streams/readable:289:9)
	at Readable.push (node:internal/streams/readable:228:10)
	at Pipe.onStreamRead (node:internal/stream_base_commons:190:23)
2023-05-01 21:11:01.383 [trace] forwarded error to extension? false {"value":"paulober.pico-w-go","c":"paulober.pico-w-go"}
```

It doesn't seem to be json in the project that is causing the issue. Anything else I can try?

paulober, 2023-05-05:
Maybe you global settings.json of you vscode installation has a json error which causes my extension to crash when it tries to read it...

kingtimm, 2023-05-08:
> @kingtimm Maybe you global settings.json of you vscode installation has a json error which causes my extension to crash when it tries to read it...

I can upload small projects no problem. It seems to have this issue only when I include the [nanogui "gui" folder](https://github.com/peterhinch/micropython-nano-gui/tree/master/gui) as part of my project.

Steps are:

1. Create a new folder and Configure Project using PIco-W-Go > Configure Project
2. Add the gui folder linked above to the directory
3. Select PIco-W-Go > Upload Project

1.

expor, 2023-05-09:
I'm observing the same issue. I setup a blank project, configure it and add a basic 'flash led' program. When pressing Run in the bottom of vscode (cannot right-click and run, option greyed out) it runs the first time. After that the connection seems to hang. I cannot interact with REPL nor Run another program.

Everything works again when restarting VS Code, but only once after which the above repeats.

MainzerKaiser, 2023-05-14:
Thanks for giving the hint on print() command, that changes the Run Button in the extension button list in the bottom to a stop button and I can interact with vs code without closing it.

batfink5000, 2023-05-15:
I'm getting similar issue to expoor.
I can run start / stop  code using the tool bar at the bottom provided there is a print statement in the code.
(cannot right-click and run, option greyed out)

Thanks :)

paulober, 2023-06-14:
@sdmomo @kingtimm Does the issue still exists with the newest version?

@batfink5000 @MainzerKaiser @expor Your issue should finally be resolved with v3.0.11 in the next few hours

kingtimm, 2023-06-17:
> @sdmomo @kingtimm Does the issue still exists with the newest version?
 
The issue still exists. I traced the json.parse error to this line: https://github.com/paulober/pyboard-serial-com/blob/28270fc7a6019aca7d7f492b701076c4c2d8c2bc/src/pyboardRunner.ts#L729

It looks like there's a version.json crested for file hashes when doing an upload. For whatever reason, the [nangui](https://github.com/peterhinch/micropython-nano-gui/tree/master/gui) library, when hashed, produced invalid json. I didn't continue my troubleshooting, but perhaps it is due to too many levels of folder nesting? I am not very skilled at debugging extensions.

paulober, 2023-06-17:
@kingtimm Thanks for the details. They helped to identify the source of the uploading issues mentioned in this issue thread. The fix also patches downloading project with only one file.

Let me know if the next patch of Pico-W-Go resolved your issue.

closed 2023-06-17 by paulober

kingtimm, 2023-06-20:
Can confirm - it's fixed! Thanks!! 🚀

PR #108 (2023-06-17): Patch v3.0.14
### Changed
- Fix #66, Problems with uploading/downloading a project
- Store listing improvements
- Upgrade to pyboard-serial-com v1.4.31
- Update terminal profile image
commit 07b63abe (2023-06-17): Fix #66, Problems with uploading/downloading a project

- Update to pyboard-serial-com v1.4.31