paulober/MicroPico #68: Commands/REPL Hangs After Run

jordantryon, 2023-04-14:
## What are the steps to reproduce this issue?
1. Launch VS Code (with Pico W already connected or connecting Pico W afterward).
2. Run a python file that contains time.sleep delays or while(True) loops using the "Run current file" Pico-W-Go command.

## What happens?
After running, the REPL terminal and commands become unresponsive. The "Run" button does not change to "Stop", cannot perform soft or hard resets, or send commands of any kind. The current python program does start and continue to run on the Pico W. 

## What were you expecting to happen?
I was expecting to be able to interrupt and stop the running program. This functionality is works in Thonny. 

## Any other comments?
- The program ending on it's own or reloading VS Code makes it responsive again until the next run. 
- A few days ago I don't think I was having this issue, so it could have been introduced with a recent update. 
- I specified the python path in the extension settings: C:\Users\"user"\AppData\Local\Programs\Python\Python311\python.exe.
- pyserial was installed automatically by Pico-W-Go successfully. 

## Support info

Copy this from the `Help -> Info/About -> Copy` option in Visual Studio Code:

```
Version: 1.77.3 (user setup)
Commit: 704ed70d4fd1c6bd6342c436f1ede30d1cff4710
Date: 2023-04-12T09:16:02.548Z
Electron: 19.1.11
Chromium: 102.0.5005.196
Node.js: 16.14.2
V8: 10.2.154.26-electron.0
OS: Windows_NT x64 10.0.19042
Sandboxed: Yes
```

It's very possible I've done something wrong, I'm fairly new at this.

jordantryon, 2023-04-14:
Looks like this may be a previous issue that was supposed to be fixed in 3.0.3:
"Fixes [#58](https://github.com/paulober/Pico-W-Go/issues/58), Run button no stop option"

It is still not working for me.

paulober, 2023-04-14:
I could not replicate your issue with v3.0.3:
https://user-images.githubusercontent.com/44974737/232073341-c0d08db4-7d0f-493b-ad37-32726d627508.mp4

closed 2023-04-14 by paulober

jordantryon, 2023-04-15:
Thanks for taking the time to test that on your end! After some further debugging, I think what is actually happening is REPL hangs when I run a python file that doesn't have any print statements in it. 

This allows me to stop after running:
```
import time

print('start')

while(True):
    time.sleep(1)
```
This hangs after running:
```
import time

while(True):
    time.sleep(1)
```

Please see this video of the behavior: https://user-images.githubusercontent.com/22991842/232173068-15191b55-4bbe-4ffd-b03d-ef04193daad9.mov

rbrooklyn, 2023-05-10:
Confirming the print issue.

No print statement means the run button is stuck there, the pico is unresponsive, the reset button does nothing.

Adding just one print statement at the top of my code fixes all of this.