paulober/MicroPico #69: No terminal profile provider registered for id "picowgo.vrepl"

nedamaleki, 2023-04-14:
## What are the steps to reproduce this issue?
1.
2.
3.

## What happens?

## What were you expecting to happen?

## Any logs, error output, etc?
*(To get extension logs see Terminal > Output > Protcoll (Extensionhost) and copy output concerning picowgo)*
*(If it’s long, please paste to https://gist.github.com and insert the link here)*

## Any other comments?

## Support info

Copy this from the `Help -> Info/About -> Copy` option in Visual Studio Code:

```
< Paste Here >
```

nedamaleki, 2023-04-14:
When I need to upload my project to PicoW, this error pumps up. I removed and reinstalled both PicoW-Go extension and Visual Studio and restart my Mac but still gets error.

paulober, 2023-04-14:
That could be because you have not run "> Pico-W-Go > Connect" command or opened a project with a .picowgo file before to activate the extension and connect to a Pico.  I've currently not registered my extension for activation by the terminal profile. Maybe I find a way to hide it before you connect to a Pico or for a future update an option to use it without a workspace but that would need some more work to be done...

paulober, 2023-06-03:
@nedamaleki Does the issue with version v3.0.9 still exist?
I tested the extension in Vscode 1.78.2 with Python 3.11 on a new Apple Silicon Mac with macOS 13.4 and had no issues.

paulober, 2023-10-16:
@nedamaleki Can you confirm that the issue is solved now?

closed 2024-11-18 by paulober