paulober/MicroPico #70: All the commands related to Pico-W-Go in command palette do not working and shows error message.

nedamaleki, 2023-04-14:
## What are the steps to reproduce this issue?
1. Setting -> Command Palette -> Pico-W-Go > Configure project 
2. Setting -> Command Palette -> Pico-W-Go > Connect
3. Setting -> Command Palette -> Pico-W-Go > Global settings 
4. ... 

## What happens?
When I want to run the command, it shows an error message box and says Command 'Pico-W-Go > Connect' resulted in an error (command not found)

## What were you expecting to happen?
I expect the commands work since I installed Pico-W-Go as an extension in VSCode. I connected and disconnected PicoW and reinstalled VSCode and Pico-W-Go. Also installed the Python version higher than 3.9 and put it in the PATH: export PATH="/Library/Frameworks/Python.framework/Versions/3.9.13/bin:$PATH"

## Any logs, error output, etc?
[image]

paulober, 2023-04-14:
Do you have any error output in the output panel under "Extension Host"?

nedamaleki, 2023-04-14:
No, no error. But the good news is that finally after 6 hours and trying with an Ubuntu machine, I figured out the latest version of Pico-W-Go (version3) does not work with the VSCode version (any versions I tried. The current one is the latest one 1.77.3). So I downgraded the Pico-W-Go version to version 2. I could execute the command Configure project and it created .vscode and .picowgo config files. :)

audira77, 2023-04-15:
I have the same issue on Windows 10. Just starting with Pico W and VS Code. Today installed all latest versions that were offered: Pyhton 3.11, latest VS Code and I am getting this error: 
[image]

audira77, 2023-04-15:
P.S. Downgrade from 3 to 2.1.8 solved the problem.

hvanschaick, 2023-04-15:
Having the same problem on MacOS. Pico W now defunc. It consistently proposes to Auto install pyserial. Which fails with the message check python path. pySerial is installed when I load python3 and 'import serial' also works.

enzopicas, 2023-04-16:
Hi, same issue for me with the latest version of vscode python and pico-w-go.
It works fine with 2.8.1

krhnh63, 2023-04-16:
Yep. Pretty much the same thing here on win 10. Was working fine, started vscode yesterday and much the same errors as above. Downgraded to 2.1.8 as suggested above and is now working again. Great extension. Love it but looks like something isn't playing well under the hood with the new rev.

bjoernh, 2023-04-17:
same problem under macOS

martinowitsch, 2023-04-30:
Same issue here. Nothing is working, cant configure project or upload anything without any error message. Did a complete reinstall of visual studio code but still got that issue. I downgraded to 2.1.8 and it seems to work properly again.

zopp99, 2023-04-30:
Same problem on Raspberry Pi4 (amd64).  Running version 1.76 of VSCODE, which is the latest supported by Raspberry.  Downgrade to 2.1.8 works.  (And thank you for a for a very useful tool)

breezin320, 2023-05-05:
Same issue as others.  Just finished all new installation of VSCode, Python 311, and Python-w-go, etc.  None of the pico-w-go commands are found, as in Adira77 screen grab.  Pico board shows as disconnected, and won't connect.  The flash the LED test script shows Import "machine" could not be resolved.  When I did a pip install machine, system flagged VStudio C code tools not installed, so installed all of those.  

Was all working okay in Thonny, twenty minutes before, but wanted to migrate to VSCode for the better tools, etc....very new to this environment, so any help would be great!
thanks!!
dave

hvanschaick, 2023-05-05:
Use the 2.x release for now. Look at the installed extensions and select an earlier release.

paulober, 2023-05-05:
@breezin320 Could it be that you have multiple versions of Python installed on your system?

paulober, 2023-05-05:
@nedamaleki The issue seems to be cause by Python 3.9. I fixed this compatibility issue with Patch v3.0.8. Does this solve the extension not activating issue with Python 3.9 and the extension V3?

breezin320, 2023-05-05:
Thanks, all, for the comments and advice!  I dropped back to version Pico-w-go v2.1.8.  That enabled the Pico-w-go commands to work, the Pico W connected to the PC, and am able to now run the flashing LED program on the pico!  awesome!

I probably do have multiple versions of Python on the PC, but I did put the path to 3.11 everywhere that appeared...not sure I got them all...I'm pretty new at all this

Paul, I did see your early morning update this morning, and reloaded v3.0.8, b4 I went to  v2.1.8, and had the same negative result.  Now that I know all is working, I can try it again, if that'll help :-)

Just some other comments....RUN on the top of the window didn't work.  Run at the "bottom of the window" next to the Pico W connected message, did work, allowing the program to run.

I was unable to install "machine".  But after dropping back to v2.1.8, and activating the Configure Project, the flag on missing "machine" went away.  i was still unable to pip install machine, as it threw up trying to build pycrypto.  I tried pycryptodome and pycryptodomex (which did install okay), but nothing worked....evidently something happens with PWG v2.1.8 that made it happy regarding "machine", even though it appears to be unable to be installed thru pip.  

Thanks, Paul, for the great work....!  Much appreciated

breezin320, 2023-05-05:
Hi, Paul.  So I just tried installing and running v3.0.8.  While the extension "Connect" did run, it did not connect to the Pico W board.  When I dropped back to v2.1.8,  the board immediately connected, and the LED program ran.  Just another data point....

thanks for your help!

breezin320, 2023-05-05:
Hi, Paul.got it on the run button.  

I am using Windows 11Pro.....In settings.json workspace, I do have: "picowgo.openOnStart": true, but don't see autoconnect in the list.  However, in PWG global settings, I see both Open-on-Start checked, and Auto-connect checked.  This is v2.1.8.

I grabbed the last bunch of stuff in the Output>Extension Host window (not sure what I'm looking at):

2023-05-05 09:52:59.998 [info] ExtensionService#_doActivateExtension vscode.debug-auto-launch, startup: false, activationEvent: 'onStartupFinished'
2023-05-05 09:53:00.001 [info] ExtensionService#_doActivateExtension vscode.merge-conflict, startup: false, activationEvent: 'onStartupFinished'
2023-05-05 09:53:00.006 [info] ExtensionService#_doActivateExtension ms-vscode-remote.remote-wsl-recommender, startup: false, activationEvent: 'onStartupFinished'
2023-05-05 09:58:42.928 [info] Extension host terminating: renderer closed the MessagePort
2023-05-05 09:58:42.942 [info] Extension host with pid 5188 exiting with code 0
2023-05-05 09:58:43.673 [info] Extension host with pid 6820 started
2023-05-05 09:58:43.673 [info] Skipping acquiring lock for c:\Users\dbrab\AppData\Roaming\Code\User\workspaceStorage\419a21128c416f8933ec89b706e57169.
2023-05-05 09:58:43.726 [info] ExtensionService#_doActivateExtension vscode.emmet, startup: false, activationEvent: 'onLanguage'
2023-05-05 09:58:43.732 [info] ExtensionService#_doActivateExtension ms-python.python, startup: false, activationEvent: 'onLanguage:python'
2023-05-05 09:58:43.779 [info] ExtensionService#_doActivateExtension VisualStudioExptTeam.intellicode-api-usage-examples, startup: false, activationEvent: 'onLanguage:python'
2023-05-05 09:58:43.815 [info] ExtensionService#_doActivateExtension VisualStudioExptTeam.vscodeintellicode, startup: false, activationEvent: 'onLanguage:python'
2023-05-05 09:58:43.851 [error] ProxyResolver#getCaCertificates error Error: Cannot find module 'vscode-windows-ca-certs'
Require stack:
- c:\Users\dbrab\AppData\Local\Programs\Microsoft VS Code\resources\app\node_modules.asar\@vscode\proxy-agent\out\index.js
- c:\Users\dbrab\AppData\Local\Programs\Microsoft VS Code\resources\app\out\bootstrap-amd.js
- c:\Users\dbrab\AppData\Local\Programs\Microsoft VS Code\resources\app\out\bootstrap-fork.js
- 
	at Module._resolveFilename (node:internal/modules/cjs/loader:1010:15)
	at s._resolveFilename (node:electron/js2c/utility_init:2:2740)
	at Module._load (node:internal/modules/cjs/loader:858:27)
	at f._load (node:electron/js2c/asar_bundle:2:13330)
	at b._load (c:\Users\dbrab\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:126:29948)
	at h._load (c:\Users\dbrab\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:126:26698)
	at I._load (c:\Users\dbrab\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:90:24489)
	at Module.require (node:internal/modules/cjs/loader:1082:19)
	at g (c:\Users\dbrab\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\loader.js:4:647)
	at c:\Users\dbrab\AppData\Local\Programs\Microsoft VS Code\resources\app\node_modules.asar\@vscode\proxy-agent\out\index.js:380:71
	at runNextTicks (node:internal/process/task_queues:61:5)
	at listOnTimeout (node:internal/timers:528:9)
	at process.processTimers (node:internal/timers:502:7)
2023-05-05 09:58:43.917 [info] ExtensionService#_doActivateExtension vscode.github-authentication, startup: false, activationEvent: 'onAuthenticationRequest:github'
2023-05-05 09:58:43.990 [info] ExtensionService#_doActivateExtension ms-python.vscode-pylance, startup: false, activationEvent: 'onLanguage:python'
2023-05-05 09:58:44.079 [info] ExtensionService#_doActivateExtension vscode.microsoft-authentication, startup: false, activationEvent: 'onAuthenticationRequest:microsoft'
2023-05-05 09:58:44.712 [info] ExtensionService#_doActivateExtension vscode.git-base, startup: true, activationEvent: '*', root cause: vscode.git
2023-05-05 09:58:44.715 [info] ExtensionService#_doActivateExtension paulober.pico-w-go, startup: true, activationEvent: 'workspaceContains:.picowgo'
2023-05-05 09:58:45.145 [info] ExtensionService#_doActivateExtension vscode.git, startup: true, activationEvent: '*'
2023-05-05 09:58:45.160 [info] ExtensionService#_doActivateExtension vscode.github, startup: true, activationEvent: '*'
2023-05-05 09:58:45.482 [info] Eager extensions activated
2023-05-05 09:58:45.553 [info] ExtensionService#_doActivateExtension vscode.debug-auto-launch, startup: false, activationEvent: 'onStartupFinished'
2023-05-05 09:58:45.557 [info] ExtensionService#_doActivateExtension vscode.merge-conflict, startup: false, activationEvent: 'onStartupFinished'
2023-05-05 09:58:45.563 [info] ExtensionService#_doActivateExtension ms-vscode-remote.remote-wsl-recommender, startup: false, activationEvent: 'onStartupFinished'
2023-05-05 10:09:17.662 [info] ExtensionService#_doActivateExtension vscode.json-language-features, startup: false, activationEvent: 'onLanguage:jsonc'
2023-05-05 10:09:17.679 [info] ExtensionService#_doActivateExtension vscode.typescript-language-features, startup: false, activationEvent: 'onLanguage:jsonc'
2023-05-05 10:09:22.918 [info] ExtensionService#_doActivateExtension vscode.extension-editing, startup: false, activationEvent: 'onLanguage:json'
[... cut ...] anguage:json'
2023-05-05 10:09:23.635 [info] ExtensionService#_doActivateExtension vscode.markdown-language-features, startup: false, activationEvent: 'onLanguage:markdown'
2023-05-05 10:09:23.881 [info] ExtensionService#_doActivateExtension vscode.markdown-math, startup: false, activationEvent: 'api', root cause: vscode.markdown-language-features
2023-05-05 10:09:28.754 [info] ExtensionService#_doActivateExtension vscode.simple-browser, startup: false, activationEvent: 'onOpenExternalUri:https'

breezin320, 2023-05-05:
Here's the rest (earlier) stuff from that window:

2023-05-05 09:52:48.956 [info] Extension host with pid 5188 started
2023-05-05 09:52:48.956 [info] Skipping acquiring lock for c:\Users\dbrab\AppData\Roaming\Code\User\workspaceStorage\419a21128c416f8933ec89b706e57169.
2023-05-05 09:52:49.063 [info] ExtensionService#_doActivateExtension vscode.emmet, startup: false, activationEvent: 'onLanguage'
2023-05-05 09:52:49.070 [info] ExtensionService#_doActivateExtension ms-python.python, startup: false, activationEvent: 'onLanguage:python'
2023-05-05 09:52:49.120 [info] ExtensionService#_doActivateExtension VisualStudioExptTeam.intellicode-api-usage-examples, startup: false, activationEvent: 'onLanguage:python'
2023-05-05 09:52:49.153 [info] ExtensionService#_doActivateExtension VisualStudioExptTeam.vscodeintellicode, startup: false, activationEvent: 'onLanguage:python'
2023-05-05 09:52:49.191 [error] ProxyResolver#getCaCertificates error Error: Cannot find module 'vscode-windows-ca-certs'
Require stack:
- c:\Users\dbrab\AppData\Local\Programs\Microsoft VS Code\resources\app\node_modules.asar\@vscode\proxy-agent\out\index.js
- c:\Users\dbrab\AppData\Local\Programs\Microsoft VS Code\resources\app\out\bootstrap-amd.js
- c:\Users\dbrab\AppData\Local\Programs\Microsoft VS Code\resources\app\out\bootstrap-fork.js
- 
	at Module._resolveFilename (node:internal/modules/cjs/loader:1010:15)
	at s._resolveFilename (node:electron/js2c/utility_init:2:2740)
	at Module._load (node:internal/modules/cjs/loader:858:27)
	at f._load (node:electron/js2c/asar_bundle:2:13330)
	at b._load (c:\Users\dbrab\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:126:29948)
	at h._load (c:\Users\dbrab\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:126:26698)
	at I._load (c:\Users\dbrab\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\workbench\api\node\extensionHostProcess.js:90:24489)
	at Module.require (node:internal/modules/cjs/loader:1082:19)
	at g (c:\Users\dbrab\AppData\Local\Programs\Microsoft VS Code\resources\app\out\vs\loader.js:4:647)
	at c:\Users\dbrab\AppData\Local\Programs\Microsoft VS Code\resources\app\node_modules.asar\@vscode\proxy-agent\out\index.js:380:71
	at runNextTicks (node:internal/process/task_queues:61:5)
	at listOnTimeout (node:internal/timers:528:9)
	at process.processTimers (node:internal/timers:502:7)
2023-05-05 09:52:49.367 [info] ExtensionService#_doActivateExtension ms-python.vscode-pylance, startup: false, activationEvent: 'onLanguage:python'
2023-05-05 09:52:49.454 [info] ExtensionService#_doActivateExtension vscode.github-authentication, startup: false, activationEvent: 'onAuthenticationRequest:github'
2023-05-05 09:52:49.562 [info] ExtensionService#_doActivateExtension vscode.microsoft-authentication, startup: false, activationEvent: 'onAuthenticationRequest:microsoft'
2023-05-05 09:52:49.995 [info] ExtensionService#_doActivateExtension vscode.git-base, startup: true, activationEvent: '*', root cause: vscode.git
2023-05-05 09:52:49.998 [info] ExtensionService#_doActivateExtension paulober.pico-w-go, startup: true, activationEvent: 'workspaceContains:.picowgo'
2023-05-05 09:52:50.010 [info] ExtensionService#_doActivateExtension vscode.git, startup: true, activationEvent: '*'
2023-05-05 09:52:50.024 [info] ExtensionService#_doActivateExtension vscode.github, startup: true, activationEvent: '*'
2023-05-05 09:52:50.042 [info] ExtensionService#_doActivateExtension vscode.json-language-features, startup: false, activationEvent: 'onLanguage:jsonc'
2023-05-05 09:52:50.057 [info] ExtensionService#_doActivateExtension vscode.typescript-language-features, startup: false, activationEvent: 'onLanguage:jsonc'
2023-05-05 09:52:50.106 [error] Error: spawn C:\Users\dbrab\AppData\Local\Programs\Python\Python311 ENOENT
	at ChildProcess._handle.onexit (node:internal/child_process:283:19)
	at onErrorNT (node:internal/child_process:478:16)
	at process.processTicksAndRejections (node:internal/process/task_queues:83:21)

paulober, 2023-06-01:
@breezin320 The output does not help has you say its from 2.1.8 which works for you. What would help if you try 3.0.8 and then paste the output (better would be in a gist and you link it here).

KilinW, 2023-06-08:
Having similar issue here. 
Machine is Ubuntu 22.04
Vscode version is 1.78.2
Python is 2023.8.0

I downgrade Pico-W-Go to 2.1.8 and it works.

paulober, 2023-06-08:
@KilinW
- What version of the extension have you installed?
- 2023.8.0 is not a valid Python version, maybe that causes the issue
- The information that a downgrade works, does not help to solve the issue at all
- What platform are you on?

augustbering, 2023-09-14:
I just had the same issue on linux, latest vscode (1.82.1) version and latest (3.2.5) MicroPico. Needed to downgrade to 2.1.8 to make it work

paulober, 2023-10-16:
@augustbering Do you still have the issue after `v3.3.2`?

paulober, 2023-10-16:
@nedamaleki Can you confirm the issue is solved?

nedamaleki, 2023-10-16:
Hi Paul, @paulober 
Sorry I was not notified by your message on May and now I got a notification from you. 
As you now the issue has been solved using version 2. However, I have not tried the pico-go-w version 3 + Python 3.9 to check if it works or not. I will check it asap and let you know.  
/Neda

augustbering, 2023-10-18:
> @augustbering Do you still have the issue after `v3.3.2`?

Yes, sorry, same problem on 3.3.3

paulober, 2023-10-18:
@augustbering What arch do you have?

augustbering, 2023-10-19:
> @augustbering What arch do you have?

I run Linux Mint 20.3 if that's what you mean

paulober, 2023-10-19:
> > @augustbering What arch do you have?
> 
> I run Linux Mint 20.3 if that's what you mean

No I mean x86_64, arm64 or armhf

augustbering, 2023-10-19:
ah, x86_64

paulober, 2024-11-18:
@nedamaleki @augustbering Hi, just want to check if this is still an issue for you with v4 of the extension?

paulober, 2026-09-14:
I'm closing this as I couldn't get the details needed to look into it, and a lot has changed since. If it still happens with the latest version, please open a new issue using the bug form. Thanks!

closed 2026-09-14 by paulober