paulober/MicroPico #73: Unplugging USB crashes extension

crimsonbytes, 2023-04-17:
## What are the steps to reproduce this issue?
1. Configure for Pico-W-Go
2. Run a program
3. Unplug USB

## What happens?
Terminal window prints !!ERR!! and you need to restart VS Code to regain control of the pico as none of the buttons on the status bar work any longer, including soft and hard resets or disconnecting, reconnecting.

## What were you expecting to happen?
My stuck program would be reset and at worst the plugin would show the Pico was no longer connected. 

## Any logs, error output, etc?
*(To get extension logs see Terminal > Output > Protcoll (Extensionhost) and copy output concerning picowgo)*
*(If it’s long, please paste to https://gist.github.com and insert the link here)*
2023-04-16 22:23:41.654 [info] Extension host with pid 35196 started
2023-04-16 22:23:41.654 [info] Skipping acquiring lock for c:\Users\flash\AppData\Roaming\Code\User\workspaceStorage\33e8b0954ec9b535e1a864150cd0311b.
2023-04-16 22:23:41.865 [info] ExtensionService#_doActivateExtension ms-python.python, startup: false, activationEvent: 'onLanguage:python'
2023-04-16 22:23:42.078 [info] ExtensionService#_doActivateExtension ms-toolsai.jupyter, startup: false, activationEvent: 'onLanguage:python'
2023-04-16 22:23:42.231 [info] ExtensionService#_doActivateExtension VisualStudioExptTeam.intellicode-api-usage-examples, startup: false, activationEvent: 'onLanguage:python'
2023-04-16 22:23:42.269 [info] ExtensionService#_doActivateExtension VisualStudioExptTeam.vscodeintellicode, startup: false, activationEvent: 'onLanguage:python'
2023-04-16 22:23:42.372 [info] ExtensionService#_doActivateExtension vscode.emmet, startup: false, activationEvent: 'onLanguage'
2023-04-16 22:23:42.719 [info] ExtensionService#_doActivateExtension ms-toolsai.jupyter-renderers, startup: false, activationEvent: 'api', root cause: ms-toolsai.jupyter
2023-04-16 22:23:42.729 [info] ExtensionService#_doActivateExtension vscode.github-authentication, startup: false, activationEvent: 'onAuthenticationRequest:github'
2023-04-16 22:23:42.790 [info] ExtensionService#_doActivateExtension ms-python.vscode-pylance, startup: false, activationEvent: 'onLanguage:python'
2023-04-16 22:23:42.916 [info] ExtensionService#_doActivateExtension vscode.git-base, startup: true, activationEvent: '*', root cause: vscode.git
2023-04-16 22:23:42.919 [info] ExtensionService#_doActivateExtension ecmel.vscode-html-css, startup: true, activationEvent: '*'
2023-04-16 22:23:42.952 [info] ExtensionService#_doActivateExtension formulahendry.auto-rename-tag, startup: true, activationEvent: '*'
2023-04-16 22:23:42.959 [info] ExtensionService#_doActivateExtension Zignd.html-css-class-completion, startup: true, activationEvent: '*'
2023-04-16 22:23:43.050 [info] ExtensionService#_doActivateExtension vscode.microsoft-authentication, startup: false, activationEvent: 'onAuthenticationRequest:microsoft'
2023-04-16 22:23:43.159 [info] ExtensionService#_doActivateExtension vscode.git, startup: true, activationEvent: '*'
2023-04-16 22:23:43.220 [info] ExtensionService#_doActivateExtension vscode.github, startup: true, activationEvent: '*'
2023-04-16 22:23:43.314 [info] Eager extensions activated
2023-04-16 22:23:43.333 [info] ExtensionService#_doActivateExtension vscode.debug-auto-launch, startup: false, activationEvent: 'onStartupFinished'
2023-04-16 22:23:43.336 [info] ExtensionService#_doActivateExtension vscode.merge-conflict, startup: false, activationEvent: 'onStartupFinished'
2023-04-16 22:23:43.341 [info] ExtensionService#_doActivateExtension ms-vscode-remote.remote-wsl-recommender, startup: false, activationEvent: 'onStartupFinished'
2023-04-16 22:23:43.364 [info] ExtensionService#_doActivateExtension dbaeumer.vscode-eslint, startup: false, activationEvent: 'onStartupFinished'
2023-04-16 22:23:43.370 [info] ExtensionService#_doActivateExtension eamodio.gitlens, startup: false, activationEvent: 'onStartupFinished'
2023-04-16 22:23:43.396 [info] ExtensionService#_doActivateExtension vscode.npm, startup: false, activationEvent: 'onStartupFinished', root cause: esbenp.prettier-vscode
2023-04-16 22:23:43.417 [info] ExtensionService#_doActivateExtension ms-vscode-remote.remote-containers, startup: false, activationEvent: 'onStartupFinished'
2023-04-16 22:23:43.442 [info] ExtensionService#_doActivateExtension ms-vscode-remote.remote-wsl, startup: false, activationEvent: 'onStartupFinished'
2023-04-16 22:23:43.450 [info] ExtensionService#_doActivateExtension vscode-icons-team.vscode-icons, startup: false, activationEvent: 'onStartupFinished'
2023-04-16 22:23:44.333 [info] ExtensionService#_doActivateExtension esbenp.prettier-vscode, startup: false, activationEvent: 'onStartupFinished'
2023-04-16 22:23:44.787 [info] ExtensionService#_doActivateExtension vscode.configuration-editing, startup: false, activationEvent: 'onLanguage:jsonc'
2023-04-16 22:23:44.795 [info] ExtensionService#_doActivateExtension vscode.json-language-features, startup: false, activationEvent: 'onLanguage:jsonc'
2023-04-16 22:23:44.807 [info] ExtensionService#_doActivateExtension vscode.typescript-language-features, startup: false, activationEvent: 'onLanguage:jsonc'
2023-04-16 22:24:17.373 [info] ExtensionService#_doActivateExtension paulober.pico-w-go, startup: false, activationEvent: 'onCommand:picowgo.initialise'
2023-04-16 22:24:20.490 [error] Error: EISDIR: illegal operation on a directory, symlink 'C:\Users\flash\AppData\Roaming\Code\User\Pico-W-Stub' -> 'g:\My Drive\whitestone\.vscode\Pico-W-Stub'

## Any other comments?

## Support info

Copy this from the `Help -> Info/About -> Copy` option in Visual Studio Code:

```
Version: 1.77.3 (system setup)
Commit: 704ed70d4fd1c6bd6342c436f1ede30d1cff4710
Date: 2023-04-12T09:16:02.548Z
Electron: 19.1.11
Chromium: 102.0.5005.196
Node.js: 16.14.2
V8: 10.2.154.26-electron.0
OS: Windows_NT x64 10.0.22621
Sandboxed: Yes
```

paulober, 2023-04-18:
@crimsonbytes Could confirm your issue but the error in the logs you've posted refer to a different issue.

Did you just want to report this bug or do you have a usecase that needs such behavior of the extension?
- Start a script
- Unplug the pico
- Plug it back in 
- And continue working
-

crimsonbytes, 2023-04-22:
I'm not sure what the different issue logged is showing then, I posted the
logs when it behaved as I described.

In any case, I wanted to report a bug/use case where:
- Start a script
- Pi crashes (in my current project I'm controlling a DC motor, so a crash
means that it keeps running)
- Recover gracefully
- Continue working

This may mean:
- Unplug because soft/hard resets do not work
- Plug it back in
- Reconnect to pi

At the moment this means:
- Unplug because soft/hard resets do not work
- Plug it back in
- Attempt to reconnect
- Restart VS Code and reconfigure project for Pico plugin (due to no
response to reconnects)

If you'd like, I can share the code with you via dm.

paulober, 2023-04-22:
Thank you for providing more details on this issue and the usecase it blocks. I'll like to fix this, but for now I'm unfortunately to bussy but I come back to this issue once i have more time to work on the project. This would mean a delay of about 30 days. I'm you can still use the Extension as it is now with some workarounds need.

paulober, 2023-05-29:
@crimsonbytes Does the issue still exits with the most recent version of the extension?

commit 1891dd55 (2023-06-02): Fixes #73, Extension freezes on Pico disconnection during code execution
PR #98 (2023-06-02): Patch v3.0.9
### Added
- Status Bar button `togglepicowfs` (Suggested in #95)

### Changed
- Fixed #73, Extension freezes when Pico is disconnected during code execution
- Upgrade to [pyboard-serial-com v1.4.26](https://github.com/paulober/pyboard-serial-com/releases/tag/v1.4.26)
commit 9bcd9d65 (2023-06-02): Fix #73, Extension freezes on Pico disconnection during code execution
closed 2023-06-02 by paulober