paulober/MicroPico #79: Configure Project cannot complete after install on WIndows VS Code

nigelkennington, 2023-04-24:
## What are the steps to reproduce this issue?
1. Install Extension into VS code on windows
2. Attempt to use the pico-w-go > configure project command

## What happens?
Error dialog: "Command Pico-W-Go > Configure Project resulted in an error . Command picowgo.initialise not found"

## What were you expecting to happen?
Not that :-/

## Any logs, error output, etc?
*(To get extension logs see Terminal > Output > Protcoll (Extensionhost) and copy output concerning picowgo)*
*(If it’s long, please paste to https://gist.github.com and insert the link here)*
There is no Output option in the Terminal.

## Any other comments?

## Support info

Copy this from the `Help -> Info/About -> Copy` option in Visual Studio Code:

```
Version: 1.77.3 (user setup)
Commit: 704ed70d4fd1c6bd6342c436f1ede30d1cff4710
Date: 2023-04-12T09:16:02.548Z
Electron: 19.1.11
Chromium: 102.0.5005.196
Node.js: 16.14.2
V8: 10.2.154.26-electron.0
OS: Windows_NT x64 10.0.19044
Sandboxed: Yes
```

paulober, 2023-04-26:
What python installation do you have? Is it user install (so installed in AppData)? Do you get an error notification in vscode?

nigelkennington, 2023-04-29:
Python is installed in "D:\Program Files\Python\Python311", path includes "D:\Program Files\Python\Python311\Scripts\;D:\Program Files\Python\Python311\;" There's no error notification other than "Command picowgo.initialise not found" when the command is run.

paulober, 2023-05-05:
What output do you get when executing following commands in PowerShell:
```powershell
cd $env:USERPROFILE\.vscode\extensions\paulober.pico-w-go-3.0.8-win32-x64\dist\scripts
python.exe .\wrapper.py --scan-ports
```

paulober, 2023-05-29:
@nigelkennington What is your response to my last comment?
Does the issue still persists?

Artem-Sotnikov, 2023-06-02:
I have managed to reproduce this issue by running into it during an attempt to set up the extension on my computer. My python is installed in AppData and vscode seems to have the capability of talking to it ("Run Python File in Terminal" command in vscode successfully executes a 'hello world' script). The vscode information is:

```
Version: 1.78.2 (user setup)  
Commit: b3e4e68a0bc097f0ae7907b217c1119af9e03435
Date: 2023-05-10T14:39:26.248Z
Electron: 22.5.2
Chromium: 108.0.5359.215
Node.js: 16.17.1
V8: 10.8.168.25-electron.0
OS: Windows_NT x64 10.0.19045
Sandboxed: Yes
```

I have navigated to the vscode folder `.vscode\extensions\paulober.pico-w-go-3.0.8-win32-x64\dist\scripts` in powershell and ran the wrapper.py --scan-ports command. The resulting output was:

```
COM5
!!EOO!!
```

Let me know what other steps I can take at this point. I'd really rather not use Thonny for development with the pico and I'm keen to try and get this to work.

Thanks!

paulober, 2023-06-03:
@Artem-Sotnikov Sorry to hear you are having the same issue. I can totally understand that you don't want to use the other "IDE" and I appreciate your help to fix this issue.

So first of all:
- How do you get the issue? When running pico-w-go > Configure Project in an empty folder?
- What output do you get in Output pane > "Extension Host"?

Artem-Sotnikov, 2023-06-05:
To respond to your first point:

1. Last week, I had tried to run the command on both an empty folder and one that had a sub-folder within it; neither worked

However, the plot thickened as I tried to come up with an answer to point 2. After getting back to my desk this week, I set about trying to debug the issue in greater detail by:
- Changing version of python (utilizing anaconda to create new envs with python 3.9, 3.10 and then setting the pico-w-go python path to the env folder with that python version)
- Changing the extension settings and restarting vscode (Namely, Open on Start and Sync All file Types)
- Trying to run the command on a local folder (I was originally running it on a synced folder similar to OneDrive)

None of this seemed to help, although almost immediately I stopped getting the error and the command simply did nothing (no files are added to the folder or any other action). Then suddenly, it started working as intended. 

I have been able to reproduce the 'does nothing' error by adding quotations around the python path setting, which is what 
I had originally and leads to the following output in ExtensionHost (which I can reproduce reliably):
 
```
spawn "C:\Users\AUSERNAME\AppData\Local\Programs\Python\Python311\python.exe" ENOENT
	at ChildProcess._handle.onexit (node:internal/child_process:285:19)
	at onErrorNT (node:internal/child_process:497:16)
	at process.processTicksAndRejections (node:internal/process/task_queues:83:21)
```

However, I have not been able to reproduce the original error message despite repeated attempts. I'll be sure to report it if I ever come across it again. However, the advice for anyone else running into the issue would be:
1. Ensure that the python path of pico-w-go does not have quotations around it and is valid -> no errors in extension host
2. Restart VsCode (and maybe the computer)

Beyond that, I'm really not sure.

paulober, 2023-06-20:
@Artem-Sotnikov Thanks for putting all the work into researching the cause of this problem on you machine. 
I can confirm the issue with the wrong python path often results in no error message and no feedback, nothing happens when running configure project command. That's especially bad if you have settings sync and switch between different OS then you have this bug every time you change the python path and switch the OS.

I'm happy to hear that the extension is now working for you. I'll try to find a better way to give the user feedback on why the extension is not running and what they can do to fix it. (or better remove this point of failure)

paulober, 2023-10-16:
@nigelkennington Do you still have the issue after `v3.3.2`?

closed 2023-10-18 by paulober