paulober/MicroPico #84: Incorporating other languages besides Python

NonRidiculousAdjective, 2023-04-28:
# Feature Request - adding © to the mix?

## Is your feature request related to a problem? Please describe.
Hey, I love this extension and I am happy it has been revived by someone. At the moment I shifted from Python for a bit and onto C to learn some generic basics regarding embedded systems and to have faster programs. The thing is that my head hurts from all the banging of trying to compile a single short code and needing to repeat dozens of steps and fix another dozen of errors just to get it on my RPI Pico.

## Describe the solution you'd like
The suggestion that I am getting at here, is whether it would be possible to include other languages besides python to the mix. As I described in [this post](https://forums.raspberrypi.com/viewtopic.php?t=349086) in the RPI Pico forums, it would be amazing for the C process to get more streamlined.

## Describe alternatives you've considered
using cmake, checking for cmake alternatives..

## Teachability, Documentation, Adoption, Migration Strategy
I am not sure how to help with this part, after all I am not that experienced and don't know how VSCode extensions work..

paulober, 2023-05-05:
@NonRidiculousAdjective There might be something in the works 😉

doge-gif, 2023-07-13:
If you're targeting C (or C++), I would have said platformio should be the choice, until I saw the [huge drama](https://github.com/platformio/platform-raspberrypi/pull/36).
TLDR: platformio doesn't want it, but you can use it with [this](https://community.platformio.org/t/raspberry-pi-pico-w-in-platformio/32006/2) config.
The Arduino extension for VSCode should work too.

Both options above use an Arduino library layer.
If you don't want that but just the pico-sdk - I hate being that guy telling "just learn it", but learning CMake might be the right direction.
If using VSCode, I suggest using the following extensions:
```text
twxs.cmake
ms-vscode.cmake-tools
ms-vscode.cpptools
marus25.cortex-debug (needs a debugger, r.g. picoprobe)

xaver.clang-format
or
llvm-vs-code-extensions.vscode-clangd
(clangd has some quirks since pico-sdk doesn't support clang as a compiler, but it's way faster and better than intellisense)
```

Good luck!

NonRidiculousAdjective, 2023-10-10:
> @NonRidiculousAdjective There might be something in the works 😉

Are there any updates, I am too curious.

paulober, 2023-10-10:
@NonRidiculousAdjective It's in its final stage of development before the first public release. I'll keep this issue updated...

paulober, 2024-03-27:
@NonRidiculousAdjective @doge-gif Just launched: https://github.com/raspberrypi/pico-vscode

closed 2024-03-27 by paulober

NonRidiculousAdjective, 2024-04-01:
@paulober Yay, finally!! Thanks for the update!