paulober/MicroPico #85: Auto Connect setting is ignored

ezio416, 2023-04-30:
## What are the steps to reproduce this issue?
1. disable auto connect in settings ui
2. disable in settings json file

## What happens?
Pico-W-Go always says AutoConnect enabled and tries to find devices. When the terminal is killed, it always comes back.

## What were you expecting to happen?
AutoConnect should be disabled when I disable it using either method in settings.

## Any logs, error output, etc?
2023-04-29 23:50:12.455 [info] ExtensionService#_doActivateExtension paulober.pico-w-go, startup: true, activationEvent: 'workspaceContains:.picowgo'

## Any other comments?
N/A

## Support info
```
Version: 1.74.3 (system setup)
Commit: 97dec172d3256f8ca4bfb2143f3f76b503ca0534
Date: 2023-01-09T16:59:02.252Z
Electron: 19.1.8
Chromium: 102.0.5005.167
Node.js: 16.14.2
V8: 10.2.154.15-electron.0
OS: Windows_NT x64 10.0.19045
Sandboxed: No
```

paulober, 2023-05-05:
Which version of the extension are you currently on?

ezio416, 2023-05-05:
My version is the current latest, 2.1.8. After some more restarts and using separate workspace settings, it's finally not auto connecting. However upon launching VSCode the terminal still always opens and says it's searching for devices, even if it doesn't actually connect.

paulober, 2023-05-06:
v2.1.8 isn't the latest. The reason why you don't get the most recent updates is that you use an outdated version of vscode. The current minimum requirement is 1.77.0. The newer versions of the extension should solve you issues.

closed 2023-05-06 by paulober