paulober/MicroPico #86: Pico vREPL not open

RnoldR, 2023-05-04:
## What are the steps to reproduce this issue?
1. Run vscode 1.78.0 on Ubuntu 22.04
2. install latest version of pico-w-go (3.0.7)
3. run the simple micropython blink.py program (or any other program)

## What happens?
Message: Pico vREPL not open.

## What were you expecting to happen?
A running program

## Any logs, error output, etc?
*(To get extension logs see Terminal > Output > Protcoll (Extensionhost) and copy output concerning picowgo)*
*(If it’s long, please paste to https://gist.github.com and insert the link here)*
023-05-04 20:48:40.179 [info] ExtensionService#_doActivateExtension ms-vscode.vscode-serial-monitor, startup: false, activationEvent: 'onView:vscode-serial-monitor.monitor0'
2023-05-04 20:49:49.094 [info] Extension host terminating: renderer closed the MessagePort
2023-05-04 20:49:49.122 [info] Extension host with pid 42057 exiting with code 0
2023-05-04 20:49:49.781 [info] Extension host with pid 45026 started
2023-05-04 20:49:49.781 [info] Skipping acquiring lock for /home/arnold/.config/Code/User/workspaceStorage/98aaee4281510fed3ccbcb20d44a7ed3.
2023-05-04 20:49:49.872 [info] ExtensionService#_doActivateExtension vscode.emmet, startup: false, activationEvent: 'onLanguage'
2023-05-04 20:49:49.877 [info] ExtensionService#_doActivateExtension ms-python.python, startup: false, activationEvent: 'onLanguage:python', root cause: ms-python.isort
2023-05-04 20:49:49.918 [info] ExtensionService#_doActivateExtension ms-toolsai.jupyter, startup: false, activationEvent: 'onLanguage:python'
2023-05-04 20:49:49.953 [info] ExtensionService#_doActivateExtension njpwerner.autodocstring, startup: false, activationEvent: 'onLanguage:python'
2023-05-04 20:49:50.156 [info] ExtensionService#_doActivateExtension ms-toolsai.jupyter-renderers, startup: false, activationEvent: 'api', root cause: ms-toolsai.jupyter
2023-05-04 20:49:50.162 [info] ExtensionService#_doActivateExtension vscode.github-authentication, startup: false, activationEvent: 'onAuthenticationRequest:github'
2023-05-04 20:49:50.222 [info] ExtensionService#_doActivateExtension ms-python.isort, startup: false, activationEvent: 'onLanguage:python'
2023-05-04 20:49:50.229 [info] ExtensionService#_doActivateExtension ms-python.vscode-pylance, startup: false, activationEvent: 'onLanguage:python'
2023-05-04 20:49:50.396 [info] ExtensionService#_doActivateExtension vscode.microsoft-authentication, startup: false, activationEvent: 'onAuthenticationRequest:microsoft'
2023-05-04 20:49:50.457 [info] ExtensionService#_doActivateExtension vscode.git-base, startup: true, activationEvent: '*', root cause: vscode.git
2023-05-04 20:49:50.458 [info] ExtensionService#_doActivateExtension deitry.colcon-helper, startup: true, activationEvent: '*'
2023-05-04 20:49:50.461 [info] ExtensionService#_doActivateExtension paulober.pico-w-go, startup: true, activationEvent: 'workspaceContains:.picowgo'
2023-05-04 20:49:50.488 [info] ExtensionService#_doActivateExtension vscode.git, startup: true, activationEvent: '*'
2023-05-04 20:49:50.500 [info] ExtensionService#_doActivateExtension vscode.github, startup: true, activationEvent: '*'
2023-05-04 20:49:50.529 [info] ExtensionService#_doActivateExtension vscode.json-language-features, startup: false, activationEvent: 'onLanguage:jsonc'
2023-05-04 20:49:50.539 [info] ExtensionService#_doActivateExtension vscode.typescript-language-features, startup: false, activationEvent: 'onLanguage:jsonc'
2023-05-04 20:49:51.128 [info] Eager extensions activated
2023-05-04 20:49:51.130 [info] ExtensionService#_doActivateExtension vscode.debug-auto-launch, startup: false, activationEvent: 'onStartupFinished'
2023-05-04 20:49:51.131 [info] ExtensionService#_doActivateExtension vscode.merge-conflict, startup: false, activationEvent: 'onStartupFinished'
2023-05-04 20:50:23.230 [info] ExtensionService#_doActivateExtension vscode.debug-server-ready, startup: false, activationEvent: 'onDebugResolve'

## Any other comments?
As suggested elsewhere, a lower version should work. That's right: the error is introduced since version 3.0.0, the last version before that (2.1.8) works. Partly.

Partly because 
- when a program is already running
-  vscode is exited
- and is entered again
- another run command is given

another error occurs:
...
Connecting to /dev/ttyACM0...

> Connection timed out. Click the "Pico Disconnected" button to try again.
...

reset, stop disconnect/connect all have no effect other than the message: "Connecting to /dev/ttyACM0..." (which is the correct linux usb port for the Pico W).

The package pyserial is installed. 

## Support info
```
Version: 1.78.0
Commit: 252e5463d60e63238250799aef7375787f68b4ee
Date: 2023-05-03T20:09:46.685Z
Electron: 22.4.8
Chromium: 108.0.5359.215
Node.js: 16.17.1
V8: 10.8.168.25-electron.0
OS: Linux x64 5.19.0-41-generic
Sandboxed: Yes
```

paulober, 2023-05-05:
That's no bug, you have to open the vREPL first by opening a new terminal and the select "Pico-W-Go vREPL" in the new terminal dropdown. This must be done manually, as opening this terminal by code currently has some issues with the vscode api.

closed 2023-05-05 by paulober

RnoldR, 2023-05-07:
Sorry, I missed that. Thanks for your clarification, all is well now.

paulober, 2023-05-07:
Will be fixed in the future when i find a solution to this problem...