paulober/MicroPico #89: Module installed but reporting not found

breezin320, 2023-05-06:
## What are the steps to reproduce this issue?
1. Using pico-w-go in VSCode
2.Pip install untangle (reports installed, and VSCode reports it as installed with path, etc)
3."import untangle as ut" in program

## Program fails to run, due to report of no module named untangle
Traceback (most recent call last):
  File "<stdin>", line 5, in <module>
ImportError: no module named 'untangle'

## What were you expecting to happen?
Program should have run

## Any logs, error output, etc?
Traceback (most recent call last):
  File "<stdin>", line 5, in <module>
ImportError: no module named 'untangle'

## Any other comments?  New to this development environment, so not sure what I'm missing

thanks for any and all help and advice

paulober, 2023-05-07:
@breezin320 If you do pip install on you dev machine it will install this package on you dev machine. So if you then run you code on the Pico the module will not be available to the python script at runtime.

breezin320, 2023-05-07:
thanks, paul!  got that....so how do i run pip (or mip?) on the pico w?  at the >>> prompt, it doesn't know pip....?  appreciate the help, working to understand the relationships between the two environments....

appreciate the help!

paulober, 2023-05-07:
@breezin320 Please read the official [MicroPython Docs](https://docs.micropython.org) if you need help getting started with MicroPython.

">>> " indicates you are in a Python REPL, where you use Python syntax for 'commands'. So `mip` is a custom package manager by MicroPython which allows you to install packages to your MicroPython based board if it has an internet connection. So you can use the `mip` module according to their documentation like so:
```python
>>> import mip
>>> mip.install("pkgname')
```
Also the install functions has some parameters which could be used to define how you would like a module being installed.
Keep in mind that this just installs a certain package to your Pico W but does not show up in VS Code Python auto-completion like your locally installed packages.

closed 2023-05-07 by paulober