paulober/MicroPico #90: wondering what is the expected /supported workflow

mess-maker, 2023-05-07:
# Feature Request 🚀

Thanks for this extension, finally a simple way to have autocomplete on micropython

I've just some concern and feedback

could you describe a bit more what are the expected workflow with this extension?

- do we need to set a venv before / after "configure project" (or maybe a venv is set automagically and hidden from the user?)
- is it nevertheless possible to set a venv ? and if so I think we need to install pyserial in it ? and point to the venv python interpreter in the parameter ?

- what are the purpose of the virtual FS
    -> code directly on the MCU flash ? what is the workflow in that case to restart / update the program
    -> just a convenient way to see / transfert files on /from the MCU ?

- the pyserial package seems to install even if we answer "no" to the auto install query 

finally last feature request, do you plan to integrate with mip (the new micropython pip)

cheers

paulober, 2023-05-07:
@mess-maker 
- No, but you can. The extension just needs a path to a python executable which pip installed. So you could setup a venv and then give the path to this python executable to the extension. It uses `<pythonexecutable> -m pip ...`, so it'll install the package in the venv.
- You can set (for example for Windows) `"picowgo.pythonPath": "<path-to-venv>/python.exe"`. It will then install pyserial for you in this venv like it would do in a global python installation
- The virtual workspace is for a more convinient way to browse the filesystem on the pico:
    - It allows to edit files on the Pico
    - Running these files is faster because it don't have to upload the file every time you want to run it
    - Transferring files via drag and drop
    - See what directory structure is present on the file system without having to use the vREPL
    - Easy delete, rename and move operations without having to use the vREPL
    - Quickly view files on the Pico, so you don't have to download, view and then delete the file again if you just want to browse some files
    
    It was mainly created as replacement for the old virtual FTP server implementation and for those who requested a simple way, like in other GUI tools for developing for the Raspberry Pi Pico, for browsing the remote fs
- In the most recent versions of the extension it should not ask you to install `pyserial` as it is now seen as requirement for the extension and automatically installed on activation if not already present. And in the old version you could not say no but something like "I install it manually" this causes then the extension to deactivate it self until it check for pyserial again after a reactivation

Maybe. I'll put this request on the list for upcoming features. Maybe I can add an installer GUI as a sidebar pane which allows you to install packages. But auto-completion for packages installed by mip is something totally different. If i would autogenerate these they would only auto-complete names (like method, classes and other class members) but they would not contain type annotations.

KernelBypass, 2023-07-02:
@paulober 

I am curious about `.venv` and how it affects `upload project to pico` function.
By default VS Code creates `.venv` folder when setting up venv, and pico-w-go excludes only `venv` (no dot) from uploads. I added `.venv` to exclusion list manually, but it still seems to try to upload `.venv` to pico and then pico loses connection to VS Code  (probably due to flash overflow with `.venv` being over 20MB).

Is `Pico (W) Remote Workspace` the `virtual FS` from OP's question?

Should we therefore avoid `upload project to pico` when using `.venv` and just drag and drop multiple `.py` files of the project to the `Pico (W) Remote Workspace` -- is that the preferred workflow?

[image]

Thanks

paulober, 2023-10-16:
@KernelBypass Sorry that I've overlooked your question. 
No, uploading the `venv` or `.venv` folder to the pico is strongly not recommended as it doesn't make sense, uses a lot of space and slows down the upload process. The MicroPython firmware has it's own custom Python interpreter to run on microcontroller hardware so your generic virtual environment doesn't work on the Pico.