paulober/MicroPico #91: Per project upload exclusions

bellingeri, 2023-05-08:
# Feature Request 🚀

First of all, I really appreciate your work!

It would be nice to be able to configure upload exclusions per project (in addition to the generic ones).
...perhaps adding a dedicated section in the file .picowgo.

paulober, 2023-05-14:
Thanks for the kind words.

Do you mean that you want to exclude diffrent things in multiple picowgo projects within the same workspace?
Currently if you want to overwrite your global exclusion settings, just set the `"picowgo.syncFileTypes"` or `"picowgo.pyIgnore"` in `.vscode/settings.json` for your selected vscode workspace.

bellingeri, 2023-05-15:
This means I don't know how to use workspaces!!
I usually manage projects by folder without creating any workspaces.

I was convinced that the settings were global to the extension... I will try to learn...

Thank you!

paulober, 2023-06-20:
Ok, so you can set the `syncFileTypes` and `pyIgnore` settings also in you project folder in `.vscode/settings.json`.

closed 2023-06-20 by paulober