paulober/MicroPico #92: Monitoring running program

bellingeri, 2023-05-08:
## What are the steps to reproduce this issue?

Here's an issue we've encountered since version 3:
It is no longer possible to see the output of self running programs (`main.py`).

## What happens?

When the extension connects to the Pico, the execution of the program stops.

## What were you expecting to happen?

Previously by running the extension (or Pico-W-Go -> Connect) it was possible to see the output of the program; it was also possible to see the error that caused the program to close.

## Any logs, error output, etc?

n/a

## Any other comments?

n/a

## Support info

Pico-W-Go v3.0.8

Visual Studio Code:
```
Versione: 1.78.0 (system setup)
Commit: 252e5463d60e63238250799aef7375787f68b4ee
Data: 2023-05-03T20:09:00.748Z
Electron: 22.4.8
Chromium: 108.0.5359.215
Node.js: 16.17.1
V8: 10.8.168.25-electron.0
Sistema operativo: Windows_NT x64 10.0.22621
In modalità sandbox: Yes
```

paulober, 2023-05-14:
@bellingeri @coderberry
As a result of the inclusion of new features and minor enhancements in version 3, it became necessary to implement this change. While it may be possible to revert this change at a later time, doing so would require a significant amount of effort and resources.

At present, the only way to observe the functionality of startup scripts is to either run the main.py file on the Pico remotely (if you happen to have it locally and it does not require other python scripts), or to access the remote Pico file system and execute the main.py or boot.py directly on the Pico (after uploading them).

Would you kindly confirm if these workarounds are suitable for your particular use cases?

bellingeri, 2023-05-15:
Thanks for the reply.

In the development of my little scripts, it happens that these work immediately, and then run into problems after a few days.
In my case it happens especially if I use the pico as a web server.

In these cases, to try to identify the problem I always kept the pico connected (and powered) to the pc, even for a few days - then open the project on vscode when I realized that things weren't going as expected.

Now, clearly from your words, this is no longer an option.

What could be the alternatives?!

paulober, 2023-05-18:
@bellingeri Take a look at [listen2pico](https://github.com/paulober/listen2pico) utility for now. It allows you to listen to the MicroPython output without interrupting running programs.

bellingeri, 2023-05-19:
Thanks @paulober.

I will definitely try it on the weekend.
I hope you didn't work just for me!!!

Thanks again.