paulober/MicroPico #96: Support for other chips?

steeley, 2023-05-24:
# Feature Request 🚀
First of all, thanks for this plug-in!
## Is your feature request related to a problem? Please describe.
*A clear and concise description of what the problem is. E.g. I have an issue when...*
Not an issue really, but are you planning to support other MCUs?

## Describe the solution you'd like
*A clear and concise description of what you want to happen. Add any considered drawbacks.*
I would really like this to work with ESP32, (and Teensy for example).

## Describe alternatives you've considered
*A clear and concise description of any alternative solutions or features you've considered.*
Using Thonny maybe.

## Teachability, Documentation, Adoption, Migration Strategy
*If you can, explain how users will be able to use this and possibly write out a version the docs. Maybe a screenshot or design?*
There are many users of ESP32 and this would make the plugin very popular, and maybe more people to help you develop it.

steeley, 2023-06-14:
So I'm guessing thats a 'no' then ... ??

MortenGuldager, 2023-07-02:
I could too have used it for the ESP32's running micropython.

It's kinda odd to lock out other micropythons, like, why??? 

Well, I guess it is back to Pymakr then..

ChrisChoke, 2024-01-11:
i would be interested, too. this extension could be what i searched for.
But i usually work with esp32-wroom and esp32-c3/c6.

@paulober is that soemthing you can realize? that would be nice. 😉 

Chris

paulober, 2024-01-11:
I would like adding support for ESP32 based boards to the extension. I'm already working on it, but I don't have much time at the moment.
I can test on esp32-wroom and esp32-s3. C6 isn't currently supported by Micropython but it seems to be WIP. And C3 would require to be tested by the community.

paulober, 2024-01-11:
> I could too have used it for the ESP32's running micropython.
> 
> It's kinda odd to lock out other micropythons, like, why???
> 
> Well, I guess it is back to Pymakr then..

Cause the extension needs to be adjusted to work with other micropython board and I didn't have the time and devices to do so. But I'm working on it.

ChrisChoke, 2024-01-12:
That's sounds pretty good I think.
You're right. Crashes have to be fixed before releasing new features.

I can help testing for the esp32-c3. I have two mini form factor of this devices and a bunch of esp32 wroom. The Micropython c6 support got through to me. I haven't saw it 😃

If you need help testing or something else. Feel free to address. I am not JavaScript/typescript hero so helping by coding would be difficult 😃

Chris

paulober, 2024-01-18:
@ChrisChoke @MortenGuldager ESP32-WROOM-32 experimental support with [v3.7.0](https://github.com/paulober/MicroPico/releases/tag/v3.7.0).

paulober, 2024-01-18:
@steeley I don't have access to a Teensy board but I could add detection for it if you could tell me what product and vendor id the board shows on your os after flashing MicroPython on it?

ChrisChoke, 2024-01-19:
Yeah nice, thx I will test it today in the evening.

My esp32-c6 boards vendor and product id is ```303a:1001```
Espressif USB JTAG/serial debug unit

I can use a baudrate of 115200 as usual.

Chris

ChrisChoke, 2024-01-20:
oh my god. @paulober thats so nice. v3.7.1 works like a charm with esp32-wroom. this will make work so much easier. i love it already.

looking forward to interact with esp32-c3, too 😊

so my first feedback to you is just: THANK YOU!

Chris

steeley, 2024-01-20:
> @steeley I don't have access to a Teensy board but I could add detection for it if you could tell me what product and vendor id the board shows on your os after flashing MicroPython on it?

Micropython supports Teensy 4.0 and 4.1 - I have a Teensy 4.0 here and it shows this:

Teensy 4.0:

  Product ID:	0x9802
  Vendor ID:	0xf055
  Version:	1.00
  Serial Number:	6929F865D269051D
  Speed:	Up to 480 Mb/s
  Manufacturer:	MicroPython
  Location ID:	0x14320000 / 5
  Current Available (mA):	500
  Current Required (mA):	250
  Extra Operating Current (mA):	0

hope this helps.

paulober, 2024-01-21:
@steeley @ChrisChoke Added C3 and Teensy

closed 2024-01-21 by paulober

paulober, 2024-01-21:
>  oh my god. @paulober thats so nice. v3.7.1 works like a charm with esp32-wroom. this will make work so much easier. i love it already.
> 
> 
> 
> looking forward to interact with esp32-c3, too 😊
> 
> 
> 
> so my first feedback to you is just: THANK YOU!
> 
> 
> 
> Chris

I am glad that MicroPico is useful to you.

PR #186 (2024-01-21): Patch v3.7.2
### Added
- `ESP32-C3` and `Teensy 4.0` support (experimental) (Fixes #96)