paulober/MicroPico #97: Pico W wont connect

RyuunosukeDS3, 2023-05-29:
## What are the steps to reproduce this issue?
1. Installed Firmware
2. Started a pico project

## What happens?
Simply wont connect and spams "Connection to Pico lost"

[image]

## What were you expecting to happen?
To be able to connect to my pico

## Any logs, error output, etc?
2023-05-29 20:02:28.861 [error] Error: spawn /usr/local/bin/python ENOENT
	at ChildProcess._handle.onexit (node:internal/child_process:285:19)
	at onErrorNT (node:internal/child_process:497:16)
	at process.processTicksAndRejections (node:internal/process/task_queues:83:21

## Support info

Copy this from the `Help -> Info/About -> Copy` option in Visual Studio Code:

```
Version: 1.78.2
Commit: b3e4e68a0bc097f0ae7907b217c1119af9e03435
Date: 2023-05-10T14:47:05.613Z
Electron: 22.5.2
Chromium: 108.0.5359.215
Node.js: 16.17.1
V8: 10.8.168.25-electron.0
OS: Linux x64 6.2.0-76060200-generic
Sandboxed: Yes
```

paulober, 2023-05-30:
That's because it either cannot detect the Pico device file by its product code or it's the permission problem on Linux. The Pico will be "mounted" only for root and a special system group depending on your distro. See Google or the Wiki https://github.com/paulober/Pico-W-Go/wiki/Linux

RyuunosukeDS3, 2023-05-30:
It took me a couple of tries to make this work but apparently it did now!

closed 2023-05-30 by RyuunosukeDS3

paulober, 2023-05-30:
@RyuunosukeDS3 Glad to hear. So it was the permission thing with serial ports on Linux?

RyuunosukeDS3, 2023-05-30:
Aparently yes! I rerun the commands on the docs a couple times and was able to connect. Kinda weird how it took a couple tries but well...