paulober/MicroPico #100: Not able to attach vREPL or re-run "Run" after "Run current file"

laurencepike, 2023-06-07:
Using VS Code Remote SSH - Host on Rasbian OS 32bit (RPI 4B), client Windows 11
Rasbperry Pi set up using the wiki (serial port enabled with login shell disabled)

## What are the steps to reproduce this issue?
1. Open picowgo project, see "Pico Connected"
2. Click "Run" or cntrl+shift+P "Pico-W-Go > Run current file"

## What happens?
Code is uploaded and runs on Pico W (yay)
Notification pops up: "Pico vREPL not open" (maybe oh no?)
"✔️ Pico Connected " in bottom bar

From this point forward I cannot attach vREPL and if I click "Run"/cntrl+shift+P "Pico-W-Go > Run current file" nothing happens

If I physically disconnect the pico usb and reconnect then I can "Run" new code again (one time, like back to the start).

If vREPL is open before I run "Run" for the first time, then "None" is printed in the terminal and then becomes unresponsive

## What were you expecting to happen?
vREPL to open and being able to continue to click "Run" as I update my code

## Any logs, error output, etc?
Protcoll is not listed in output options

## Any other comments?
If this is intended behavior, or something I'm unfamiliar with on the Pico W, very sorry for the inconvenience. I'm not yet very familiar with embedded systems development

## Support info

Copy this from the `Help -> Info/About -> Copy` option in Visual Studio Code:

```
Version: 1.78.2 (user setup)
Commit: b3e4e68a0bc097f0ae7907b217c1119af9e03435
Date: 2023-05-10T14:39:26.248Z
Electron: 22.5.2
Chromium: 108.0.5359.215
Node.js: 16.17.1
V8: 10.8.168.25-electron.0
OS: Windows_NT x64 10.0.22621
Sandboxed: Yes
```
(Note that this is using vscode remote ssh, another bug raised said it was not working, it's working for me minus having to unplug my pico after every code run, not sure if this particular issue is specific to this environment)

paulober, 2023-06-14:
Maybe the problem is related to remote ssh vscode integration like an other issue reports. Does the issue sill persists after v3.0.11?

martinowitsch, 2023-06-15:
I've got the same issue. Device gets connected, I can run code, but I cant get any output in my terminal and got this warning:

[image]

VisualStudioCode Version: 1.79.0
Python: 3.11.3
Pico-W-Go: 3.0.12
Pyserial: 3.5

brehor, 2023-06-17:
I ran into this issue as well.
I found that if I was in the Workspace created by Pico-W-Go when the virtual file system is activated, the vREPL would not open. If I close the workspace and open the folder, everything works normally.

Steps to reproduce:
1. open a configured folder **File>Open Folder**
The vREPL auto opened and is operational.
[image]

2. Click `Toggle Pico-W-FS`
[image]

3. If the REPL is still open, it still works
[image]

4. If I close the REPL and run the program, I get the error
[image]

5. I can no longer open the REPL from there until I open the folder once again.

paulober, 2023-06-18:
@laurencepike Does the issue still exists for you.
I tried following steps:
- Install remote ssh extension on desktop
- Connect to headless Raspberry Pi (with remote ssh)
- It installs the vscode server and sets up vscode
- Install the Pico-W-Go extension on the target raspberry pi
- Open new folder > run configure project
- I connects to the Pico connected to the raspberry pi without any problems 
- When I try to run a file it says vREPL not open
- New Terminal > vREPL
- vREPL opens normally and I can not hit the run button again and it runs

paulober, 2023-06-18:
@martinowitsch New Terminal > Pico (W) vREPL

laurencepike, 2023-06-19:
I just tried again using the steps provided by @brehor and was able to get the vREPL open after toggling the Pico-W-FS (toggling the workspace caused my connection to ssh vscode to need to reload). After that I was able to see vREPL fine, and run/stop code as expected.

paulober, 2023-07-03:
@laurencepike Your issue should be resolve with the release of v3.1.0, can you confirm?

paulober, 2023-10-16:
Closed due to inactivity

closed 2023-10-16 by paulober