paulober/MicroPico #101: After established connection to the board, the main.py program crashes

webdeb, 2023-06-10:
## What are the steps to reproduce this issue?
1. Upload a program with an infinite loop
2. Restart the board, the program runs
3. Connect the board to Pico-W-Go, the program crashes (no logs, it just stops)

## What were you expecting to happen?
The program should run keep running like in the previous version, and the logs from the main program should be accessible

## Any logs, error output, etc?
* Thats the problem, no logs, no outputs.

## Any other comments?
* It makes the reboot on Upload option misleading. One would expect that the board reboots and the extension connects to the board
* In previous versions the workflow was: upload the program, reset the board, everything is showing.
Now the workflow is: upload the program, reset the board, program starts, crashes after pico-w-go connected, now go to main.py start manually.

## Support info

```
Version: 1.79.0
Commit: b380da4ef1ee00e224a15c1d4d9793e27c2b6302
Date: 2023-06-07T14:29:00.206Z (3 days ago)
Electron: 22.5.5
Chromium: 108.0.5359.215
Node.js: 16.17.1
V8: 10.8.168.25-electron.0
OS: Darwin x64 22.3.0
```

webdeb, 2023-06-11:
After reading the other issues, I guess this is the reason: https://github.com/paulober/Pico-W-Go/issues/92#issuecomment-1546905165

@paulober as a quick fix I would like to ask to implement an auto-run script on upload.

paulober, 2023-06-12:
@webdeb Have you tried: [listen2pico](https://github.com/paulober/listen2pico) utility as a quick fix?
The rebootAfterUpload option soft resets the MicroPython REPL to clear memory and old variables but currently it does not support running main or boot after upload anymore, you're right. I'll look into it for one of the next patches.

webdeb, 2023-06-13:
@paulober Hmm.. I don't quite understand how it would fix my problem:

I use extensively upload, and always run the app from main.py because its a multiprogram application, which I am building so, in previous versions it was a nice workflow: 

Upload (per Shortkey), the app reloads and executes the main.py program

Now, its uploading, then the board restarts, but crashes after connections is established. Then you have to execute another reset, and then execute main.py by changing the selected file and pressing run. It doesn't feel straightforward. 

Like suggested, you could provide a simple option `resetAndAutorunOnConnect: main.py` and this would be quick fix. It would not be the same, but it would be a workable solution.

closed 2023-06-20 by paulober