paulober/MicroPico #103: No connection to PICO on Ubuntu 22.04

PySchilling, 2023-06-14:
## What are the steps to reproduce this issue?
1. Flash rp2-pico-20230426-v1.20.0.uf2
2. Reconnect Raspberry Pi Pico
3. Connect Pico

## What happens?

Get message popup "Connection to Pico lost"
No Connection

## What were you expecting to happen?

Pico gets connected

## Any logs, error output, etc?

## Any other comments?

Do the same thing on Windows 10 no problems there.

Device: Raspberry Pi Pico

Permissions
crw-rw---- 1 root dialout 166, 0 Jun 14 14:30 ttyACM0
User Groups
xxx : xxx adm dialout cdrom sudo dip plugdev lpadmin lxd sambashare

## Support info

Copy this from the `Help -> Info/About -> Copy` option in Visual Studio Code:

using extension Pico-W-Go v3.0.10

```
Version: 1.79.1
Commit: 4cb974a7aed77a74c7813bdccd99ee0d04901215
Date: 2023-06-12T16:17:13.990Z
Electron: 22.5.7
Chromium: 108.0.5359.215
Node.js: 16.17.1
V8: 10.8.168.25-electron.0
OS: Linux x64 5.19.0-43-generic
```

paulober, 2023-06-14:
@PySchilling Can you connect to you pico with the [listen2pico tool](https://github.com/paulober/listen2pico) without sudo?
Have you tried to unplug and plug the pico back in (or maybe click on the Pico Disconnected button, wait, and click again to connect) after getting the connection to pico lost notification?

PySchilling, 2023-06-15:
@paulober Thanks for your response. Actually I tried it severeal times. I have the feeling that I don't get a connection at all. I tried it using thonny on the same machine with the same configuration and it works well. listen2pico tool seems to work!

nqdrizzt, 2023-06-19:
I have the same issue running vsc in WSL with remote extension enabled. 
Error: [error] [Extension Host] [[31mERROR[0m] [Activator] Pyboard exited with code 1

I am using the usb2ip tool to pass the usb device from windows to WSL2 running debian bullseye:
https://learn.microsoft.com/en-us/windows/wsl/connect-usb
drizzt@lexa:~/listen2pico$ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 002: ID 2e8a:0005 MicroPython Board in FS mode
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

running the listen2pico tool fails with: 
python3 listen2pico.py /dev/ttyACM0
Traceback (most recent call last):
  File "/home/drizzt/listen2pico/listen2pico.py", line 57, in <module>
    main(sys.argv[1])
  File "/home/drizzt/listen2pico/listen2pico.py", line 38, in main
    pyb = pyboard.Pyboard(dev, baudrate=115200, exclusive=True)
  File "/home/drizzt/listen2pico/pyboard.py", line 317, in __init__
    raise PyboardError("failed to access " + device)
pyboard.PyboardError: failed to access /dev/ttyACM0

drizzt@lexa:~/listen2pico$ groups
drizzt root adm dialoug cdrom sudo dip plugdev

drizzt@lexa:~/listen2pico$ ls -alh /dev/ttyACM0
crw------- 1 root root 166, 0 Jun 19 22:42 /dev/ttyACM0

paulober, 2023-06-19:
@nqdrizzt The reason for that is you have added your self to the group "dialoug" not "dialout". Also in your case (on your distro installation) the device file you think is the Pico is not owned by a special group only by root. That could be a problem cause then only if you run something as root it will have access to this device. I'm not an linux system expert but I think it would be a good step to make this device file co owned by a special group which you can call "dialout" like on other distros.

nqdrizzt, 2023-06-19:
yes you are absolutely right. the following fixed it for me:

sudo chmod g+rw /dev/ttyACM0
sudo chown root:dialout /dev/ttyACM0
useradd <user> dialout

after that u have to logout and login again.

Also worth to mention, if u pass the usb device with the usbipd into the WSL like described [here](https://learn.microsoft.com/en-us/windows/wsl/connect-usb), each restart of the wsl login shell releases the bind and the device gets lost. u need reattach it again and the permissions need to be set again. maybe it can be automated withe a udev rule ... at least the permission setting.

paulober, 2023-10-16:
@PySchilling Does the issue still persists for you with v3.3.2?

ToasterUwU, 2024-02-19:
For other people who might have my issue: If you have a Valve Index VR Headset connected, it might hog the ttyACM0 dev for its audio (idk why), rebooting fixed all my issues in that case, relogging wasn't enough in my case.

paulober, 2024-11-18:
@PySchilling Do you still have this problem with the extension?

paulober, 2026-09-14:
I'm closing this as I couldn't get the details needed to look into it, and a lot has changed since. If it still happens with the latest version, please open a new issue using the bug form. Thanks!

closed 2026-09-14 by paulober