paulober/MicroPico #105: vREPL occasionally hangs upon opening

paulober, 2023-06-14:
Sometimes the REPL occasionally hangs upon opening and causing the other commands (connect/disconnect, run, reset etc) to become unresponsive. Can't seem to find a reliable way to reproduce it.
[image]

_Originally posted by @dburles in https://github.com/paulober/Pico-W-Go/issues/39#issuecomment-1506365738_

paulober, 2023-07-02:
@dburles Does this still happen after v3.1.0?

dburles, 2023-07-03:
Hey @paulober, I haven't been actively developing anything lately to confirm that, however I'll create a new issue if I encounter it again in future.

paulober, 2023-07-03:
> Hey @paulober, I haven't been actively developing anything lately to confirm that, however I'll create a new issue if I encounter it again in future.

Ok, thanks.

closed 2023-07-03 by paulober