paulober/MicroPico #112: Repl jumbled and usable

callen6123, 2023-06-28:
I have been running Pico-W-Go on a different Mac for a while with no issues, however I just installed VS Code on an older iMac (late 2009) running MacOS 10.13.6, and installed Pico-W-Go along with Python, Pylance, and IntelliCode.

My Pi Pico is connected and recognized just fine by VS Code, and a I am able to upload and run code on the Pico just fine, but the Repl gets all jumbled and  unreadable when trying to use it.  I have tried different versions of Pico-W-Go, as well as an older version of VS Code with same results.  I have tried multiple Picos as well, all showing the same.

The Gif below shows what I'm seeing.

[image]

## Any other comments?
I have IntelliCode, Jupyter, Python, Pylance, and Pico-W-Go extensions installed.  

## Which version of Pico-W-Go are you using?
v3.1.1

## Support info

Version: 1.79.2 (Universal)
Commit: 695af097c7bd098fbf017ce3ac85e09bbc5dda06
Date: 2023-06-14T08:58:52.392Z
Electron: 22.5.7
Chromium: 108.0.5359.215
Node.js: 16.17.1
V8: 10.8.168.25-electron.0
OS: Darwin x64 17.7.0

paulober, 2023-06-28:
Do you get any output in "Output > Extension Host" or "Output > Pico-W-Go"?

callen6123, 2023-06-28:
No Warnings or Issue on Extension Host Output, but seeing below for Pico-W-Go output:

2023-06-28 13:17:37.746 [info] [Activator] Connection to wrapper successfully established
2023-06-28 13:17:39.093 [error] [PicoWFs] readFile: File not found
2023-06-28 13:17:39.127 [warning] [PicoWFs] stat: item not found: /.devcontainer/devcontainer.json
2023-06-28 13:17:39.188 [error] [PicoWFs] readFile: File not found
2023-06-28 13:17:39.227 [warning] [PicoWFs] stat: item not found: /pom.xml
2023-06-28 13:17:39.247 [error] [PicoWFs] readFile: File not found
2023-06-28 13:17:39.283 [warning] [PicoWFs] stat: item not found: /build.gradle
2023-06-28 13:17:39.404 [error] [PicoWFs] readFile: File not found
2023-06-28 13:17:39.436 [warning] [PicoWFs] stat: item not found: /pyproject.toml
2023-06-28 13:17:40.064 [warning] [PicoWFs] stat: item not found: /node_modules
2023-06-28 13:17:40.129 [warning] [PicoWFs] stat: item not found: /pom.xml
2023-06-28 13:17:40.316 [warning] [PicoWFs] stat: item not found: /app/src/main/AndroidManifest.xml

callen6123, 2023-06-28:
Looks like it's a VS Code issue.  I went to VS Code Settings and changed "terminal.integrated.gpuAcceleration" to "off".  Seems to be working just fine now.  Thanks for the help, and for the great extension!

closed 2023-06-28 by callen6123