paulober/MicroPico #115: Only enable Pico-W-Go extension for a Pico project

Odud, 2023-07-06:
# Feature Request 🚀

I have a number of projects in my workspace, only one relates to the Pico. The commands such as Run, Reset, All Commands show up in the bottom bar on vscode for everthing. It would be great if there was a way to only display these for a Pico project.  

There is a **workaround** in that I create two workspaces: one for Pico projects, one for everything else. But  I find it clumsy to have to switch between them (there are often cases when I need to copy and paste code between projects)

**Advantages**: Reduce screen clutter, remove "Conection to Pico lost" messages

doge-gif, 2023-07-13:
One way is to use vscode's workspace settings:
1. disable this extension (globally)
2. open the workspace you want to enable
3. in the workspace, enable locally this extension

It's a bit clunky to do this for every new project related to pi pico, but it works regardless of the extension.

paulober, 2023-07-15:
Sorry, I'm not currently working on a fix as many people have their pico projects in a sub folder of their workspace, so if I would only activate if a `.picowgo` file is at the root of the workspace many of them would have problems activating the extension. 
Also this is a common problem among vscode extensions that assist for a certain kind of projects like cmake-tools (one example) which often activates where I don't want it to activate. 
You can temporarily solve this by following the approach @doge-gif mentioned or edit the package.json of the installed extension in your `~/.vscode/extensions` folder after every update and remove the line `"workspaceContains:.picowgo",` from the `activationEvents` that way you would have to manually activate the extension by opening a vREPL.

If you can think of a better solution to this problem, please send it in this issue so I can consider it.

Josverl, 2023-08-03:
VSCode has profiles that enables the use of different sets of extensions and settings across workspaces and projects

https://code.visualstudio.com/docs/editor/profiles

paulober, 2026-09-14:
MicroPico only activates in folders with a `.micropico` file (created by `MicroPico: Initialize MicroPico Project`), so it stays out of other projects, and as of [4.4.0](https://github.com/paulober/MicroPico/releases/tag/v4.4.0) the plotter panel is hidden there too. Thanks for the suggestion!

closed 2026-09-14 by paulober