paulober/MicroPico #117: Python module, untangle, not able to install using 'mip', reporting package not found, but found and installed with PIP

breezin320, 2023-07-21:
> Please provide a clear and concise description of the bug along with logs

## using VSCODE, attempting to install 'untangle', using PICO W REPL

## What happens?
>>> import mip
>>> mip.install('untangle')
Installing untangle (latest) from https://micropython.org/pi/v2 to /lib
Package not found: https://micropython.org/pi/v2/package/6/untangle/latest.json
Package may be partially installed

## What were you expecting to happen?
The package installs with pip, and is found when import in VSCODE Python, but doesn't install with 'mip' (doesn't appear to find it) and can't be found when running on the PICO: reports module not found
## Any logs, error output, etc?
Not sure if this is a mip problem?

Thanks, so much for the help!
dave

paulober, 2023-07-22:
@breezin320 MIP is not using pypi registry so it could be that that's the problem: https://docs.micropython.org/en/latest/reference/packages.html. And have you made sure that you Pico is connected to the internet so it can download the package?

```python3
# As well as downloading packages from the micropython-lib index, mip can also install third-party libraries. The simplest way is to download a file directly:

>>> mip.install("http://example.com/x/y/foo.py")
>>> mip.install("http://example.com/x/y/foo.mpy")
When installing a file directly, the target argument is still supported to set the destination path, but mpy and version are ignored.

# The URL can also start with github: as a simple way of pointing to content hosted on GitHub:

>>> mip.install("github:org/repo/path/foo.py")  # Uses default branch
>>> mip.install("github:org/repo/path/foo.py", version="branch-or-tag")  # Optionally specify the branch or tag
```

breezin320, 2023-07-22:
Thanks, Paul...so I think the issue is untangle is not in the micropython
library, or any other lib that mip pokes at.   The problem I'm trying to
solve is parse an XML page....I can't find anything loadable by mip to
parse.  DOM, untangle, xmltok, etree, SAX, etc...nothing will install ...

Appreciate the quick reply....
Best
Dave

paulober, 2023-07-23:
You can define the download url manually. Look at the documentation link I provided above.

closed 2023-07-23 by paulober